raspiblitz/raspiblitz #134: PoC: Integrate with RTL

saubyk, 2018-10-08:
@rootzoll Opening this issue as discussed.
Please test integration of RTL web UI with Raspiblitz. The instructions for installation are on the repo:
https://github.com/ShahanaFarooqui/RTL/blob/master/README.md

Thanks

geco91, 2018-10-28:
Once npm install ran thru - use the following to test start:

`sudo -u bitcoin node rtl --lndir /mnt/hdd/lnd/data/chain/bitcoin/mainnet`

geco91, 2018-10-28:
This is the service file I am using:

`sudo nano /etc/systemd/system/RTL.service`

This is the content:

```
# Raspibolt RTL: systemd unit for RTL
# /etc/systemd/system/RTL.service

[Unit]
Description=RTL daemon
Wants=lnd.service
After=lnd.service

[Service]
/usr/bin/node /home/admin/RTLFullApplication/RTL/rtl --lndir /home/admin/.lnd/data/chain/bitcoin/mainnet
User=root
Restart=always
TimeoutSec=120
RestartSec=30

[Install]
WantedBy=multi-user.target
```

Then enable and start it:

```
sudo systemctl enable RTL
sudo systemctl start RTL
```

Also need to open port 3000 on the firewall:

```
sudo ufw allow 3000
sudo ufw enable
```

geco91, 2018-10-28:
TODO: adapt RTL.service to litecoin/bitcoin network

rootzoll, 2018-11-04:
Added RTL install/deinstall to the mainmenu. Will be part of next SD card release.

Also added a feature request to the RTL development:
https://github.com/ShahanaFarooqui/RTL/issues/4

rootzoll, 2018-12-05:
OK. RTL is now part if the service options and test ran OK. Will close ticket for now.

closed 2018-12-05 by rootzoll