raspiblitz/raspiblitz #135: Feature request: Check HDD for preloaded blockchain

Smiggel, 2018-10-12:
I would like to see a future added for a preloaded blockchain. I had an Raspibolt setup running and wanted to use the same blockchain with Raspiblitz. I don't see this option and the setup scripts just formats the hdd when it's attached. Even when a preloaded blockchain is on it. Which is too bad.

Would be a nice extra feature to the setup process.

rootzoll, 2018-10-16:
This is already a feature - wo do this when we hand out raspiblitzes on hackevents. Important is that the HDD is formatted to ext4 and named BLOCKCHAIN ... the "bitcoin" folder then needs to be in the root of the HDD.

closed 2018-10-16 by rootzoll

Smiggel, 2018-10-16:
Hmmm ok. I guess my problem was the name of the HDD. It was not called BLOCKCHAIN i guess.