raspiblitz/raspiblitz #136: Measure Power Consumption

rootzoll, 2018-10-12:
As a statistic it would be great to have some powerconsumption measuremnts of the RaspiBlitz (in sysned up - regular running mode).

vmavromatis, 2018-10-12:
Wow that would be an extremely cool feature. If we assume the voltage stays at 5V you need to measure just the current right? So if it's something like 1A (because we are touchscreen + HDD, am I right or completely off???) that's 5W or 5*24=120Wh daily or 43.8kWh annually. 

Now to make this even cooler you can get the average kWh price (e.g. 0.2eurocents) from the country based on the external IP? 

43.8kWh * 0.2eurocents = 8.76Euros / year

Just thinking out loud maybe my math is way off

rootzoll, 2018-10-20:
With my cheap measuremen box from amazon its showing the following balues:
around 5,5W on idle
up to 7,0W on regular workload
up to 8,5W on short peaks

So if we calculate with 7W we should be on the safe side.

closed 2018-10-20 by rootzoll