raspiblitz/raspiblitz #138: Locale errors

torkelrogstad, 2018-10-13:
When building my own image, the console output is littered with warnings from primarily Perl, but also other programs. The Perl warning:

```
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LC_TIME = "nb_NO.UTF-8",
        LC_MONETARY = "nb_NO.UTF-8",
        LC_ADDRESS = "nb_NO.UTF-8",
        LC_TELEPHONE = "nb_NO.UTF-8",
        LC_NAME = "nb_NO.UTF-8",
        LC_MEASUREMENT = "nb_NO.UTF-8",
        LC_IDENTIFICATION = "nb_NO.UTF-8",
        LC_NUMERIC = "nb_NO.UTF-8",
        LC_PAPER = "nb_NO.UTF-8",
        LANG = "en_GB.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_GB.UTF-8").
locale: Cannot set LC_ALL to default locale: No such file or directory
```

Here's a [tutorial](https://daker.me/2014/10/how-to-fix-perl-warning-setting-locale-failed-in-raspbian.html) on how to fix this problem interactively. There's probably a way of doing this through a script. It would be nice to have a cleaner output on building, that way I could actually follow what's being done.

rootzoll, 2018-11-04:
Thanks. I will take a quick look at this and see if I can blind fix it for version 1.0, because its no reproducing on my machine.

rootzoll, 2018-12-03:
OK I added locales and set the LANG explicitly for building. Works good on my tests. Hope this will fix this. Closing issue.

closed 2018-12-03 by rootzoll