raspiblitz/raspiblitz #141: Handling Errors from Power Outage Safely - Repair script?

mr-dakota, 2018-10-18:
After running successfully for a month, my raspiblitz has hit into a first issue. Appears to have lost power briefly which has caused an EXT4-fs error when rebooted.

Process I think may help will be to unmount the hdd, perform fsck and remount. Hopefully that will be sufficient, but otherwise I may need to reindex-chainstate for bitcoind. 

Has anyone successfully resolved a similar issue at all? This could be a useful script to put together for version 1.0 to handle 'repair' functions.

mr-dakota, 2018-11-18:
For what it is worth, the fsck & reindex process I mentioned was successful - it took the best part of 14 days to reindex-chainstate though.

rootzoll, 2018-11-27:
I would need more details on what would be the repair-options. So removing this from milestone vor 1.0 and maybe adding repair-scripts later on.

geco91, 2019-02-26:
In v1.0 there is now basic detection if re-index/repair is needed and the torrent and scp scripts for getting the blockchain can already be used to replace corrupt data again. I would recommend to make this a better UX and offer repairs more user friendly for v1.1

rootzoll, 2019-03-14:
Also check if anything of the following can be added to a repairscript: https://www.tecmint.com/check-linux-hard-disk-bad-sectors-bad-blocks/

rootzoll, 2019-03-17:
To simulate the "data corruption" (for testing) its possible to just go into the directory `/mnt/hdd/bitcoin/blocks/index` and `echo "" > [latest-ldb-file]` and then make reboot.

rootzoll, 2019-03-17:
For v1.1 when on SSH into the RaspiBlitz its detecting a reindex/data corruption it will offer now multiple options:

- download torrent again
- copy blockchain from another computer
- start resync by itself

also a rescue backup option is offerred to do first optional.

rootzoll, 2019-03-19:
testing worked fine. closing issue.

closed 2019-03-19 by rootzoll