raspiblitz/raspiblitz #143: [lncli] rpc error: code = Unavailable desc = grpc: the connection is unavailable

Smiggel, 2018-10-18:
Did a fresh install tonight. Noticed in can generate a new address. When I get info in the terminal I see the following:

[lncli] rpc error: code = Unavailable desc = grpc: the connection is unavailable

Looking in the lnd.conf it all seems to look ok. Password is also set to my own. 

Can not unlock my wallet now and sync it, or do anything else.

Someone else experiencing this issue?

vnnkl, 2018-10-18:
see if the following helps as my issue was similar,

go to your bitcoin.conf, as user admin:
`cd /home/bitcoin/.bitcoin/`

`sudo nano bitcoin.conf`

change in `bitcoin.conf` the following lines:
```
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333
```

to 
```
zmqpubrawblock=tcp://*:28332
zmqpubrawtx=tcp://*:28333
```

try restarting bitcoind by `sudo systemctl restart bitcoind`
then try `lncli unlock` again

wonder if that does the trick for you as well

rootzoll, 2018-10-18:
added your changes to the general config .. will run tests.

vnnkl, 2018-10-18:
got it from an LND slack answer, not sure if this is good as I guess it works as a placeholder for whatever (allow non localhost to access zmq), but i guess that would be caught by lazy firewall?

check your slack messages btw :P

vnnkl, 2018-10-19:
seems like i was lucky one time, did break it again and am back to same issue

Smiggel, 2018-10-19:
No luck here. When I look at the logs now I see:
Oct 19 08:57:47 Raspibolt systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Oct 19 08:57:47 Raspibolt systemd[1]: lnd.service: Unit entered failed state.
Oct 19 08:57:47 Raspibolt systemd[1]: lnd.service: Failed with result 'exit-code'.

Smiggel, 2018-10-19:
> seems like i was lucky one time, did break it again and am back to same issue

You don't have any special characters in your password, like I had? That caused my problem.

rootzoll, 2018-11-04:
About the special characters I opened a seperate issue:
https://github.com/rootzoll/raspiblitz/issues/162