raspiblitz/raspiblitz #144: bootloop

vnnkl, 2018-10-18:
keep being stuck in bootloop after fresh install (build myself), 
did manually apply changes from this commit (https://github.com/rootzoll/raspiblitz/commit/f9becbc29b0eaea5471af43e086392ef7f7af702), 
which did not help.

any pointers what to check? seems similar or same issue as #143 

Below log:
```
2018-10-18 23:48:29.129 [INF] LTND: Version: 0.5.0-beta commit=v0.5-beta-101-g61e867741926bcb318432a6344b80161fabd1455, build=production, logging=default
2018-10-18 23:48:29.129 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2018-10-18 23:48:29.130 [INF] CHDB: Checking for schema update: latest_version=6, db_version=6
2018-10-18 23:48:29.159 [INF] RPCS: password RPC server listening on [::]:10009
2018-10-18 23:48:29.159 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
2018-10-18 23:48:29.161 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
2018-10-18 23:48:47.847 [INF] LNWL: Opened wallet
2018-10-18 23:48:48.533 [INF] LTND: Primary chain is set to: bitcoin
2018-10-18 23:48:48.590 [INF] LTND: Shutdown complete

2018-10-18 23:49:48.928 [INF] LTND: Version: 0.5.0-beta commit=v0.5-beta-101-g61e867741926bcb318432a6344b80161fabd1455, build=production, logging=default
2018-10-18 23:49:48.929 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2018-10-18 23:49:48.929 [INF] CHDB: Checking for schema update: latest_version=6, db_version=6
2018-10-18 23:49:48.935 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
2018-10-18 23:49:48.936 [INF] RPCS: password RPC server listening on [::]:10009
2018-10-18 23:49:48.938 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
2018-10-18 23:53:40.658 [INF] LNWL: Opened wallet
2018-10-18 23:53:41.205 [INF] LTND: Primary chain is set to: bitcoin
2018-10-18 23:53:41.384 [INF] LTND: Shutdown complete

2018-10-18 23:54:42.194 [INF] LTND: Version: 0.5.0-beta commit=v0.5-beta-101-g61e867741926bcb318432a6344b80161fabd1455, build=production, logging=default
2018-10-18 23:54:42.194 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2018-10-18 23:54:42.195 [INF] CHDB: Checking for schema update: latest_version=6, db_version=6
2018-10-18 23:54:42.203 [INF] RPCS: password RPC server listening on [::]:10009
2018-10-18 23:54:42.203 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
2018-10-18 23:54:42.205 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
```

vnnkl, 2018-10-18:
i was able to break my bootloop:

changed in bitcoin.conf:
```
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333
```

to 
```
zmqpubrawblock=tcp://*:28332
zmqpubrawtx=tcp://*:28333
```

did work after a restart of bitcoind

rootzoll, 2018-10-18:
added your changes to the general config .. will run tests.

rootzoll, 2018-10-26:
seems to run fine so far. closing the issue for now.

closed 2018-10-26 by rootzoll