raspiblitz/raspiblitz #145: reboot resets autopilot to 0

vnnkl, 2018-10-19:
after every reboot the in `lnd.conf` the autopilot is being reset to
```
autopilot.active=0
```

geco91, 2018-10-26:
The autopilot is the first test for the new raspiblitz settings/config file - see details:
https://github.com/rootzoll/raspiblitz/blob/master/background_raspiblitzSettings.md

At the moment make sure that there is a "autoPilot=on" in the /mnt/hdd/raspiblitz.conf, because this setting gets enforced on every start.

geco91, 2018-10-26:
@rootzoll on test runs there is a double entry of "autoPilot=on" in the config file. check why.

rootzoll, 2018-11-04:
yep  - this has to be fixed for version 1.0. I am putting it on the list.

rootzoll, 2018-12-08:
ok - this is fixed for next release.

closed 2018-12-08 by rootzoll