raspiblitz/raspiblitz #148: Illegal instruction

CryptoSigmund, 2018-10-20:
> TAIL Chain Network Log
> tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory
> 
> UNKOWN STATE - there is blockain data folder, but blockchaind is not running
> It seems that something went wrong during sync/download/copy of the blockchain.
> If you want start fresh --> sudo rm -r /mnt/hdd/bitcoin
> Or something with the config is not correct.

After copy from hdd to hdd and the same issue after just downloading the blockchain via torrent. 
raspiblitz-2018-09-22

Do you have any advice that could help in this situation?

CryptoSigmund, 2018-10-22:
After testing this with the new version from raspiblitz-2018-10-20.img I get the same error. This was a fresh sync of the blockchain.

Any advice where to start for the failure?

> *** Update System ***
> raspberrypi-bootloader was already set on hold.
> Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
> Get:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
> Fetched 15.0 kB in 4s (3,047 B/s)
> Reading package lists... Done
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> Calculating upgrade... Done
> 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
> OK - System is now up to date
> HDD was already inited or prepared
> Now needs to be mounted
> 
> *** Adding HDD to the System ***
> OK - HDD found as sda
> 
> *** Check HDD ***
> OK - HDD is formatted with ext4 and is named BLOCKCHAIN
> Adding line to /etc/fstab ...
> UUID=8d33bc18-c675-455e-acb2-c9360d4eb56b /mnt/hdd ext4 noexec,defaults 0 0
> OK - HDD is listed in /etc/fstab
> 
> *** Mount HDD ***
> OK - HDD is mounted
> 
> *** Analysing HDD Content ***
> Looks like the HDD is prepared with the Blockchain.
> Continuing with finishing the system setup ...
> 
> *** Checking HDD ***
> OK - HDD is ready.
> 
> *** Prepare bitcoin ***
> mkdir: cannot create directory ‘/mnt/hdd/lnd’: File exists
> OK - bitcoin setup ready
> 
> *** Start bitcoin ***
> This can take a while ..
> Created symlink /etc/systemd/system/multi-user.target.wants/bitcoind.service → /etc/systemd/system/bitcoind.service.
> Job for bitcoind.service failed because a fatal signal was delivered to the control process.
> See "systemctl status bitcoind.service" and "journalctl -xe" for details.
> Giving bitcoind service 180 seconds to init - please wait ...
> OK - bitcoind started
> 
> Illegal instruction
> TAIL Chain Network Log
> tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory
> 
> UNKOWN STATE - there is blockain data folder, but blockchaind is not running
> It seems that something went wrong during sync/download/copy of the blockchain.
> If you want start fresh --> sudo rm -r /mnt/hdd/bitcoin
> Or something with the config is not correct.
> admin@RaspiBlitz:~ $

Emzy, 2018-10-22:
The problem was that the CPU is a "ARMv6-compatible processor rev 7 (v6l)".
This can't execute the bitcoind binary "Illegal instruction"

geco91, 2018-10-25:
@Emzy so this a problem with the hardware (not a raspberry 3) or is there something to change in the RaspBlitz setup.

CryptoSigmund, 2018-10-27:
It was the raspi 2 V1.2 which couses some problems with the insctruction set in couse by hardware limitation. So RaspiBlitz is fine.

rootzoll, 2018-11-04:
@CryptoSigmund Ok thanks. Will close ticket for now.

closed 2018-11-04 by rootzoll