raspiblitz/raspiblitz #151: Verification check post-Torrent download is failing...

georgevaccaro, 2018-10-22:
This is the message that keeps looping (here's 3 iterations):

*** Test if chaninnetwork is ready ...
1540216573
result()
error(error code: -28
error message:
Verifying blocks...)
2018-10-22T12:35:25Z Opened LevelDB successfully
2018-10-22T12:35:25Z Using obfuscation key for /home/bitcoin/.bitcoin/indexes/txindex: 0000000000000000
2018-10-22T12:35:25Z Upgrading txindex database... [0%]
2018-10-22T13:05:26Z Upgrading txindex database... [10%]
2018-10-22T13:42:43Z Upgrading txindex database... [20%]
Waiting 1 minute and then trying again ...

*** Test if chaninnetwork is ready ...
1540216633
result()
error(error code: -28
error message:
Verifying blocks...)
2018-10-22T12:35:25Z Opened LevelDB successfully
2018-10-22T12:35:25Z Using obfuscation key for /home/bitcoin/.bitcoin/indexes/txindex: 0000000000000000
2018-10-22T12:35:25Z Upgrading txindex database... [0%]
2018-10-22T13:05:26Z Upgrading txindex database... [10%]
2018-10-22T13:42:43Z Upgrading txindex database... [20%]
Waiting 1 minute and then trying again ...

*** Test if chaninnetwork is ready ...
1540216693
result()
error(error code: -28
error message:
Verifying blocks...)
2018-10-22T12:35:25Z Opened LevelDB successfully
2018-10-22T12:35:25Z Using obfuscation key for /home/bitcoin/.bitcoin/indexes/txindex: 0000000000000000
2018-10-22T12:35:25Z Upgrading txindex database... [0%]
2018-10-22T13:05:26Z Upgrading txindex database... [10%]
2018-10-22T13:42:43Z Upgrading txindex database... [20%]
Waiting 1 minute and then trying again ...

Unfortunately I lost the download details in the scrollback buffer but when I looked at it, it seemed like the actual download size was 4 bytes larger than expected.

Other than that the install has been very smooth.

geco91, 2018-10-25:
pull request #155 adds some information on the error -28

@georgevaccaro which sd card image did you use? The latest raspiblitz-2018-10-20.img.gz should already have torrents updated txindex, so this waiting time should not be necessary anymore.

georgevaccaro, 2018-10-26:
Thanks geco91.  I should have mentioned it in the issue.  I was using raspiblitz-2018-09-22.img.

Ah, I see, I grabbed it just before it got updated in the README.md here:

https://github.com/rootzoll/raspiblitz/commit/2737abc6eb84d9afbbf37a5d8d31e8d90bcbb766#diff-04c6e90faac2675aa89e2176d2eec7d8

Do you think it's worth a complete rebuild?  It seems to be running OK otherwise though I'm not sure where to look for issues.  I was hoping to get this all going before LighningHackDay in NYC this weekend.

I can always toss a new microsd in it if you think it will build before then...

Any advice is appreciated. - Thank you!

rootzoll, 2018-10-26:
The october release has the final bitcoind 0.17.0 - not the release candidate anymore. But for basic lightning use cases the old image should be fine also.

About updating an sd card image see: https://github.com/rootzoll/raspiblitz#update-to-a-new-sd-card-release

See you in NY :) Closing issue for now.

closed 2018-10-26 by rootzoll