raspiblitz/raspiblitz #152: Add more through info on what to expect post-torrent but pre lightning service start

IvanThinking, 2018-10-22:
I was following the instructions verbatim and found two confusing things so far. It would be nice to fix the documentation to describe them.

1. When my torrent was done (100%) it just sat there, refreshing at 100% forever... I had to press X to quit out of it. The on-screen instructions suggested that pressing X = ABORT only and I thought I might screw up my download. Better instructions there or on the github walkthrough would be cool.

2. After I exited the torrent, I got the same thing that was reported here: https://github.com/rootzoll/raspiblitz/issues/112
The issue is that "error" suggests, well, a PROBLEM, and a n00b like me assumes the script is blown up in some way, especially with no warning in the documentation (onscreen or online). Nothing in the walkthrough prepares you for this and so it seems like something failed or is broken. It's only after you wait a while that it flips to 10%....20% etc. It takes a good long time (for my system) to finish up. Would be nice if the "error" was more clear AND if the walkthrough printed this message/screen along with "you will now see this junk... it's the block verification step and will take a couple hours. It's also not really an error in that something is wrong, so be cool and patient."

IvanThinking, 2018-10-25:
> @IvanThinking <https://github.com/IvanThinking> thanks for the feedback.
>
>    1.
>
>    Normally the torrent download should detect that its ready and
>    continue by itself. But you are not the first one to report, that is not
>    happening reliable. So for this case I would like to add some more text
>    information to the screen, letting people know thats OK to press X to close
>    the download manually. Because I cannot reproduce this scenario on my side
>    I have one question: When you pressed x on 100% did you afterwards got an
>    dialog telling you that the download failed or was its just starting to
>    moving files and worked on from then?
>    2.
>
>    rebuilding the index should anymore needed with the new sd card image
>    and torrent files (raspiblitz-2018-10-20.img.gz) ... but the error -28
>    could still happen, so will add an additional info to let people know at
>    least, that error -28 is normally means just to wait a bit more.

1. When I pressed X, and held my breath, the scripts kept executing (no
fail dialog) it just kept going...so I figured "cool? I think it's ok?"

2. Very nice. Yes, the "error" with 0% showing and no clue what comes next
(is it stuck in an infinite loop?) sent me to Google where I found exactly
one (1) match for the error someone mentioned last month. Thankfully, they
also mentioned that you just have to sit back and wait, so I did, but it
was a bit bothersome as I thought something had failed.

rootzoll, 2018-10-26:
I merged the additional info. Thanks for the fresh look. Closing issue.

closed 2018-10-26 by rootzoll