raspiblitz/raspiblitz #160: re-check for dhcp-server from time to time

jagottsicher, 2018-10-27:
after a broken internet connection or a broken connection to the dhcp-server in the local network the raspiblictz gets an ip like 169.x.x.x 

you can't connect and get a proper ip again. All you can do is power off, which in many cases leads to a corrupted database. 

the dhcp client should check from time to time for a new ip if the ip starts with 169 OR there should be an emergency reboot or something!

rootzoll, 2018-11-04:
Yep a background checking process would make sense to keep track of that and trigger actions. 

Stopping and restarting LND seems like the way to go for now in case the global IP changed, because there is no manual way to add/remove IPs while running.

Adding to my version 1.0 list.

rootzoll, 2018-12-11:
OK this is now part of the new background service - will be part of next sd card release
was not able to test it - but reboot is triggered after max 5min when local IP starts with 169.x.x.x

closed 2018-12-11 by rootzoll

phenomeus, 2020-07-07:
can we please try to differentiate between eth0 + wlan0 but also **v**ethxxxx

on my rpi4 I have docker and other stuff running and the virtual eth gets checked and the reboot triggered. I would suggest that the DHCP only checks eth0 or/ and wifi wlan0

phenomeus, 2020-07-12:
please see #1337