raspiblitz/raspiblitz #161: Could not locate RPC credentials

charona, 2018-10-28:
Set up 2 raspiblitzes, and synced the blockchain from an external HDD. final blockchain sync finshed. all fine.

except that the mainnet sync shows up empty (even after 12 hours):
               RaspiBlitz v0.95  h....z
               bitcoin Fullnode + Lightning Network + TOR
        ,/     -------------------------------------------
      ,'/      load average: 0.32, 0.32, 0.29, CPU 60°C
    ,' /       Free Mem 596M / 927M  Free HDD 648G (70%)
  ,'  /_____,  ssh admin@192.168.1.106 ▼8.4MiB ▲4.5MiB
 .'____    ,'  
      /  ,'    bitcoin v0.17.0 mainnet Sync  ()
     / ,'      Public : 
    /,'        
   /'          LND v0.5-beta  waiting for chain sync
               scanning 547646/547649

0....................e@3.....26:9735

The cli output shows this:
****************************************************************************
Unlock LND Wallet --> lncli --chain=bitcoin unlock
****************************************************************************
HELP: Enter your PASSWORD C
You may wait some seconds until you get asked for password.
****************************************************************************
error: Could not locate RPC credentials. No authentication cookie could be found, and RPC password is not set.  See -rpcpassword and -stdinrpcpass.  Configuration file: (/home/bitcoin/.bitcoin/bitcoin.conf)
Input wallet password: 

lnd successfully unlocked!
tail: cannot open '/mnt/hdd/lnd/logs/bitcoin/net/lnd.log' for reading: No such file or directory

However, /home/bitcoin/.bitcoin/bitcoin.conf exists, is readable and seems to have the right RPC credentials in it.

Any idea how I can cure this?

charona, 2018-10-29:
One problem solved by creating a symbolic link thru ssh command:
sudo ln -s /mnt/hdd/lnd/logs/bitcoin/mainnet /mnt/hdd/lnd/logs/bitcoin/net

Also added stdinrpcpass to the bitcoin.conf file, but still getting the same error message.

rootzoll, 2018-11-04:
@charona  "/mnt/hdd/lnd/logs/bitcoin/net/lnd.log" is for sure an unvalid directory by standard. There seems something wrong with the "chain" parameter detection within the shell script.

I put this on my checking list for version 1.0.

playak, 2018-11-12:
Same here. Main problem is that bitcoin sync stays on empty, and public as well. I assume the bitcoin node is thus not really running.

playak, 2018-11-24:
No help here really? I built 2 blitzes, but both are standing here doing nothing. That is, I assume that the status screen showing:

bitcoin v0.17.0 mainnet Sync ()
Public :

means that something is wrong.

Please somebody help me get this stuff running. Otherwise I'll just go for a Casa or nodl I'm afraid....

rootzoll, 2018-11-26:
@playak it needs more info for a diagnose of "what is wrong". Please generate a debug report and share it - see how to do here: https://github.com/rootzoll/raspiblitz/blob/master/README.md#get-a-debug-report

playak, 2018-11-26:
Great, thanks, happy to share that:

$ ./XXdebugLogs.sh

*** RASPIBLITZ LOGS ***
blitzversion: 0.96
chainnetwork: bitcoin / main
 17:48:54 up 5 min,  2 users,  load average: 3.70, 2.19, 0.95

*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2018-11-26 17:43:23 UTC; 5min ago
  Process: 606 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 608 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─608 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

Nov 26 17:43:22 MYraspiblitz systemd[1]: Starting Bitcoin daemon...
Nov 26 17:43:23 MYraspiblitz systemd[1]: Started Bitcoin daemon.

*** LAST 20 CHAINNETWORK LOGS ***
2018-11-26T17:45:29Z dnsseed thread exit
2018-11-26T17:45:31Z txindex is enabled at height 551591
2018-11-26T17:45:31Z txindex thread exit
2018-11-26T17:45:34Z New outbound peer connected: version: 70015, blocks=551591, peer=1
2018-11-26T17:45:35Z New outbound peer connected: version: 70015, blocks=551591, peer=2
2018-11-26T17:45:41Z New outbound peer connected: version: 70015, blocks=551591, peer=3
2018-11-26T17:45:48Z New outbound peer connected: version: 70015, blocks=551591, peer=5
2018-11-26T17:45:49Z New outbound peer connected: version: 70015, blocks=551591, peer=6
2018-11-26T17:46:02Z New outbound peer connected: version: 70015, blocks=551591, peer=7
2018-11-26T17:46:03Z New outbound peer connected: version: 70015, blocks=0, peer=8
2018-11-26T17:46:04Z New outbound peer connected: version: 70015, blocks=551591, peer=9
2018-11-26T17:48:03Z ThreadRPCServer incorrect password attempt from 127.0.0.1:36646
2018-11-26T17:48:20Z ThreadRPCServer incorrect password attempt from 127.0.0.1:36676
2018-11-26T17:48:20Z ThreadRPCServer incorrect password attempt from 127.0.0.1:36678
2018-11-26T17:48:21Z ThreadRPCServer incorrect password attempt from 127.0.0.1:36682
2018-11-26T17:48:35Z ThreadRPCServer incorrect password attempt from 127.0.0.1:36740
2018-11-26T17:48:36Z ThreadRPCServer incorrect password attempt from 127.0.0.1:36742
2018-11-26T17:48:36Z ThreadRPCServer incorrect password attempt from 127.0.0.1:36746
2018-11-26T17:48:54Z ThreadRPCServer incorrect password attempt from 127.0.0.1:36774
2018-11-26T17:48:54Z ThreadRPCServer incorrect password attempt from 127.0.0.1:36776

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2018-11-26 17:43:23 UTC; 5min ago
 Main PID: 609 (lnd)
   CGroup: /system.slice/lnd.service
           └─609 /usr/local/bin/lnd --externalip=31.164.173.26

Nov 26 17:48:39 MYraspiblitz lnd[609]: 2018-11-26 17:48:39.722 [DBG] RPCS: [walletbalance] Total balance=0 BTC
Nov 26 17:48:40 MYraspiblitz lnd[609]: 2018-11-26 17:48:40.012 [INF] RPCS: [listchannels] fetched 0 channels from DB

*** LAST 20 LND LOGS ***
-- Logs begin at Mon 2018-11-26 17:43:08 UTC, end at Mon 2018-11-26 17:48:55 UTC. --
Nov 26 17:48:27 MYraspiblitz lnd[609]: 2018-11-26 17:48:27.732 [INF] LNWL: Started rescan from block 000000000000000000173c811e79858808abc3216af607035973f002bef60a7a (height 551591) for 1 address
Nov 26 17:48:27 MYraspiblitz lnd[609]: 2018-11-26 17:48:27.742 [INF] LNWL: Starting rescan from block 000000000000000000173c811e79858808abc3216af607035973f002bef60a7a
Nov 26 17:48:27 MYraspiblitz lnd[609]: 2018-11-26 17:48:27.829 [INF] HSWC: Payment circuits loaded: num_pending=0, num_open=0
Nov 26 17:48:28 MYraspiblitz lnd[609]: 2018-11-26 17:48:28.005 [INF] LNWL: Rescan finished at 551591 (000000000000000000173c811e79858808abc3216af607035973f002bef60a7a)
Nov 26 17:48:28 MYraspiblitz lnd[609]: 2018-11-26 17:48:28.005 [INF] LNWL: Catching up block hashes to height 551591, this might take a while
Nov 26 17:48:28 MYraspiblitz lnd[609]: 2018-11-26 17:48:28.129 [INF] LNWL: Done catching up block hashes
Nov 26 17:48:28 MYraspiblitz lnd[609]: 2018-11-26 17:48:28.129 [INF] LNWL: Finished rescan for 1 address (synced to block 000000000000000000173c811e79858808abc3216af607035973f002bef60a7a, height 551591)
Nov 26 17:48:28 MYraspiblitz lnd[609]: 2018-11-26 17:48:28.814 [INF] RPCS: RPC server listening on [::]:10009
Nov 26 17:48:28 MYraspiblitz lnd[609]: 2018-11-26 17:48:28.817 [INF] RPCS: gRPC proxy started at 127.0.0.1:8080
Nov 26 17:48:28 MYraspiblitz lnd[609]: 2018-11-26 17:48:28.860 [INF] LTND: Waiting for chain backend to finish sync, start_height=551591
Nov 26 17:48:28 MYraspiblitz lnd[609]: 2018-11-26 17:48:28.941 [INF] LTND: Chain backend is fully synced (end_height=551591)!
Nov 26 17:48:29 MYraspiblitz lnd[609]: 2018-11-26 17:48:29.380 [INF] NTFN: New block epoch subscription
Nov 26 17:48:29 MYraspiblitz lnd[609]: 2018-11-26 17:48:29.389 [INF] HSWC: Starting HTLC Switch
Nov 26 17:48:29 MYraspiblitz lnd[609]: 2018-11-26 17:48:29.394 [INF] NTFN: New block epoch subscription
Nov 26 17:48:29 MYraspiblitz lnd[609]: 2018-11-26 17:48:29.422 [INF] DISC: Authenticated Gossiper is starting
Nov 26 17:48:29 MYraspiblitz lnd[609]: 2018-11-26 17:48:29.422 [INF] BRAR: Starting contract observer, watching for breaches.
Nov 26 17:48:29 MYraspiblitz lnd[609]: 2018-11-26 17:48:29.422 [INF] NTFN: New block epoch subscription
Nov 26 17:48:29 MYraspiblitz lnd[609]: 2018-11-26 17:48:29.451 [INF] CRTR: FilteredChainView starting
Nov 26 17:48:39 MYraspiblitz lnd[609]: 2018-11-26 17:48:39.722 [DBG] RPCS: [walletbalance] Total balance=0 BTC
Nov 26 17:48:40 MYraspiblitz lnd[609]: 2018-11-26 17:48:40.012 [INF] RPCS: [listchannels] fetched 0 channels from DB

playak, 2018-11-26:
Thanks for your help. The values of bitcoind.rpcpass and rpcpassword are exactly the same in all 4 files you mentioned, so that doesn't seem to be the problem.

playak, 2018-11-27:
$ cat /home/bitcoin/.bitcoin/bitcoin.conf
# bitcoind configuration

# mainnet/testnet - to switch just comment/uncomment
#testnet=1

# Bitcoind options
server=1
daemon=1
txindex=1
disablewallet=1

# Connection settings
rpcuser=raspibolt
rpcpassword=T3stpassword.
zmqpubrawblock=tcp://127.0.0.1:28332
zmqpubrawtx=tcp://127.0.0.1:28333

# jeroen added next line. does not help it seems
stdinrpcpass=T3stpassword.

# Raspberry Pi optimizations
dbcache=100
maxorphantx=10
maxmempool=50
maxconnections=40
maxuploadtarget=5000

datadir=/home/bitcoin/.bitcoin

Screenshot of error message:
https://imgur.com/a/9rAxzYz

playak, 2018-12-04:
No more ideas on how to get my Raspiblitz working?

rootzoll, 2018-12-05:
I cannot put my finger on it from remote  - I would recommend waiting for the next SD-Card release and starting fresh. If its then still not working for you - maybe a casa or a nodl are a better fit for you.

closed 2018-12-05 by rootzoll