raspiblitz/raspiblitz #164: failed connection to rpc

nilz90, 2018-10-31:
I have a problem with the roc. Got this error:
`error: Could not connect to the server 127.0.0.1:8332`

**debug.log in .blockchain :** 
[...]
2018-10-31T22:09:15Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2018-10-31T22:09:15Z Using 4 threads for script verification
2018-10-31T22:09:15Z scheduler thread start
2018-10-31T22:09:15Z Binding RPC on address ::1 port 8332 failed.
2018-10-31T22:09:15Z HTTP: creating work queue of depth 16
2018-10-31T22:09:15Z No rpcpassword set - using random cookie authentication.
2018-10-31T22:09:15Z Generated RPC authentication cookie /home/admin/.bitcoin/.cookie
[...]

**on startup:**
****************************************************************************
Unlock LND Wallet --> lncli --chain=bitcoin unlock
****************************************************************************
HELP: Enter your PASSWORD C
You may wait some seconds until you get asked for password.
****************************************************************************
error: Could not connect to the server 127.0.0.1:8332

Make sure the bitcoind server is running and that you are connecting to the correct RPC port.
Input wallet password: 

**after unlocking the wallet, going to info:**
               RaspiBlitz v0.95  
               bitcoin Fullnode + Lightning Network + TOR
        ,/     -------------------------------------------
      ,'/      load average: 2.44, 1.06, 0.41, CPU 55°C
    ,' /       Free Mem 665M / 927M  Free HDD 622G (67%)
  ,'  /_____,  ssh admin@192.168.178.32 ▼30.2KiB ▲33.8KiB
 .'____    ,'  
      /  ,'    bitcoin v0.17.0 mainnet Sync  ()
     / ,'      Public : 
    /,'        
   /'          LND v0.5-beta  Not Started | Not Ready Yet
               

Screen is not updating ... press ENTER to continue. 

some ideas for troubleshoot ?

redmonkey80, 2018-11-02:
Hi everyone, 

I have the same problem, after i activated TOR. Before everything was running without problem, now i get the same messages:

`error: Could not connect to the server 127.0.0.1:8332
Make sure the bitcoind server is running and that you are connecting to the correct RPC port.`

Does someone know how to solve that?

Thanks

WetPuppyKisses, 2018-11-10:
I am having the same issue. Did you find a fix?

nilz90, 2018-11-10:
> I am having the same issue. Did you find a fix?

no, I'm continuing using the working ip4 connection. 

I installed and set up everything on the ip6 network and then I switched without any changes to the system to ip4 and there it started to work. So I think it's some kind of a router issue.

WetPuppyKisses, 2018-11-10:
@nilz90 how did you manage to force a ipv4 connection?

I tried:
`sudo nano /etc/sysctl.conf``
and then add the following
`net.ipv6.conf.all.disable_ipv6=1`

save and reboot, but I am getting the same error.

GeorgeSpark-Stahl, 2018-11-11:
I'm also having this issue - I was able to temporarily(until the next reboot) fix it by killing the bitcoin process and re-launching it: `sudo /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid`

I am running it as `admin` so, maybe that has something to do with it?  Maybe the `bitcoin` user doesn't have the permissions to open the ports?  Or maybe there is a timing issue with how it's launched on boot?  I tried launching `bitcoind` as the `bitcoin` user but I was seeing different permission issues around accessing the chain db.  

I also had to restart the lnd service with `sudo systemctl restart lnd`

CryptoSigmund, 2018-11-11:
Same problem here. Did anyone investigate it further?

>                RaspiBlitz v0.95  Wallet Locked
>                bitcoin Fullnode + Lightning Network + TOR
>                -------------------------------------------
>                load average: 0.03, 0.06, 0.03, CPU 45°C
>                Free Mem 821M / 927M  Free HDD 631G (68%)
>                ssh admin@192.168.2.142 ▼599.8KiB ▲91.6KiB
>   
>                bitcoin v0.17.0 mainnet Sync  ()
>                Public :
>      
>                LND v0.5-beta -

> ****************************************************************************
> Unlock LND Wallet --> lncli --chain=bitcoin unlock [...]

kangoala, 2018-11-16:
I got the same error messages and partially still have the same problems.

The rpc-error seems to be fixed by adding following entries at bitcoin.conf:
rpcallowip = 127.0.0.1
rpcport = 8332
(see also https://www.reddit.com/r/btc/comments/5jwh0s/bitcoincli_not_connecting_help/)
e.g. by "sudo nano /mnt/hdd/bitcoin/bitcoin.conf"

However LND doesn't seem to start automatically. I tried to start by "sudo systemctl start lnd" what sometimes sporadically might have worked, but most of the time it still shows:

LND v0.5-beta  Not Started | Not Ready Yet

and the LND wallet falls back into locked status after 2-3 minutes.

Has anyone an idea how to start LND?

WetPuppyKisses, 2018-11-17:
@kangoala  try with

  sudo systemctl enable lnd
  sudo systemctl start lnd

kangoala, 2018-11-17:
@WetPuppyKisses 
Thanks, from mainMenu I started the console and entered your commands but anyway didn't received to start lnd. Is there any config-file where to add the commands to start lnd automatically on any restart?

When trying to show channel list, I always get the message "lnd not running". As mentioned before, sometimes I already got a running system. So it was possible to establish a channel to another peer. However, right now I can not get lnd back running anymore. :-( 

Maybe this topic should be handled in separate thread.

dimarkov, 2018-11-21:
I am getting the same error after a fresh install 

`error: Could not connect to the server 127.0.0.1:8332

Make sure the bitcoind server is running and that you are connecting to the correct RPC port.`

Adding
rpcallowip = 127.0.0.1
rpcport = 8332

to bitcoin.conf did not resolve the issue even after a reboot.

kangoala, 2018-11-21:
yes ... that doesn't seem to be the solution. At least in my case I don't get an error message anymore. However there still seem to be a kind of communication error which lets shuting down the lnd deamon quite regularly. I have no idea.

WetPuppyKisses, 2018-11-22:
I could give out my bitcoin.conf file, but the command nano bitcoin.conf doesn't show anything.

WetPuppyKisses, 2018-11-22:
and the lnd conf file

# lnd configuration

[Application Options]
debuglevel=debug
maxpendingchannels=5
alias=Palmi
color=#68F442
nat=false

# RPC open to all connections on Port 10009
rpclisten=0.0.0.0:10009
# Domain, could use https://freedns.afraid.org
#tlsextradomain=lightning.yourhost.com

[Bitcoin]
bitcoin.active=1
bitcoin.node=bitcoind
# enable either testnet or mainnet
#bitcoin.testnet=1
bitcoin.mainnet=1

[Bitcoind]
bitcoind.rpcuser=raspibolt
bitcoind.rpcpass=[Password]
bitcoind.rpchost=127.0.0.1
bitcoind.zmqpubrawblock=tcp://*:28332
bitcoind.zmqpubrawtx=tcp://*:28333

[autopilot]
autopilot.active=0
autopilot.maxchannels=5
autopilot.allocation=0.6

ComradePashka, 2018-12-25:
Hello guys!

I encountered same trouble in the issue https://github.com/rootzoll/raspiblitz/issues/191 but I think it's better to continue here.

After some research, seems that I found exact explanation of the issue with initial bitcoind binding problem. It's well explained here: https://bitcointalk.org/index.php?topic=3179045.msg32927489#msg32927489

The only one difference is that raspiblitz project's file `bitcoind.service` depends on `bootstrap.service` so probably it's better to change line: https://github.com/rootzoll/raspiblitz/blob/ed748e433ff3548da05530d97756143e5b4f5991/home.admin/assets/bootstrap.service#L6
to:

```
After=multi-user.target
```

Going to check this and report soon.
Cheers!

kangoala, 2018-12-27:
I did the change, and after a first reboot everythings works fine. 
Great! That seems to fix the problem.

kangoala, 2018-12-27:
Also a second reboot looks ok. In a first trial, I followed this link: https://www.reddit.com/r/btc/comments/5jwh0s/bitcoincli_not_connecting_help/ and added those two lines to bitcoin.conf. I am not a very experienced user, so I can't say what exactly fixed the problem (hopefully it's fixed).

kangoala, 2018-12-28:
Ok, after several reboots everything worked fine. Then I switched on Autopilot and after a reboot it got stuck several times on starting up bitcoind. When getting back access I realized that another channel was opened. Maybe it was related to that or the Autopilot behaves sometimes a kind of buggy. I switched Autopilot off and in my case it seems to reboot more smoothly now.

rootzoll, 2019-01-10:
It sounds related to this issue - that is on my fix list for verison 1.0 now:
https://github.com/rootzoll/raspiblitz/issues/217

Closing this issue for now.

closed 2019-01-10 by rootzoll