raspiblitz/raspiblitz #167: ifconfig RX/TX bytes wrong

frennkie, 2018-11-05:
Not sure whether I have wrong (e.g. sysctl) setting somewhere on my RaspiBlitz.. but the figures to up/downloaded packets/bytes that `ifconfig` shows are completely wrong. My guess is that there is a counter that is overflowing at resets to 0. Anyone else have this issue?

I have `vnstat` running and those figures make much more sense.

``` 
$: vnstat -i eth0 -s

                      rx      /      tx      /     total    /   estimated
 eth0:
       Oct '18      4.45 GiB  /   18.84 GiB  /   23.29 GiB
       Nov '18      2.03 GiB  /    7.48 GiB  /    9.52 GiB  /   58.82 GiB
     yesterday    403.28 MiB  /    3.17 GiB  /    3.57 GiB
         today    417.13 MiB  /  780.17 MiB  /    1.17 GiB  /    1.31 GiB
``` 

Ifconfig:

``` 
$: ifconfig eth0 | grep "X packets"
        RX packets 45984919  bytes 2813556088 (2.6 GiB)
        TX packets 52741189  bytes 2693327837 (2.5 GiB)
```

rootzoll, 2019-01-15:
I was giving this some thought ... in the end the up/down total traffic is not that useful for monitoring. And the RaspiBlitz would also lose this data with enery update of the sd card.

What can be useful for the future: 

1. If your connection is on a monthly limit - maybe a 1month statistic is useful. 
2. A general info how much traffic bandwidth up&down is at the moment (like with`vnstat -tr` ) to see how much of your local bandwidth resource is used by the RaspiBlitz (and if its maybe the reason why you have problems watching netflix)

@all: Please feel free to add further use cases there network statistics can make sense for you.

I was not able to find a quick output of vnstat giving me the total traffic. So for now I decided to keep the ifconfig data in the infoscreen. It might get replaced in the future by the statistics service anyway soon. But I added vnstat as a service to the SD card build script, so that its available for the future features.

I will remove the issue from the version 1.0 - but it stays open for future enhacement.

frennkie, 2019-01-15:
I plugged in my RaspiBlitz some time on Dec. 31st and this is my output (both ifconfig and vntstat):

```
$: uptime
 19:54:01 up 15 days,  6:24,  5 users,  load average: 0.80, 0.91, 0.88
$: ifconfig eth0 | grep byt
        RX packets 50720396  bytes 1728714298 (1.6 GiB)
        TX packets 49131936  bytes 1682254930 (1.5 GiB)
$: vnstat -i eth0
Database updated: Tue Jan 15 19:52:21 2019

   eth0 since 21/10/18

          rx:  35.01 GiB      tx:  98.34 GiB      total:  133.35 GiB

   monthly
                     rx      |     tx      |    total    |   avg. rate
     ------------------------+-------------+-------------+---------------
       Dec '18      8.22 GiB |   13.21 GiB |   21.43 GiB |   67.13 kbit/s
       Jan '19      9.41 GiB |   17.32 GiB |   26.73 GiB |  175.52 kbit/s
     ------------------------+-------------+-------------+---------------
     estimated     19.73 GiB |   36.31 GiB |   56.04 GiB |

   daily
                     rx      |     tx      |    total    |   avg. rate
     ------------------------+-------------+-------------+---------------
     yesterday    538.76 MiB |  452.28 MiB |  991.04 MiB |   93.97 kbit/s
         today    421.87 MiB |    1.57 GiB |    1.98 GiB |  232.41 kbit/s
     ------------------------+-------------+-------------+---------------
     estimated       508 MiB |    1.90 GiB |    2.39 GiB |

```

So the kernel counters rolled over at something like ~24 GB...  I just was surprised by this.. 

@rootzoll I agree with your approach to consider the actual use cases. I think the total for a given range (current month - for me `26.73 GiB`) really is the most obvious one...

frennkie, 2020-01-07:
I'm not sure for what we are keeping this issue open. As `vnstat` was included in the release image it is now available for users to check it. Maybe we add a nice shortcut for the command (and maybe a reference in the FAQ). But I would not change the 00infoBlitz.sh dashboard to show it.

closed 2020-04-18 by rootzoll