raspiblitz/raspiblitz #170: mv: cannot stat

WetPuppyKisses, 2018-11-09:
Once I donwloaded the blockchain using torrent I get the following:

****************************************************
Monitoring Screen Session: Torrent base+update
If needed press key x to stop TORRENT download
NOTICE: This can take multiple hours or days !!
Its OK to close terminal now and SSH back in later.
****************************************************

*** 1) Status Torrent 'blockchain':
Completed

*** 2) Status Torrent 'update':
Completed

OK - all torrents finished

*** Moving Files ***
1541726779
can take some minutes ... öease wait
mv: cannot stat '/mnt/hdd/torrent/raspiblitz-bitcoin1-2018-10-13-base/*': No such file or directory

and then it doesn't do anything. I don't have a clue of what step I messed up. The instructions are very straight forward.

is there a way to re initiate the setup process without downloading the whole blockchain again?

rootzoll, 2018-11-13:
At this point just restart the pi and it should try to keep up - not much set yet.

To make it contunue: The torrents data needs to be moved into the correct spot - the directory mnt/hdd/bitcoin .. then ist should be able to continue.

But I wonder why the directory is missing. Can you post the output of running: sudo ls /mnt/hdd/torrent/

WetPuppyKisses, 2018-11-14:
@rootzoll Apparently something happened when I downloaded the blockchain via torrent.  The raspberry showed that the download was completed, but the HDD was showing only ~40GB of used disk space.

I formatted the HDD, flashed the SD and proceed to download de blockchain via FTP. Everything is working fine now.

rootzoll, 2018-11-18:
I added some testing for a valid data to the torrent download - so that if such case comes up agian it at least should detect and offer FTP download by itself.

closed 2018-12-11 by rootzoll