raspiblitz/raspiblitz #172: Auto clear logs?

Smiggel, 2018-11-11:
I would like to see an option to enable auto clearing logs.

Today I came across an issue, where my SD card (8GB) was full. It seems like the system dit not clear the available logs by itself and kept it all. I want it to clean it automatically.

I could not set a cronjob, because it's not available for admin, etc.

rootzoll, 2018-11-13:
Which log files exactly, which path? Normally in the RaspiBlitz Setup the bitcoin and lnd logs are on the HDD with much space. 

A cron job is maybe a but much and could delete the files at the wrong time. Would a delete on startup be a start?

If you run the logs on SD card, maybe set log level in config files to somethingelse then debug - this in comibination in deleting the logs on startup should keep them in check.

Smiggel, 2018-11-14:
I noticed I had a lot of logs on the SD card. I assume it had to do with Raspbian then, as the bitcoin logs are on the hdd? I will check it out. Thanks!

rootzoll, 2018-11-29:
I closing the ticket for now - if you can provide more details on growing log files on the SD card please let me know.

closed 2018-11-29 by rootzoll