raspiblitz/raspiblitz #175: lnd not running

kangoala, 2018-11-17:
After unlocking LND wallet I get:

"LND v0.5-beta  Not Started | Not Ready Yet"

Trying to show channel list:

"lnd not running"

I can't get lnd to run. I already tried by command line:
sudo systemctl enable lnd
sudo systemctl start lnd
sudo systemctl restart lnd

So far without success. Afterwards I get asked to unlock the wallet again.

rootzoll, 2018-11-18:
LND only locks again in case of a restart. So it seems that LND is running into some kind of error and gets restarted by systemd. 

Can you check the LND log file?

You should be able to create a DebugReport with:
./XXupdateScripts.sh
./XXdebugLogs.sh

Then copy and paste the report here.

Giksi, 2018-11-18:
Have the same issue. Here is my report: 

*** RASPIBLITZ LOGS ***
blitzversion: 0.95
chainnetwork: bitcoin / main
 15:34:58 up  3:27,  2 users,  load average: 0.33, 0.25, 0.16

*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2018-11-18 15:34:44 CET; 14s ago
  Process: 21972 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 21974 (code=exited, status=1/FAILURE)

Nov 18 15:34:44 Nimbus systemd[1]: bitcoind.service: Unit entered failed state.
Nov 18 15:34:44 Nimbus systemd[1]: bitcoind.service: Failed with result 'ex…de'.
Hint: Some lines were ellipsized, use -l to show in full.

*** LAST 20 CHAINNETWORK LOGS ***
2018-11-18T14:34:43Z scheduler thread start
2018-11-18T14:34:43Z HTTP: creating work queue of depth 16
2018-11-18T14:34:43Z Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcauth for rpcauth auth generation.
2018-11-18T14:34:43Z HTTP: starting 4 worker threads
2018-11-18T14:34:43Z Cache configuration:
2018-11-18T14:34:43Z * Using 2.0MiB for block index database
2018-11-18T14:34:43Z * Using 12.2MiB for transaction index database
2018-11-18T14:34:43Z * Using 8.0MiB for chain state database
2018-11-18T14:34:43Z * Using 77.8MiB for in-memory UTXO set (plus up to 47.7MiB of unused mempool space)
2018-11-18T14:34:43Z init message: Loading block index...
2018-11-18T14:34:43Z Opening LevelDB in /home/bitcoin/.bitcoin/blocks/index
2018-11-18T14:34:44Z Fatal LevelDB error: IO error: /home/bitcoin/.bitcoin/blocks/index/LOCK: Structure needs cleaning
2018-11-18T14:34:44Z You can use -debug=leveldb to get more complete diagnostic messages
2018-11-18T14:34:44Z Fatal LevelDB error: IO error: /home/bitcoin/.bitcoin/blocks/index/LOCK: Structure needs cleaning
2018-11-18T14:34:44Z : Error opening block database.
Please restart with -reindex or -reindex-chainstate to recover.
2018-11-18T14:34:44Z Aborted block database rebuild. Exiting.
2018-11-18T14:34:44Z Shutdown: In progress...
2018-11-18T14:34:44Z scheduler thread interrupt
2018-11-18T14:34:44Z Shutdown: done

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2018-11-18 15:34:43 CET; 15s ago
 Main PID: 21975 (lnd)
   CGroup: /system.slice/lnd.service
           └─21975 /usr/local/bin/lnd --externalip=80.110.72.151

Nov 18 15:34:43 Nimbus lnd[21975]: 2018-11-18 15:34:43.748 [INF] RPCS: pass…8080
Nov 18 15:34:43 Nimbus lnd[21975]: 2018-11-18 15:34:43.749 [INF] LTND: Wait… it.
Hint: Some lines were ellipsized, use -l to show in full.

*** LAST 20 LND LOGS ***
-- Logs begin at Sat 2018-11-17 23:17:01 CET, end at Sun 2018-11-18 15:34:58 CET. --
Nov 17 23:17:31 Nimbus lnd[762]: 2018-11-17 23:17:31.294 [INF] LTND: Version 0.5.0-beta commit=25145acc46cc5d18e4e348eed097300b1391d2a7
Nov 17 23:17:31 Nimbus lnd[762]: 2018-11-17 23:17:31.296 [INF] LTND: Active chain: Bitcoin (network=mainnet)
Nov 17 23:17:31 Nimbus lnd[762]: 2018-11-17 23:17:31.401 [INF] CHDB: Checking for schema update: latest_version=6, db_version=6
Nov 17 23:17:31 Nimbus lnd[762]: 2018-11-17 23:17:31.415 [INF] RPCS: password RPC server listening on [::]:10009
Nov 17 23:17:31 Nimbus lnd[762]: 2018-11-17 23:17:31.416 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
Nov 17 23:17:31 Nimbus lnd[762]: 2018-11-17 23:17:31.417 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
Nov 18 15:33:40 Nimbus lnd[762]: 2018-11-18 15:33:40.148 [INF] LNWL: Opened wallet
Nov 18 15:33:40 Nimbus lnd[762]: 2018-11-18 15:33:40.761 [INF] LTND: Primary chain is set to: bitcoin
Nov 18 15:33:40 Nimbus systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Nov 18 15:33:40 Nimbus systemd[1]: lnd.service: Unit entered failed state.
Nov 18 15:33:40 Nimbus systemd[1]: lnd.service: Failed with result 'exit-code'.
Nov 18 15:34:41 Nimbus systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Nov 18 15:34:41 Nimbus systemd[1]: Stopped LND Lightning Daemon.
Nov 18 15:34:43 Nimbus systemd[1]: Started LND Lightning Daemon.
Nov 18 15:34:43 Nimbus lnd[21975]: 2018-11-18 15:34:43.739 [INF] LTND: Version 0.5.0-beta commit=25145acc46cc5d18e4e348eed097300b1391d2a7
Nov 18 15:34:43 Nimbus lnd[21975]: 2018-11-18 15:34:43.740 [INF] LTND: Active chain: Bitcoin (network=mainnet)
Nov 18 15:34:43 Nimbus lnd[21975]: 2018-11-18 15:34:43.742 [INF] CHDB: Checking for schema update: latest_version=6, db_version=6
Nov 18 15:34:43 Nimbus lnd[21975]: 2018-11-18 15:34:43.748 [INF] RPCS: password RPC server listening on [::]:10009
Nov 18 15:34:43 Nimbus lnd[21975]: 2018-11-18 15:34:43.748 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
Nov 18 15:34:43 Nimbus lnd[21975]: 2018-11-18 15:34:43.749 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.

rootzoll, 2018-11-18:
OK - there is something strange with your bitcoind. It has a "Fatal LevelDB error" and is on exit status.

I can see two options: Try to recover/cleanup the bitcoind data or start again with fresh formatted and HDD and SD card image.

kangoala, 2018-11-21:
@rootzoll 
... can you please take a look at the corresponding log file? Is it an communication issue? Initially I had the same problem as in #164 described. I could fix it by adding some lines in the bitcoin.conf file. But now this problem with the "Stopped LND Daemon" is remaining.

WetPuppyKisses, 2018-11-22:
@kangoala How did you downloaded the blockchain?

When I tried via torrent I had multiple similar issues. FTP worked fine.

Did you get the SD image from:
https://wiki.fulmo.org/downloads/raspiblitz-2018-10-20.img.gz ?

If I were you I would try to start fresh. (Format HDD and SD). 
For me is very hard and frustrating to debug on the Pi

osilloh, 2018-11-22:
Having similar issue. Ran the update script twice and see this:

admin@RASPI:~ $ ./XXupdateScripts.sh

*** UPDATING SHELL SCRIPTS FROM GITHUB ***
justincase, not the final upadte mechanism
******************************************
Already up-to-date.
******************************************
OK - shell scripts and assests are up to date
Reboot recommended

redacted IP and device name. It keeps cycling and asking me for the lncli unlock password.

*** RASPIBLITZ LOGS ***
blitzversion: 0.96
chainnetwork: bitcoin / main
 07:19:20 up 10 min,  2 users,  load average: 0.34, 0.37, 0.22
 
*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-11-22 07:09:26 UTC; 9min ago
  Process: 657 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 659 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─659 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin…id
 
Nov 22 07:09:25 RASPI systemd[1]: Starting Bitcoin daemon...
Nov 22 07:09:26 RASPI systemd[1]: Started Bitcoin daemon.
 
*** LAST 20 CHAINNETWORK LOGS ***
2018-11-22T07:11:08Z Bound to 0.0.0.0:8333
2018-11-22T07:11:08Z init message: Loading P2P addresses...
2018-11-22T07:11:08Z torcontrol thread start
2018-11-22T07:11:08Z Imported mempool transactions from disk: 0 succeeded, 0 failed, 0 expired, 0 already there
2018-11-22T07:11:08Z Loaded 1088 addresses from peers.dat  57ms
2018-11-22T07:11:08Z init message: Loading banlist...
2018-11-22T07:11:08Z init message: Starting network threads...
2018-11-22T07:11:08Z net thread start
2018-11-22T07:11:08Z dnsseed thread start
2018-11-22T07:11:08Z addcon thread start
2018-11-22T07:11:08Z msghand thread start
2018-11-22T07:11:08Z opencon thread start
2018-11-22T07:11:08Z init message: Done loading
2018-11-22T07:11:19Z Loading addresses from DNS seeds (could take a while)
2018-11-22T07:11:28Z 233 addresses found from DNS seeds
2018-11-22T07:11:28Z dnsseed thread exit
2018-11-22T07:12:54Z ThreadRPCServer incorrect password attempt from 127.0.0.1:57226
2018-11-22T07:14:55Z ThreadRPCServer incorrect password attempt from 127.0.0.1:57474
2018-11-22T07:16:20Z ThreadRPCServer incorrect password attempt from 127.0.0.1:57674
2018-11-22T07:18:09Z ThreadRPCServer incorrect password attempt from 127.0.0.1:57910
 
*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-11-22 07:19:10 UTC; 10s ago
 Main PID: 8917 (lnd)
   CGroup: /system.slice/lnd.service
           └─8917 /usr/local/bin/lnd --externalip=xxx.xxx.xxx.xxx
 
Nov 22 07:19:10 RASPI lnd[8917]: 2018-11-22 07:19:10.586 [INF] RPCS: password gRPC proxy started ….1:8080
Nov 22 07:19:10 RASPI lnd[8917]: 2018-11-22 07:19:10.586 [INF] LTND: Waiting for wallet encryptio…ock it.
Hint: Some lines were ellipsized, use -l to show in full.
 
*** LAST 20 LND LOGS ***
-- Logs begin at Thu 2018-11-22 07:09:14 UTC, end at Thu 2018-11-22 07:19:20 UTC. --
Nov 22 07:17:20 RASPI lnd[7298]: 2018-11-22 07:17:20.613 [INF] RPCS: password RPC server listening on [::]:10009
Nov 22 07:17:20 RASPI lnd[7298]: 2018-11-22 07:17:20.615 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
Nov 22 07:17:20 RASPI lnd[7298]: 2018-11-22 07:17:20.615 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
Nov 22 07:18:08 RASPI lnd[7298]: 2018-11-22 07:18:08.805 [INF] LNWL: Opened wallet
Nov 22 07:18:09 RASPI lnd[7298]: 2018-11-22 07:18:09.851 [INF] LTND: Primary chain is set to: bitcoin
Nov 22 07:18:10 RASPI lnd[7298]: unable to create chain control: unable to connect to bitcoind: status code: 401, response: ""
Nov 22 07:18:10 RASPI lnd[7298]: 2018-11-22 07:18:10.127 [INF] LTND: Shutdown complete
Nov 22 07:18:10 RASPI lnd[7298]: unable to connect to bitcoind: status code: 401, response: ""
Nov 22 07:18:10 RASPI systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Nov 22 07:18:10 RASPI systemd[1]: lnd.service: Unit entered failed state.
Nov 22 07:18:10 RASPI systemd[1]: lnd.service: Failed with result 'exit-code'.
Nov 22 07:19:10 RASPI systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Nov 22 07:19:10 RASPI systemd[1]: Stopped LND Lightning Daemon.
Nov 22 07:19:10 RASPI systemd[1]: Started LND Lightning Daemon.
Nov 22 07:19:10 RASPI lnd[8917]: 2018-11-22 07:19:10.579 [INF] LTND: Version: 0.5.0-beta commit=v0.5-beta-101-g61e867741926bcb318432a6344b80161fabd1455, build=production, logging=default
Nov 22 07:19:10 RASPI lnd[8917]: 2018-11-22 07:19:10.580 [INF] LTND: Active chain: Bitcoin (network=mainnet)
Nov 22 07:19:10 RASPI lnd[8917]: 2018-11-22 07:19:10.581 [INF] CHDB: Checking for schema update: latest_version=6, db_version=6
Nov 22 07:19:10 RASPI lnd[8917]: 2018-11-22 07:19:10.586 [INF] RPCS: password RPC server listening on [::]:10009
Nov 22 07:19:10 RASPI lnd[8917]: 2018-11-22 07:19:10.586 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
Nov 22 07:19:10 RASPI lnd[8917]: 2018-11-22 07:19:10.586 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.

rootzoll, 2018-11-26:
@kangoala lets start with the question about "Recover Coins". I wrote a bleeding fresh chapter about that in the README - checkout here: https://github.com/rootzoll/raspiblitz/blob/master/README.md#recover-your-coins-from-a-failing-raspiblitz

rootzoll, 2018-11-26:
@kangoala 

About the LND not starting correctly: The error seems to be "ReadVarBytes: sig is larger than the max allowed size" - the only think I could find on that was an issue on the LND github:
https://github.com/lightningnetwork/lnd/issues/2016

They closed the issue because there was not enough feedback from the node operator. Maybe ask your problem there again. 

The alternative way would be: try to recover your old coins with cypher seed and try the next fresh sd-card release. Sorry for the trouble.

kangoala, 2018-11-28:
Thanks, for your update on coin recovery.

> The alternative way would be: try to recover your old coins with cypher seed and try the next fresh sd-card release. Sorry for the trouble.

No worries ;-)
I am aware of the fact that lightning is still in early development phase (and probably will be for quite some time). Will it become less risky to loose channel funds once functioning "Watchtowers" have been implemented?

If lnd is not running anymore, or if I shut down the raspi blitz and are waiting for the next sd-card release, can someone meanwhile steal my coins from my lnd-channels?

If I restart lnd after a longer period of time, could this also cause to trigger the lightning "penalty" mechanism, or will it just try to synchronise the channel state?

ln3r, 2018-11-29:
I'm having a similar issue, which seems to stem from bitcoind.

I had a fully sync'd Raspiblitz with a couple of open channels and I wanted to improve connectivity by enabling autopilot. I had also experimented with Zap iOS on my RaspiBlitz earlier. After doing the required reboot, bitcoind failed to start up. It got stuck on "Waiting for bitcoind to get ready. ---> Starting Up  Can take longer if device was off"

My first thought was that a large block was coming through, but it has been ~30 hours and the very same screen is still showing. Immediately after I ssh into it, the following screen pops up and I can't access the CLI so no access to logs apparently.

[image]

I'm inclined to start fresh with a new SD + HDD as suggested by @rootzoll, but since more people are reporting this issue I wanted to give my 2 cents. Let me know if you need additional information. Still love the project!

vnnkl, 2018-11-29:
you can cancel out of that screen with ctrl+c

do you have `XXdebugLogs.sh` in your admin users home directory?

otherwise try printing your latest logs first, before starting freshly over.

ln3r, 2018-11-29:
Thank you @vnnkl, got the logs and it looks like I'm fully sync'd:

*** RASPIBLITZ LOGS ***
blitzversion: 0.97
chainnetwork: bitcoin / main
 15:45:44 up 36 min,  2 users,  load average: 0.56, 0.44, 1.01

*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-11-29 14:52:00 UTC; 53min ago
  Process: 679 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 680 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─680 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

Nov 29 14:51:59 RetroBlitz systemd[1]: Starting Bitcoin daemon...
Nov 29 14:52:00 RetroBlitz systemd[1]: Started Bitcoin daemon.

*** LAST 20 CHAINNETWORK LOGS ***
2018-11-29T15:28:50Z UpdateTip: new best=00000000000000000014d9723763790c3259c66970a73d65e42944b5786ce5e5 height=551927 version=0x20000000 log2_work=90.074479 tx=360665697 date='2018-11-29T13:26:21Z' progress=0.999951 cache=36.6MiB(326178txo)
2018-11-29T15:29:03Z UpdateTip: new best=0000000000000000001e36fb7acd1101f2cc74f5dfbff8d6c1d3f4466fcb79b1 height=551928 version=0x20000000 log2_work=90.07451 tx=360668025 date='2018-11-29T14:07:40Z' progress=0.999968 cache=37.4MiB(333669txo)
2018-11-29T15:29:14Z UpdateTip: new best=000000000000000000153db7abf62fe7ef1deef04bba015ccac285ede9797245 height=551929 version=0x20000000 log2_work=90.074542 tx=360669429 date='2018-11-29T14:12:16Z' progress=0.999969 cache=37.8MiB(337433txo)
2018-11-29T15:29:14Z Leaving InitialBlockDownload (latching to false)
2018-11-29T15:29:14Z New outbound peer connected: version: 70015, blocks=551934, peer=13
2018-11-29T15:29:36Z UpdateTip: new best=0000000000000000001574cc1a5abe09521d1e65234b327b45978ed96befaa25 height=551930 version=0x20c00000 log2_work=90.074574 tx=360670618 date='2018-11-29T14:21:20Z' progress=0.999973 cache=38.2MiB(341104txo) warning='28 of last 100 blocks have unexpected version'
2018-11-29T15:29:51Z UpdateTip: new best=00000000000000000005053309d83308315842aaa9a1da1aaeb4a4d1d61c08df height=551931 version=0x2000e000 log2_work=90.074605 tx=360672888 date='2018-11-29T14:34:36Z' progress=0.999978 cache=38.6MiB(344758txo) warning='29 of last 100 blocks have unexpected version'
2018-11-29T15:30:08Z UpdateTip: new best=00000000000000000005b26f4b0a5b0955b32279328715a1934cae21cf99222f height=551932 version=0x20000000 log2_work=90.074637 tx=360675182 date='2018-11-29T14:50:44Z' progress=0.999984 cache=38.9MiB(348142txo) warning='28 of last 100 blocks have unexpected version'
2018-11-29T15:30:09Z New outbound peer connected: version: 70015, blocks=551934, peer=14
2018-11-29T15:30:23Z UpdateTip: new best=000000000000000000243f8b674b5dcb3c98a0b6986e04e79292a76283d4635e height=551933 version=0x20c00000 log2_work=90.074669 tx=360677965 date='2018-11-29T15:24:54Z' progress=0.999998 cache=39.6MiB(354166txo) warning='29 of last 100 blocks have unexpected version'
2018-11-29T15:30:36Z Pre-allocating up to position 0x200000 in rev01447.dat
2018-11-29T15:30:36Z UpdateTip: new best=00000000000000000016945351b1a1d6716c09b56487fdb8172d98af775b68d7 height=551934 version=0x20000000 log2_work=90.0747 tx=360679617 date='2018-11-29T15:27:20Z' progress=0.999999 cache=39.4MiB(352484txo) warning='29 of last 100 blocks have unexpected version'
2018-11-29T15:30:38Z New outbound peer connected: version: 70015, blocks=551934, peer=15
2018-11-29T15:30:38Z New outbound peer connected: version: 70015, blocks=551934, peer=16
2018-11-29T15:30:41Z New outbound peer connected: version: 70015, blocks=551934, peer=18
2018-11-29T15:30:46Z New outbound peer connected: version: 70015, blocks=551934, peer=19
2018-11-29T15:30:47Z New outbound peer connected: version: 70015, blocks=551934, peer=20
2018-11-29T15:30:47Z New outbound peer connected: version: 70015, blocks=551934, peer=21
2018-11-29T15:32:40Z Imported mempool transactions from disk: 7152 succeeded, 12887 failed, 0 expired, 0 already there
2018-11-29T15:39:17Z UpdateTip: new best=0000000000000000001c60b22763eff726cb5285670c637f8f6a58d72fa22055 height=551935 version=0x20000000 log2_work=90.074732 tx=360682736 date='2018-11-29T15:39:14Z' progress=1.000000 cache=41.1MiB(368477txo) warning='28 of last 100 blocks have unexpected version'

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-11-29 14:52:00 UTC; 53min ago
 Main PID: 681 (lnd)
   CGroup: /system.slice/lnd.service
           └─681 /usr/local/bin/lnd --externalip=

Nov 29 14:52:02 RetroBlitz lnd[681]: 2018-11-29 14:52:02.340 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
Nov 29 14:52:02 RetroBlitz lnd[681]: 2018-11-29 14:52:02.342 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to …unlock it.
Hint: Some lines were ellipsized, use -l to show in full.

*** LAST 20 LND LOGS ***
-- Logs begin at Thu 2018-11-29 14:17:01 UTC, end at Thu 2018-11-29 15:45:44 UTC. --
Nov 29 14:52:00 RetroBlitz systemd[1]: Started LND Lightning Daemon.
Nov 29 14:52:02 RetroBlitz lnd[681]: 2018-11-29 14:52:02.219 [INF] LTND: Version: 0.5.0-beta commit=v0.5-beta-101-g61e867741926bcb318432a6344b80161fabd1455, build=production, logging=default
[... cut ...]
Nov 29 14:52:02 RetroBlitz lnd[681]: 2018-11-29 14:52:02.339 [INF] RPCS: password RPC server listening on [::]:10009
Nov 29 14:52:02 RetroBlitz lnd[681]: 2018-11-29 14:52:02.340 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
Nov 29 14:52:02 RetroBlitz lnd[681]: 2018-11-29 14:52:02.342 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.

vnnkl, 2018-11-29:
yes, you look sync, can you `lncli unlock` successfully? 

What happens if you use `./00mainMenu.sh`, can you select the info screen then?

ln3r, 2018-11-29:
> yes, you look sync, can you `lncli unlock` successfully?

yep, I can unlock it successfully.

> 
> What happens if you use `./00mainMenu.sh`, can you select the info screen then?

I'm brought back to the "Waiting for bitcoind to get ready." No selections available.

ln3r, 2018-11-29:
Not sure if it helps, but if I run `lncli closeallchannels` it returns:

`[lncli] unable to fetch open channels: rpc error: code = Unavailable desc = grpc: the connection is unavailable`

Maybe it's a problem with rpc?

rootzoll, 2018-11-29:
I just setuped a new Litecoin-Node with FTP downloaded - it also gets stuck in the "StartingUp" screen - litecoind is saying: Please restart with -reindex or -reindex-chainstate to recover.

TODO for Milestone Version 1.0:
So it seems I have to update the Litecoin FTP and improve the "StartingUP Screen" to detect such errors better - and not just getting stuck.

About the other problems - i will start to look into this over the next days. All inputs welcome.

fluidvoice, 2018-12-23:
> @kangoala
> 
> About the LND not starting correctly: The error seems to be "ReadVarBytes: sig is larger than the max allowed size" - the only think I could find on that was an issue on the LND github:
> [lightningnetwork/lnd#2016](https://github.com/lightningnetwork/lnd/issues/2016)
> 
> They closed the issue because there was not enough feedback from the node operator. Maybe ask your problem there again.
> 
> The alternative way would be: try to recover your old coins with cypher seed and try the next fresh sd-card release. Sorry for the trouble.

I'm getting the same error ... "ReadVarBytes: sig is larger than the max allowed size"
I didn't see any resolution to this problem. Help?

$ ./XXdebugLogs.sh 
./XXdebugLogs.sh: line 9: /mnt/hdd/raspiblitz.conf: No such file or directory
backup info: network
ls: cannot access '/mnt/hdd/litecoin/litecoin.conf': No such file or directory
backup info: chain

*** RASPIBLITZ LOGS ***
blitzversion: 0.97
chainnetwork: bitcoin / main
 18:01:29 up 1 day,  3:20,  3 users,  load average: 0.18, 0.17, 0.24

*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2018-12-21 13:45:55 PST; 1 day 4h ago
  Process: 813 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 814 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─814 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home…id

Dec 21 13:45:55 thundahub systemd[1]: Starting Bitcoin daemon...
Dec 21 13:45:55 thundahub systemd[1]: Started Bitcoin daemon.

*** LAST 20 CHAINNETWORK LOGS ***
2018-12-23T01:03:23Z ThreadRPCServer incorrect password attempt from 127.0.0.1:50084
2018-12-23T01:04:23Z ThreadRPCServer incorrect password attempt from 127.0.0.1:50088
2018-12-23T01:05:24Z ThreadRPCServer incorrect password attempt from 127.0.0.1:50092
2018-12-23T01:06:24Z ThreadRPCServer incorrect password attempt from 127.0.0.1:50096
2018-12-23T01:07:25Z ThreadRPCServer incorrect password attempt from 127.0.0.1:50100
2018-12-23T01:08:25Z ThreadRPCServer incorrect password attempt from 127.0.0.1:50102
2018-12-23T01:08:50Z socket recv error Connection reset by peer (104)
2018-12-23T01:09:25Z ThreadRPCServer incorrect password attempt from 127.0.0.1:50106
2018-12-23T01:10:26Z ThreadRPCServer incorrect password attempt from 127.0.0.1:50110
2018-12-23T01:11:26Z ThreadRPCServer incorrect password attempt from 127.0.0.1:50116
2018-12-23T01:12:26Z ThreadRPCServer incorrect password attempt from 127.0.0.1:50120
2018-12-23T01:13:27Z ThreadRPCServer incorrect password attempt from 127.0.0.1:50126
2018-12-23T01:23:10Z ThreadRPCServer incorrect password attempt from 127.0.0.1:50140
2018-12-23T01:23:29Z ThreadRPCServer incorrect password attempt from 127.0.0.1:50156
2018-12-23T01:23:39Z ThreadRPCServer incorrect password attempt from 127.0.0.1:50182
2018-12-23T01:26:43Z ThreadRPCServer incorrect password attempt from 127.0.0.1:50206
2018-12-23T01:35:01Z Potential stale tip detected, will try using extra outbound peer (last tip update: 1911 seconds ago)
2018-12-23T01:35:02Z New outbound peer connected: version: 70015, blocks=555012, peer=2130
2018-12-23T01:37:41Z UpdateTip: new best=00000000000000000030dfec201a6b63e5dd04f8b7c549c829cfda915d8f2e70 height=555013 version=0x20000000 log2_work=90.155592 tx=366783814 date='2018-12-23T01:37:14Z' progress=1.000000 cache=16.5MiB(122335txo) warning='29 of last 100 blocks have unexpected version'
2018-12-23T01:55:46Z socket recv error Connection reset by peer (104)

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2018-12-22 17:38:17 PST; 23min ago
 Main PID: 26508 (lnd)
   CGroup: /system.slice/lnd.service
           └─26508 /usr/local/bin/lnd --externalip=

Dec 22 17:38:17 thundahub lnd[26508]: 2018-12-22 17:38:17.990 [INF] RPCS: password RPC server lis…:10009
Dec 22 17:38:17 thundahub lnd[26508]: 2018-12-22 17:38:17.992 [INF] LTND: Waiting for wallet encr…ck it.
Hint: Some lines were ellipsized, use -l to show in full.

*** LAST 20 LND LOGS ***
-- Logs begin at Fri 2018-12-21 13:17:01 PST, end at Sat 2018-12-22 18:01:29 PST. --
Dec 22 17:37:17 thundahub lnd[26222]: 2018-12-22 17:37:17.213 [INF] LTND: LightningWallet opened
Dec 22 17:37:17 thundahub lnd[26222]: 2018-12-22 17:37:17.213 [DBG] LNWL: Birthday block has already been verified: height=553299, hash=0000000000000000002fb5012be16342623a11878216e6b9fc2b75cf4f702ece
Dec 22 17:37:17 thundahub lnd[26222]: 2018-12-22 17:37:17.331 [INF] HSWC: Restoring in-memory circuit state from disk
Dec 22 17:37:17 thundahub lnd[26222]: 2018-12-22 17:37:17.363 [INF] LNWL: Started rescan from block 0000000000000000002a719dceb1fed5a8e77eb98c52a846f980ec33effd84d5 (height 554839) for 9 addresses
Dec 22 17:37:17 thundahub lnd[26222]: 2018-12-22 17:37:17.365 [INF] LNWL: Starting rescan from block 0000000000000000002a719dceb1fed5a8e77eb98c52a846f980ec33effd84d5
Dec 22 17:37:17 thundahub lnd[26222]: 2018-12-22 17:37:17.389 [INF] HSWC: Payment circuits loaded: num_pending=0, num_open=0
Dec 22 17:37:17 thundahub lnd[26222]: 2018-12-22 17:37:17.401 [INF] HSWC: Trimming open circuits for chan_id=553852:1252:1, start_htlc_id=0
[... cut ...]
Dec 22 17:38:17 thundahub lnd[26508]: 2018-12-22 17:38:17.990 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
Dec 22 17:38:17 thundahub lnd[26508]: 2018-12-22 17:38:17.990 [INF] RPCS: password RPC server listening on [::]:10009
Dec 22 17:38:17 thundahub lnd[26508]: 2018-12-22 17:38:17.992 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.

fluidvoice, 2018-12-23:
For me what it does when I SSH in is it brings up a menu "Choose how you want to setup your Raspiblitz" with options Bitcoin or Litecoin. Upon selecting Bitcoin is sees that LND is running and a wallet exists then prompts me to unlock the wallet. Upon entering the password it says the wallet was successfully unlocked but then I get the error I mentioned above and the SSH terminal shows...
"error(error: Authorization failed: Incorrect rpcuser or rpcpassword)"

fluidvoice, 2018-12-23:
It looks like it's not using the correct sudo user "bitcoin" because it works if I explicitly use user "bitcoin" but fails with the same error if I use user "admin"...

admin@thundahub:~ $ sudo -u admin bitcoin-cli getblockchaininfo
**error: Authorization failed: Incorrect rpcuser or rpcpassword**
admin@thundahub:~ $ sudo -u bitcoin bitcoin-cli getblockchaininfo
{
  "chain": "main",
  "blocks": 555136,
  "headers": 555136,

fluidvoice, 2018-12-23:
Now whenever I unlock my lnd wallet I get this...

Dec 23 14:05:42 thundahub lnd[19573]: 2018-12-23 14:05:42.178 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
Dec 23 14:06:49 thundahub lnd[19573]: 2018-12-23 14:06:49.775 [INF] LNWL: Opened wallet
Dec 23 14:06:50 thundahub lnd[19573]: 2018-12-23 14:06:50.237 [INF] LTND: Primary chain is set to: bitcoin
Dec 23 14:06:50 thundahub lnd[19573]: unable to create chain control: unable to connect to bitcoind: Post http://127.0.0.1:8332: dial tcp 127.0.0.1:8332: connect: connection refused
Dec 23 14:06:50 thundahub lnd[19573]: 2018-12-23 14:06:50.287 [INF] LTND: Shutdown complete
Dec 23 14:06:50 thundahub systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Dec 23 14:06:50 thundahub systemd[1]: lnd.service: Unit entered failed state.
Dec 23 14:06:50 thundahub systemd[1]: lnd.service: Failed with result 'exit-code'.

fluidvoice, 2018-12-23:
OK some progress finally. After I ran **./AAunlockLND.sh** to unlock my wallet I then ran **./00infoBlitz.sh** and it showed as per below all was running.  But I still have two problems 1) it's not starting up correctly by default when rebooting and 2) the LCD is not updating with this same information.

RaspiBlitz v0.97  raspiblitz
               bitcoin Fullnode + Lightning Network + TOR
        ,/     -------------------------------------------
      ,'/      load average: 3.50, 1.36, 0.60, CPU 69°C
    ,' /       Free Mem 551M / 927M  Free HDD 179G (39%)
  ,'  /_____,  ssh admin@192.168.2.100 ▼0.0B ▲0.0B
 .'____    ,'  
      /  ,'    bitcoin v0.17.0.1 mainnet Sync OK (100.00%)
     / ,'      Public ####:###:####:####:XXXX:XXXX:XXXX:XXXX:8333  14 connections
    /,'        
   /'          LND 0.5.1-beta wallet 49589 sat 
               1/1 Channels 98243 sat 3 peers

kangoala, 2018-12-27:
In my case, it might have been an rpc-issue.
I fixed an entry in bootstrap.service and now I works fine and LND doesn't fall back into locked status. For more details see this comment: https://github.com/rootzoll/raspiblitz/issues/164#issuecomment-449841424

kangoala, 2018-12-28:
For me the problem is fixed. After changing an entry in bootstrap.service LND runs smoothly and doesn't get stopped anymore.

 After=network.target      -->    After=multi-user.target

I had a short problem with starting up bitcoind when switching on Autopilot. Nevertheless it works much more reliable than before. Many thanks to ComradePashka.

fluidvoice, 2018-12-29:
> For me the problem is fixed. After changing an entry in bootstrap.service LND runs smoothly and doesn't get stopped anymore.
>
> After=network.target --> After=multi-user.target
>
> I had a short problem [...]

This change made it worse for me. With this change, after booting up my **bitcoind** and **lnd** services were not running.

kangoala, 2018-12-29:
It's an additional Service (Activate / Deactivate Services) which establish automatically channels to other peers (Channel Autopilot). By default it should be switched off. So if you havn't switched on, then it should be ok. You can check this in the main menu.

fluidvoice, 2018-12-29:
@kangoala check out the problem I have.  Does it make any sense to you?
I'm wondering why everyone else doesn't have this problem?
https://github.com/rootzoll/raspiblitz/issues/224#issue-394821331

theotherside, 2018-12-29:
> @kangoala check out the problem I have. Does it make any sense to you?
> I'm wondering why everyone else doesn't have this problem?
> [#224 (comment)](https://github.com/rootzoll/raspiblitz/issues/224#issue-394821331)

I've actually had that problem over that last few versions. Anytime it got restarted.

kangoala, 2018-12-31:
@fluidvoice 
I can only forward the message from rootzoll. 
He wrote a chapter about coin recovery - checkout here: https://github.com/rootzoll/raspiblitz/blob/master/README.md#recover-your-coins-from-a-failing-raspiblitz
Maybe starting from scratch with a new installation would be most promissing.
I did several restarts during the last days. Mostly it works fine. However, sometimes it still might have some problems with chain synchronization. It still can happen that bitcoind or lnd is not starting up properly. In 8 out of 10 restarts it seems to work. Maybe also try to give some time between two restarts. E.g., switch off and wait for some hours before doing a restart. There's no guaranty, probably your problem is caused by a different issue. I have no idea. I just followed rootzoll's guide on how to set up the raspi blitz. Unfortunately I'm a totally unexperienced user, but I like the idea of second layer solutions.
Hopefully in 2019 everything gets better ;-)

fluidvoice, 2018-12-31:
I thought about doing that as a last resort. Don't wanna risk losing my coins and existing channel, though it's a small amount of money. Sucks there is no other re-install tutorial.

rootzoll, 2019-01-09:
OK - after taking care about a lot of other things. I try to work thru this long thread now ... are there any changes you recommend from your debugging?

fluidvoice, 2019-01-09:
> OK - after taking care about a lot of other things. I try to work thru this long thread now ... are there any changes you recommend from your debugging?

In my case, no. My problems in this issue/thread was resolved when I updated to the latest scripts.

kangoala, 2019-01-09:
For me it looks also ok. 
After changing an entry in bootstrap.service LND seems to run much more stable.
(After=network.target --> After=multi-user.target)
I'm not quite sure if autopilot somehow can cause problems after reboot. But right now I'm happy and don't use it anymore.
Anyway, thanks for your support :-)

rootzoll, 2019-01-09:
@kangoala that change for bootstrap.service is already merged :) thx
 Also I added more debug info in case it hangs on "Starting up".

So I think I at least covered the most take aways to this thread and will close issue for now.

closed 2019-01-09 by rootzoll