raspiblitz/raspiblitz #183: Update not possible, because not possible to remove balance

jagottsicher, 2018-11-29:
> close all open lightning channels you have (lncli closeallchannels --force)
> wait until all closing transactions are done
> move all on-chain funds to a wallet outside raspiblitz (lncli --conf_target 3 sendcoins [ADDRESS])

unfortunately it seems not possible ro remove me walletbalance, because Of this: wrong usage of the task.

I tried to google this without success, because it looks like everybody is focussing on keeping balance on the lightning network. (makes sense, but here we want to send on-chain to cold storage.

How to move balance properly on chain?

Is it needed that the destination address is the same the node was funded with?

How to calculate --conf-target (sat/byte?) . needed to be deducted from balance? And if yes, how much bytes will the transaction have? 

Some enlighting information is urgently needed - especially, because version 1.0 is knocking at the door! :-)

rootzoll, 2018-11-29:
I moved funds out on my test node some month ago - when i was writing those instructions.

At which point in the process do you have a problem? All your channels are closed and the funds are displayed as available on your on-chain wallet (not delayed in a force close time-lock)? Can you post the command and the error message in detail?

BTW - it is bitcoin or litecoin?

> Is it needed that the destination address is the same the node was funded with?

No - your lnd on-chain wallet should be able to send to any other valid address.

> How to calculate --conf-target (sat/byte?) . needed to be deducted from balance? And if yes, how much bytes will the transaction have?

Simply use  "--conf_target 3" parameter API doc says: "the number of blocks that the transaction *should* confirm in, will be used for fee estimation" .. so the fee should be calculated by lnd.

TODO: I need to re-test this also before version 1.0 release ... so I will put it in my milestone list. Please share any further results you get - bad or good.

rootzoll, 2018-12-05:
OK. This was a bit more complex, but I think the raspiBlitz has now a practical solution (next sd card release). Fact is when calling the sendcoins the amount needs to be less then available funds to keep room for the fees. But we dont know how hight the fees will be. If we want to CASHOUT all funds from the device its a hard task to get this right - because a "sendall" paramter is still missing in the actual lnd release. 

So what is CASHOUT script of the blitz is doing now is, it tries multiple times - starting with almost the full amount and then reduces step by step until lnd is able to create the cashout transaction. It may be that some satohis would stay behind (below 1000) - but in my tests I was able to get all funds out that way. Closing this ticket for now.

closed 2018-12-05 by rootzoll