raspiblitz/raspiblitz #185: I'm still stuck:

eurtnl, 2018-11-29:
> *** Test if chainnetwork is ready ...
> 1543400259
> result()
> error(error: Could not connect to the server 127.0.0.1:8332
> 
> Make sure the bitcoind server is running and that you are connecting to the correct RPC port.)
> 2018-11-28T09:51:12Z Syncing txindex with block chain from height 551729
> 2018-11-28T09:51:28Z txindex is enabled at height 551810
> 2018-11-28T09:51:28Z txindex thread exit
> 2018-11-28T09:56:23Z New outbound peer connected: version: 70015, blocks=551810, peer=65
> 2018-11-28T10:16:34Z UpdateTip: new best=0000000000000000000356f3ae64b14b601505199e1afd6921b750ce58dd8f61 height=551811 version=0x20000000 log2_work=90.070805 tx=360392682 date='2018-11-28T10:16:32Z' progress=1.000000 cache=107.8MiB(942756txo) warning='20 of last 100 blocks have unexpected version'
> Waiting 1 minute and then trying again ...

_Originally posted by @eurtnl in https://github.com/rootzoll/raspiblitz/issues/180#issuecomment-442395049_

eurtnl, 2018-11-29:
The output seems to be the same over and over:

> *** Test if chainnetwork is ready ...
> 1543516757
> result()
> error(error: Could not connect to the server 127.0.0.1:8332
> 
> Make sure the bitcoind server is running and that you are connecting to the correct RPC port.)
> 2018-11-29T18:15:30Z UpdateTip: new best=00000000000000000025121554861ca5cd59825dde9ca5925694e14e6f24b61d height=551951 version=0x20000000 log2_work=90.075238 tx=360720657 date='2018-11-29T18:15:23Z' progress=1.000000 cache=55.4MiB(485474txo) warning='26 of last 100 blocks have unexpected version'
> 2018-11-29T18:17:11Z UpdateTip: new best=0000000000000000000498f9ce7cf5d3a47335b9b14ef0a90938931c6d45df63 height=551952 version=0x20000000 log2_work=90.075269 tx=360723057 date='2018-11-29T18:17:01Z' progress=1.000000 cache=55.6MiB(487420txo) warning='26 of last 100 blocks have unexpected version'
> 2018-11-29T18:22:58Z UpdateTip: new best=0000000000000000000d9c7d639272172c8a57c46ee4d0322db1deb335eef4df height=551953 version=0x20000000 log2_work=90.075301 tx=360725218 date='2018-11-29T18:22:52Z' progress=1.000000 cache=55.9MiB(490318txo) warning='26 of last 100 blocks have unexpected version'
> 2018-11-29T18:24:20Z UpdateTip: new best=00000000000000000015096fa50f69502a40baf445602cbbc81d57025febef01 height=551954 version=0x3fffe000 log2_work=90.075332 tx=360726797 date='2018-11-29T18:24:33Z' progress=1.000000 cache=56.1MiB(491488txo) warning='27 of last 100 blocks have unexpected version'
> 2018-11-29T18:32:04Z UpdateTip: new best=0000000000000000000c700dc6fc74bc1330867ed748a97379daab8ef4def4cf height=551955 version=0x20000000 log2_work=90.075364 tx=360729147 date='2018-11-29T18:31:47Z' progress=1.000000 cache=56.3MiB(494181txo) warning='26 of last 100 blocks have unexpected version'
> Waiting 1 minute and then trying again ...
>

rootzoll, 2018-11-29:
Can you open a second terminal window and SSH in with
"ssh bitcoin@[YOURIP]" (PASSWORDA)

Then run "bitcoin-cli getblockchaininfo" and let me know what you get in return?

eurtnl, 2018-11-30:
it says:

> error: Could not connect to the server 127.0.0.1:8332
> 
> Make sure the bitcoind server is running and that you are connecting to the correct RPC port.

rootzoll, 2018-12-05:
The IP 127.0.0.1 is correct - thats localhost communication.

Lets confirm that bitcoind is really running - please let me know the result of the following command:
`sudo systemctl status bitcoind -n2 --no-pager`

eurtnl, 2018-12-05:
Ok. In the meantime i created another sd using 
https://github.com/rootzoll/raspiblitz#build-the-sd-card-image but that got stuck when i connected the synced blockchain which i did before. So what i just did was booting again with the first sd (the one we're talking about) and it somehow finished the setup. I didn't changed a thing...
Here's the result you asked for:

>   bitcoind.service - Bitcoin daemon
>    Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
>    Active: active (running) since Wed 2018-12-05 18:15:15 UTC; 11min ago
>   Process: 700 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
>  Main PID: 705 (bitcoind)
>    CGroup: /system.slice/bitcoind.service
>            └─705 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.…id
> 
> Dec 05 18:15:14 Eurtnl systemd[1]: Starting Bitcoin daemon...
> Dec 05 18:15:15 Eurtnl systemd[1]: Started Bitcoin daemon.

eurtnl, 2018-12-05:
When reaching the nice menu i tried the info status and it says:               

> RaspiBlitz v0.95  
>                bitcoin Fullnode + Lightning Network + TOR
>         ,/     -------------------------------------------
>       ,'/      load average: 3.51, 2.89, 1.56, CPU 62°C
>     ,' /       Free Mem 626M / 927M  Free HDD 617G (67%)
>   ,'  /_____,  ssh admin@192.168.178.60 ▼17.7MiB ▲447.7KiB
>  .'____    ,'  
>       /  ,'    bitcoin v0.17.0 mainnet Sync  ()
>      / ,'      Public : 
>     /,'        
>    /'          LND v0.5-beta  Not Started | Not Ready Yet

When prompted to unlock the LND wallet i entered the pw and i got a succes message, saying its unlocked now. How long will it take for LND to get ready?

eurtnl, 2018-12-05:
ok this is what happens when i started the main menu again (./00mainMenu.sh): it says !!! YOUR WALLET IS LOCKED !!! so i press [ok] to unlock it (again) and along the input for password it says

> error: Could not connect to the server 127.0.0.1:8332
> Make sure the bitcoind server is running and that you are connecting to the correct RPC port.

and after entering the password it says:
> tail: cannot open 'mnt/hdd/lnd/logs/bitcoin/net/lnd.log' for reading: no such file or directory

and it opens the main menu

rootzoll, 2018-12-11:
the last line shows that the information if you are on mainnet or on testnet was not able to be determined by the script and it build the path wrong. At the moment I am finishing up the version 1.0-rc1 release I changed a lot about the managing such state information. Hope this will improve in this regard. I would recommend, that you wait some more days for the new release and try it fresh with that one.

rootzoll, 2018-12-24:
@eurtnl On the README you find a new SD card relase 0.99 - thats the version 1.0RC2. I am closing this ticket for now. But if you still have problems with the new version, please reopen.

closed 2018-12-24 by rootzoll