raspiblitz/raspiblitz #187: Orientation for "some long time" - final Blockchain Sync

son-y, 2018-12-02:
I'm at the point of "Waiting for final Blockchain Sync".
It says "this can take some long time". At the moment I'm waiting longer than the FTP Download of the blockchain took - about 20 hours.
What is the orientation time this progress should take?

This is what it shows up to me:
[image]
Does the number after progress mean >99%?

I know, my issue is really near to the #121 - but can you tell me how long I have to wait to end this final sync?

son-y, 2018-12-02:
Sorry to say that I was just to eager! I waited some more hours and now everything is synced!

To the world: Wait some minutes and trust the work of rootzoll 💃

rootzoll, 2018-12-05:
OK thanks - but:  dont trust, verify ;)

closed 2018-12-05 by rootzoll

pkclyoni, 2019-01-06:
Hi,

I'm currently at the same stage and I understand the final sync will take some time. But would just like to confirm if several observations are normal.
i) Sync progress oscillating around approx. 99.9999xxx % for 5 days
ii) Bytes going into the pi amount to only approx. 10MB for the same time period (5 days)
Running on a 5400RPM hdd, and on 10Mbs broadband.

Slightly technically challenged here and would appreciate some help.

fluidvoice, 2019-01-07:
```
ssh admin@yournode
tail -f /mnt/hdd/bitcoin/debug.log
```

pkclyoni, 2019-01-07:
As soon as I ssh in, my passwords (A+C) are requested and subsequently after entering them, the sync screen appears. When/where do I enter the line ‘tail -f /mnt/hdd/bitcoin/debug.log’?

I should add that I am running a litecoin alongside the lightning node. Sorry it wasn’t mentioned before

fluidvoice, 2019-01-07:
cancel out of the menu system. Hit ctrl+c if you need to.

pkclyoni, 2019-01-07:
got it. 

but the debug.log content seems overwhelming. Anything in particular do you need me to weed out for your viewing?

fluidvoice, 2019-01-07:
as long as your sync progress has increased from what it was above, 0.993816303, and keeps increasing until 1.000000 you're prob fine. You might just have a slow connection.

pkclyoni, 2019-01-07:
A quick run through the log, and I can see multiple times (over a number of days) where I've had a progress of 1.000000. And yet on the sync display it remains fluctuating at 99.9999xx. Should this be a concern?

fluidvoice, 2019-01-07:
Not sure. Can you fund your wallet, connect to nodes, and open channels? If so then prob not.

pkclyoni, 2019-01-09:
Is it normal for the progress to be stuck at 1.000000 after each update in debug.log, yet the progress is displayed as 99.9999xxx?

fluidvoice, 2019-01-09:
I don't know. But again, if it's functioning correctly - connections, open channels, fund wallet, etc., then I'd not worry about it. I've seen on the LCD display my node slow down, lose sync at times, then some minutes later it catches back up again and everything is displayed as normal.

rootzoll, 2019-01-09:
Flipping between 0.99999.. and 1.0 is absolutly fine, because thats the usual blockchain keeping up with the lastest blocks. 

The RaspiBlitz just needs a sync close to 1.0 and finally the LND service needs to respond on the `lncli getinfo` cammand with a `synced_to_chain: true`

pkclyoni, 2019-01-10:
> Flipping between 0.99999.. and 1.0 is absolutly fine, because thats the usual blockchain keeping up with the lastest blocks.

Got it.

> The RaspiBlitz just needs a sync close to 1.0 and finally the LND service needs to respond on the lncli getinfo cammand with a synced_to_chain: true

At the current stage, this returns '[lncli] unable to read macaroon path (check the network setting!): open /home/admin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon: no such file or directory'

Images below taken at my current stage. Including them for reference.
[image]
[image]

theotherside, 2019-01-10:
How long should it typically take to go from 0-100? Running the latest version from a reformatted HDD with raspberry Pi 3. What can affect the speed?

rootzoll, 2019-01-15:
@pkclyoni try to run the lncli as bitcoin user with:
sudo -u bitcoin lncli getinfo
normally the admin should ne able to call lncli also correctly, but something seemed to got messed up. I would advise at some point (best when next release comes out) to run the update with a fresh sd card - see: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-to-update-my-raspiblitz-after-version-098

rootzoll, 2019-01-15:
> is it possible to increase the sync speed?

Thats why the RaspiBlitz offers download of a prepared blockchain via torrent or ftp if you start with an empty HDD. 

> this returns a similar message:
> '[lncli] unable to read macaroon path (check the network setting!): ...

something is messed up ... I would advise at some point (best when next release comes out) to run the update with a fresh sd card - see: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-to-update-my-raspiblitz-after-version-098

pkclyoni, 2019-02-10:
Hi, just an update to confirm if behaviour is still normal. I seem to be keeping up with the latest blocks and yet I'm still stuck at 'waiting for final blockchain sync'

[image]

rootzoll, 2019-02-13:
@pkclyoni did it catched up finally? If its still not coming up - it seems that lnd is not running or looping restarting. I that case debug logs would be helpfull: /home/admin/XXdebugLogs.sh

pkclyoni, 2019-02-15:
This is what I got from XXdebugLogs.sh:
[image]
[image]

what's next?

rootzoll, 2019-02-15:
Thats odd .. first your litecoind was running and now it seems to fail on starting up. XXdebugLogs.sh should also print the last logs from litecoind ... can you see any error/warning in there?

pkclyoni, 2019-02-15:
> XXdebugLogs.sh should also print the last logs from litecoind

Is this what you mean by litecoind logs:
[image]

They were there, but since I didn't see anything different from the logs posted in my previous posts, I just left it out.

rootzoll, 2019-02-15:
To recap: The systemd says its failing in starting the litecoind, but in the logs you see its working/running?

Hmmm .. could be that there is already a litecoind running and systemd cannot start a second one? Have you tried a reboot `sudo shutdown -r now`?

If this is not resolving. I would try with a fresh v1.0 sd card image. You can keep the already downloaded blockchain but delete the rest from the HDD with: ./XXcleanHDD.sh ... like you would update an old version - here in the FAQ starting at step3: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-to-update-a-old-raspiblitz-before-version-098

rootzoll, 2019-02-15:
Preview image of v1.0 can be downloaded here: wiki.fulmo.org/downloads/raspiblitz-v1.0-2019-02-14.img.zip

pkclyoni, 2019-02-15:
> The systemd says its failing in starting the litecoind, but in the logs you see its working/running?
looks like it :-)

> Hmmm .. could be that there is already a litecoind running and systemd cannot start a second one? Have you tried a reboot `sudo shutdown -r now`?
If this is not resolving. I would try with a fresh v1.0 sd card image. You can keep the already downloaded blockchain but delete the rest from the HDD with: ./XXcleanHDD.sh ... like you would update an old version - here in the FAQ starting at step3: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-to-update-a-old-raspiblitz-before-version-098

will give this a try and get back.

> it seems that lnd is not running or looping restarting
so no signs that this is the issue then?

pkclyoni, 2019-02-15:
Here's the result from XXdebugLogs.sh after running `sudo shutdown -r now`:
[image]

rootzoll, 2019-02-16:
Yeah shutdown/restart was not fixing it .. would go for the fresh  fresh v1.0 sd card image - you can download a preview of the v1.0 image here: wiki.fulmo.org/downloads/raspiblitz-v1.0-2019-02-14.img.zip

pkclyoni, 2019-02-17:
Everything works after updating to v1.0. Thanks for being so patient.