raspiblitz/raspiblitz #191: Different output on: Terminal vs. HDMI vs. Screen

wuestenigel, 2018-12-04:
Hey,
I just finished the setup of my Raspiblitz and at the moment I'm stock on this position in the documentation:

> Once the wallet is unlocked the setup is finally over and the main menu shows the option and features of the RaspiBlitz:

My main problem is: There are three different outputs on every device:
- My mac (via Terminal) shows the desired Info Screen
- The Raspiblitz does not show the Info Screen
- On HDMI output there is a completely different output.

Any ideas what my next step could be?

[image]
[image]
[image]

rootzoll, 2018-12-05:
ignore HDMI - its kind of deactivated to get the LCD running.

The terminal (last picture) is the "INFO" screen, correct? If its still in the startup phase, its OK that its showing missing data. But to get this clear:

1) Your LCD is showing the screen on the picture above
2) When you SSH into you get to the main menu and you can call the INFO command
3) You did the whole setup and this is happening after the final reboot

Ist that correct?

wuestenigel, 2018-12-05:
Hey @rootzoll,

thanks for getting back to me.

Ok, I will ignore HDMI.

1) My LCD shows the blue screen saying "Waiting for bitcoind"
2) Yes, I get into main meno and can call Info command (see 3rd screen)
3) Yes, flashed the SD-Card several times with the October image. Don't wanted to format the HDD because last time it took about 5 days to sync.

rootzoll, 2018-12-05:
Please SSH in and cancel the main menu. Then on terminal run the following command and let me know the output: `sudo -u bitcoin bitcoin-cli -datadir=/home/bitcoin/.bitcoin getblockchaininfo`

wuestenigel, 2018-12-05:
> admin@hedgecoin:~ $ sudo -u bitcoin bitcoin-cli -datadir=/home/bitcoin/.bitcoin getblockchaininfo
error: Could not connect to the server 127.0.0.1:8332

Make sure the bitcoind server is running and that you are connecting to the correct RPC port.

wuestenigel, 2018-12-09:
@rootzoll Hey Christian,
do you have any ideas what I can do here?

ComradePashka, 2018-12-13:
Hello!

I was involved in the process and tried to help @wuestenigel with issue.
when I checked `ps` and `netstat` output right after boot it shows bitcoind running and LISTENING on 8332 port for ipv6.
I tried to restart bitcoind manually with `sudo service bitcoind restart` and after that the command `sudo -u bitcoin bitcoin-cli -datadir=/home/bitcoin/.bitcoin getblockchaininfo` responded with message about block verifying process and in the end get JSON like response so INFO at LCD screen shown the same screen as on terminal.
so it's weird, as I understand in both cases exactly same script used with same logic -- it's polling bitcoin-cli with 5 seconds timeout interval, but for some reason it just unable to connect to bitcoind at 127.0.0.1:8332 (I also temporary add line to copy error.tmp to track command output, it was same connection problem as in manual run)

I hope that information help to apply a little fix.
Cheers!

rootzoll, 2019-01-15:
@wuestenigel @ComradePashka your problem with not being able to connect to the 8332 port could have been related to this issue #217 - please follow there if you have still this errors

Because the basic/beginning topic if this issue seems answeres Im closing for now.

closed 2019-01-15 by rootzoll