raspiblitz/raspiblitz #192: LND not ready yet ... waiting another 60 seconds. If this takes too long (more then 10min total) --> CTRL+c and report Problem

grnqrtr, 2018-12-04:
I have used both the raspiblitz-2018-10-20.img.gz card image and have also built my own using [these](https://github.com/rootzoll/raspiblitz#build-the-sd-card-image) instructions. The bitcoin side is working fine, but I continually get stuck at:
```
LND not ready yet ... waiting another 60 seconds.
If this takes too long (more then 10min total) --> CTRL+c and report Problem
```
If I do CTRL+c and then "sudo systemctl status lnd" I get:
```
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2018-12-04 14:37:23 GMT; 13s ago
  Process: 4802 ExecStart=/usr/local/bin/lnd --externalip=${PUBLICIP} (code=exited, status=1/FAILURE)
 Main PID: 4802 (code=exited, status=1/FAILURE)

Dec 04 14:37:23 RaspiBlitz systemd[1]: lnd.service: Unit entered failed state.
Dec 04 14:37:23 RaspiBlitz systemd[1]: lnd.service: Failed with result 'exit-code'.
```
I'm not sure why lnd won't run. Any suggestions? I'm using as Raspberry Pi 2 which I think should work fine since I had a lightning node setup working fine previously under the [RaspiBolt project](https://github.com/Stadicus/guides/blob/master/raspibolt).

rootzoll, 2018-12-05:
LND has problems starting up - is exiting. It would be helpful to check the logfiles:
`sudo journalctl -u lnd -b --no-pager -n20`

grnqrtr, 2018-12-05:
Here is the output of `sudo journalctl -u lnd -b --no-pager -n20`

```
admin@RaspiBlitz:~ sudo journalctl -u lnd -b --no-pager -n20
-- Logs begin at Tue 2018-12-04 14:14:42 GMT, end at Wed 2018-12-05 14:28:53 GMT. --
Dec 05 14:26:46 RaspiBlitz systemd[1]: Stopped LND Lightning Daemon.
Dec 05 14:26:46 RaspiBlitz systemd[1]: lnd.service: Ignoring invalid environment assignment '<body bgcolor=white>': /run/publicip
Dec 05 14:26:46 RaspiBlitz systemd[1]: Started LND Lightning Daemon.
Dec 05 14:26:46 RaspiBlitz systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Dec 05 14:26:46 RaspiBlitz systemd[1]: lnd.service: Unit entered failed state.
Dec 05 14:26:46 RaspiBlitz systemd[1]: lnd.service: Failed with result 'exit-code'.
Dec 05 14:27:46 RaspiBlitz systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Dec 05 14:27:46 RaspiBlitz systemd[1]: Stopped LND Lightning Daemon.
Dec 05 14:27:46 RaspiBlitz systemd[1]: lnd.service: Ignoring invalid environment assignment '<body bgcolor=white>': /run/publicip
Dec 05 14:27:46 RaspiBlitz systemd[1]: Started LND Lightning Daemon.
Dec 05 14:27:47 RaspiBlitz systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Dec 05 14:27:47 RaspiBlitz systemd[1]: lnd.service: Unit entered failed state.
Dec 05 14:27:47 RaspiBlitz systemd[1]: lnd.service: Failed with result 'exit-code'.
Dec 05 14:28:47 RaspiBlitz systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Dec 05 14:28:47 RaspiBlitz systemd[1]: Stopped LND Lightning Daemon.
Dec 05 14:28:47 RaspiBlitz systemd[1]: lnd.service: Ignoring invalid environment assignment '<body bgcolor=white>': /run/publicip
Dec 05 14:28:47 RaspiBlitz systemd[1]: Started LND Lightning Daemon.
Dec 05 14:28:47 RaspiBlitz systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Dec 05 14:28:47 RaspiBlitz systemd[1]: lnd.service: Unit entered failed state.
Dec 05 14:28:47 RaspiBlitz systemd[1]: lnd.service: Failed with result 'exit-code'.
```

grnqrtr, 2018-12-10:
I don't know what changed, but I ssh'd into my pi today and it greeted me with the option to setup a wallet. I went through the process of creating a wallet, but still seems like lnd isn't running. Now whenever I reboot it, it asks me to put in my password to unlock the wallet, but no matter what I put in for the password (correct or not) it returns:

`[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"`

I ctrl+c and check if lnd is running, but it isn't:

```
admin@RaspiBlitz:~ $ sudo systemctl status lnd
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Mon 2018-12-10 12:16:14 GMT; 35s ago
  Process: 5719 ExecStart=/usr/local/bin/lnd --externalip=${PUBLICIP} (code=exited, status=1/FAILURE)
 Main PID: 5719 (code=exited, status=1/FAILURE)

Dec 10 12:16:14 RaspiBlitz systemd[1]: lnd.service: Failed with result 'exit-code'.
```

I do `sudo journalctl -u lnd -b --no-pager -n20` again, but still returns the same as before:

```
admin@RaspiBlitz:~ $ sudo journalctl -u lnd -b --no-pager -n20
-- Logs begin at Mon 2018-12-10 07:49:27 GMT, end at Mon 2018-12-10 12:17:24 GMT. --
Dec 10 12:15:13 RaspiBlitz systemd[1]: Stopped LND Lightning Daemon.
Dec 10 12:15:13 RaspiBlitz systemd[1]: lnd.service: Ignoring invalid environment assignment '<body bgcolor=white>': /run/publicip
Dec 10 12:15:13 RaspiBlitz systemd[1]: Started LND Lightning Daemon.
Dec 10 12:15:14 RaspiBlitz systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Dec 10 12:15:14 RaspiBlitz systemd[1]: lnd.service: Unit entered failed state.
Dec 10 12:15:14 RaspiBlitz systemd[1]: lnd.service: Failed with result 'exit-code'.
Dec 10 12:16:14 RaspiBlitz systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Dec 10 12:16:14 RaspiBlitz systemd[1]: Stopped LND Lightning Daemon.
Dec 10 12:16:14 RaspiBlitz systemd[1]: lnd.service: Ignoring invalid environment assignment '<body bgcolor=white>': /run/publicip
Dec 10 12:16:14 RaspiBlitz systemd[1]: Started LND Lightning Daemon.
Dec 10 12:16:14 RaspiBlitz systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Dec 10 12:16:14 RaspiBlitz systemd[1]: lnd.service: Unit entered failed state.
Dec 10 12:16:14 RaspiBlitz systemd[1]: lnd.service: Failed with result 'exit-code'.
Dec 10 12:17:14 RaspiBlitz systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Dec 10 12:17:14 RaspiBlitz systemd[1]: Stopped LND Lightning Daemon.
Dec 10 12:17:14 RaspiBlitz systemd[1]: lnd.service: Ignoring invalid environment assignment '<body bgcolor=white>': /run/publicip
Dec 10 12:17:14 RaspiBlitz systemd[1]: Started LND Lightning Daemon.
Dec 10 12:17:15 RaspiBlitz systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Dec 10 12:17:15 RaspiBlitz systemd[1]: lnd.service: Unit entered failed state.
Dec 10 12:17:15 RaspiBlitz systemd[1]: lnd.service: Failed with result 'exit-code'.
```

Anyone have any suggestions on what to do to get lnd running?

grnqrtr, 2019-01-07:
I was messing with this again today and was finally able to solve my problem and get lnd to run.

I first edited `/etc/systemd/system/lnd.service` and manually put in my external IP address in place of `${PUBLICIP}` and lnd ran fine. So I realized that `/run/publicip` wasn't getting updated with my external IP address.

I then followed [this tutorial](https://medium.com/@stadicus/perfect-low-cost-%EF%B8%8Flightning%EF%B8%8F-node-4c2f42a4ff7b) (with small modifications) to create a "Public IP Script" and now lnd starts up no problem.

closed 2019-01-07 by grnqrtr

rootzoll, 2019-01-10:
Thanks for reporting in. The new version has a complete rewrite of setting the publicIP - hope that will run more stable.

gnipping, 2020-01-29:
I meet the samiliar question. I can use lncli create and lncli unlock, but when I use other commands. It feeds back the error. `[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"`

rootzoll, 2020-01-31:
@gnipping can you try with the new v1.4Rc3 release? see download:
https://github.com/rootzoll/raspiblitz/tree/v1.4#installing-the-software