raspiblitz/raspiblitz #196: LND v0.5-beta Not Started | Not Ready Yet

RyanSepio, 2018-12-11:
I had successfully setup a RaspiBlitz and populated with some BTC.  All was fine until a recent reboot.

After the reboot, I am able to unlock the LND wallet but when I proceed to the INFO screen

[image]

I have reformatted the SD card several times but continue with the same result.

Any ideas to help?  I would love to continue to learn about LND and support the network.

rootzoll, 2018-12-11:
Can you go to the terminal and run this command to get some debug logs from the LND?
`sudo journalctl -u lnd -b --no-pager -n20`

RyanSepio, 2018-12-11:
-- Logs begin at Thu 2016-11-03 17:16:43 UTC, end at Tue 2018-12-11 22:40:52 UTC                                                                                                                                                                                                                                             . --
Dec 11 19:00:07 MiloBigCheese lnd[11704]: 2018-12-11 19:00:07.130 [INF] LNWL: Op                                                                                                                                                                                                                                             ened wallet
Dec 11 19:00:07 MiloBigCheese lnd[11704]: 2018-12-11 19:00:07.575 [INF] LTND: Pr                                                                                                                                                                                                                                             imary chain is set to: bitcoin
Dec 11 19:00:07 MiloBigCheese lnd[11704]: unable to create chain control: unable                                                                                                                                                                                                                                              to connect to bitcoind: status code: 401, response: ""
Dec 11 19:00:07 MiloBigCheese systemd[1]: lnd.service: Main process exited, code                                                                                                                                                                                                                                             =exited, status=1/FAILURE
Dec 11 19:00:07 MiloBigCheese systemd[1]: lnd.service: Unit entered failed state                                                                                                                                                                                                                                             .
Dec 11 19:00:07 MiloBigCheese systemd[1]: lnd.service: Failed with result 'exit-                                                                                                                                                                                                                                             code'.
Dec 11 19:01:07 MiloBigCheese systemd[1]: lnd.service: Service hold-off time ove                                                                                                                                                                                                                                             r, scheduling restart.
Dec 11 19:01:07 MiloBigCheese systemd[1]: Stopped LND Lightning Daemon.
Dec 11 19:01:07 MiloBigCheese systemd[1]: Started LND Lightning Daemon.
Dec 11 19:01:08 MiloBigCheese lnd[19044]: 2018-12-11 19:01:08.204 [INF] LTND: Ve                                                                                                                                                                                                                                             rsion: 0.5.0-beta commit=v0.5-beta-101-g61e867741926bcb318432a6344b80161fabd1455                                                                                                                                                                                                                                             , build=production, logging=default
Dec 11 19:01:08 MiloBigCheese lnd[19044]: 2018-12-11 19:01:08.204 [INF] LTND: Ac                                                                                                                                                                                                                                             tive chain: Bitcoin (network=mainnet)
Dec 11 19:01:08 MiloBigCheese lnd[19044]: 2018-12-11 19:01:08.205 [INF] CHDB: Ch                                                                                                                                                                                                                                             ecking for schema update: latest_version=6, db_version=6
Dec 11 19:01:08 MiloBigCheese lnd[19044]: 2018-12-11 19:01:08.210 [INF] RPCS: pa                                                                                                                                                                                                                                             ssword gRPC proxy started at 127.0.0.1:8080
Dec 11 19:01:08 MiloBigCheese lnd[19044]: 2018-12-11 19:01:08.210 [INF] RPCS: pa                                                                                                                                                                                                                                             ssword RPC server listening on [::]:10009
Dec 11 19:01:08 MiloBigCheese lnd[19044]: 2018-12-11 19:01:08.213 [INF] LTND: Wa                                                                                                                                                                                                                                             iting for wallet encryption password. Use `lncli create` to create a wallet, `ln                                                                                                                                               [... cut ...]
Dec 11 22:40:40 MiloBigCheese systemd[1]: lnd.service: Unit entered failed state                                                                                                                                                                                                                                             .
Dec 11 22:40:40 MiloBigCheese systemd[1]: lnd.service: Failed with result 'exit-                                                                                                                                                                                                                                             code'.

rootzoll, 2018-12-11:
there is some hint that "unable to create chain control: unable to connect to bitcoind" .. lets look into the debugs of the bitcoind - to see if it has some internal problems:
`sudo tail -n 50 /mnt/hdd/bitcoin/debug.log`

Just to check: did you use some special characters in your passwords?

RyanSepio, 2018-12-11:
In my Master password yes - there is a $.  Here are the bitcoind logs

2018-12-11T20:32:48Z socket recv error Connection reset by peer (104)
2018-12-11T20:38:29Z socket recv error Connection reset by peer (104)
2018-12-11T20:40:16Z Potential stale tip detected, will try using extra outbound                                                                                                                                                                                                                                              peer (last tip update: 7898 seconds ago)
2018-12-11T20:43:56Z socket recv error Connection reset by peer (104)
2018-12-11T20:46:29Z socket recv error Connection reset by peer (104)
2018-12-11T20:48:30Z socket recv error Connection reset by peer (104)
2018-12-11T20:50:46Z Potential stale tip detected, will try using extra outbound                                                                                                                                                                                                                                              peer (last tip update: 8528 seconds ago)
2018-12-11T20:53:30Z socket send error Connection reset by peer (104)
2018-12-11T20:54:03Z socket recv error Connection reset by peer (104)
2018-12-11T20:58:31Z socket send error Connection reset by peer (104)
2018-12-11T21:01:16Z Potential stale tip detected, will try using extra outbound                                                                                                                                                                                                                                              peer (last tip update: 9158 seconds ago)
2018-12-11T21:01:43Z socket recv error Connection reset by peer (104)
2018-12-11T21:11:46Z Potential stale tip detected, will try using extra outbound                                                                                                                                                                                                                                              peer (last tip update: 9788 seconds ago)
2018-12-11T21:15:59Z socket recv error Connection reset by peer (104)
2018-12-11T21:22:16Z Potential stale tip detected, will try using extra outbound                                                                                                                                                                                                                                              peer (last tip update: 10418 seconds ago)
2018-12-11T21:23:32Z socket recv error Connection reset by peer (104)
2018-12-11T21:23:32Z socket send error Broken pipe (32)
2018-12-11T21:32:46Z Potential stale tip detected, will try using extra outbound                                                                                                                                                                                                                                              peer (last tip update: 11048 seconds ago)
2018-12-11T21:32:56Z socket recv error Connection reset by peer (104)
2018-12-11T21:43:16Z socket recv error Connection reset by peer (104)
2018-12-11T21:43:16Z Potential stale tip detected, will try using extra outbound                                                                                                                                                                                                                                              peer (last tip update: 11678 seconds ago)
2018-12-11T21:45:27Z socket recv error Connection reset by peer (104)
2018-12-11T21:46:09Z socket recv error Connection reset by peer (104)
2018-12-11T21:46:20Z socket recv error Connection reset by peer (104)
2018-12-11T21:46:51Z socket recv error Connection reset by peer (104)
2018-12-11T21:47:06Z socket recv error Connection reset by peer (104)
2018-12-11T21:48:34Z socket recv error Connection reset by peer (104)
2018-12-11T21:53:34Z socket send error Connection reset by peer (104)
2018-12-11T21:53:46Z Potential stale tip detected, will try using extra outbound                                                                                                                                                                                                                                              peer (last tip update: 12308 seconds ago)
2018-12-11T21:56:36Z socket recv error Connection reset by peer (104)
2018-12-11T21:58:35Z socket send error Connection reset by peer (104)
2018-12-11T22:04:16Z Potential stale tip detected, will try using extra outbound                                                                                                                                                                                                                                              peer (last tip update: 12938 seconds ago)
2018-12-11T22:08:35Z socket recv error Connection reset by peer (104)
2018-12-11T22:10:15Z socket recv error Connection reset by peer (104)
2018-12-11T22:13:36Z socket send error Connection reset by peer (104)
2018-12-11T22:14:46Z Potential stale tip detected, will try using extra outbound                                                                                                                                                                                                                                              peer (last tip update: 13568 seconds ago)
[... cut ...]
2018-12-11T22:53:51Z socket recv error Connection reset by peer (104)
2018-12-11T22:56:46Z Potential stale tip detected, will try using extra outbound                                                                                                                                                                                                                                              peer (last tip update: 16088 seconds ago)
2018-12-11T22:58:39Z socket send error Connection reset by peer (104)

hardwire1010, 2018-12-12:
i am currently having a similar issue as well. Also i have a ! in my password

**debug logs from the LND**
-- Logs begin at Wed 2018-12-12 03:53:31 UTC, end at Wed 2018-12-12 11:06:50 UTC                                                        . --
Dec 12 10:59:25 Node1 lnd[31528]: 2018-12-12 10:59:25.554 [INF] LTND: Primary ch                                                        ain is set to: bitcoin
Dec 12 10:59:25 Node1 lnd[31528]: unable to create chain control: unable to conn                                                        ect to bitcoind: Post http://127.0.0.1:8332: dial tcp 127.0.0.1:8332: connect: c                                                        onnection refused
Dec 12 10:59:25 Node1 lnd[31528]: 2018-12-12 10:59:25.789 [INF] LTND: Shutdown c                                                        omplete
Dec 12 10:59:25 Node1 systemd[1]: lnd.service: Main process exited, code=exited,                                                         status=1/FAILURE
Dec 12 10:59:25 Node1 systemd[1]: lnd.service: Unit entered failed state.
Dec 12 10:59:25 Node1 systemd[1]: lnd.service: Failed with result 'exit-code'.
Dec 12 11:00:26 Node1 systemd[1]: lnd.service: Service hold-off time over, sched                                                        uling restart.
Dec 12 11:00:26 Node1 systemd[1]: Stopped LND Lightning Daemon.
Dec 12 11:00:26 Node1 systemd[1]: Started LND Lightning Daemon.
Dec 12 11:00:26 Node1 lnd[32512]: 2018-12-12 11:00:26.514 [INF] LTND: Version: 0                                                        .5.0-beta commit=v0.5-beta-101-g61e867741926bcb318432a6344b80161fabd1455, build=                                                        production, logging=default
Dec 12 11:00:26 Node1 lnd[32512]: 2018-12-12 11:00:26.515 [INF] LTND: Active cha                                                        in: Bitcoin (network=mainnet)
Dec 12 11:00:26 Node1 lnd[32512]: 2018-12-12 11:00:26.515 [INF] CHDB: Checking f                                                        or schema update: latest_version=6, db_version=6
Dec 12 11:00:26 Node1 lnd[32512]: 2018-12-12 11:00:26.526 [INF] RPCS: password R                                                        PC server listening on [::]:10009
Dec 12 11:00:26 Node1 lnd[32512]: 2018-12-12 11:00:26.526 [INF] RPCS: password g                                                        RPC proxy started at 127.0.0.1:8080
Dec 12 11:00:26 Node1 lnd[32512]: 2018-12-12 11:00:26.527 [INF] LTND: Waiting fo                                                        r wallet encryption password. Use `lncli create` to create a wallet, `lncli unlo                                                        ck` to unlock an existing wallet, or `lncli changepassword` to change the passwo                                                        rd of an existing wallet and unlock it.
Dec 12 11:06:35 Node1 lnd[32512]: 2018-12-12 11:06:35.797 [INF] LNWL: Opened wal                                                        let
Dec 12 11:06:36 Node1 lnd[32512]: 2018-12-12 11:06:36.315 [INF] LTND: Primary ch                                                        ain is set to: bitcoin
Dec 12 11:06:36 Node1 systemd[1]: lnd.service: Main process exited, code=exited,                                                         status=1/FAILURE
Dec 12 11:06:36 Node1 systemd[1]: lnd.service: Unit entered failed state.
Dec 12 11:06:36 Node1 systemd[1]: lnd.service: Failed with result 'exit-code'.

**debug of the bitcoind**
2018-12-12T04:14:38Z UpdateTip: new best=0000000000000000002fa928a8c2ecf79f1d8f962745c49a8ba4f98e32de3f05 height=553482 version=0x20000000 log2_work=90.117743 tx=363838797 date='2018-12-12T04:14:27Z' progress=1.000000 cache=56.1MiB(492197txo) warning='33 of last 100 blocks have unexpected version'
2018-12-12T04:15:42Z UpdateTip: new best=000000000000000000189ffff78b77d6c408a36b99098c41fb4be2d4ff6ffceb height=553483 version=0x20000000 log2_work=90.117769 tx=363839216 date='2018-12-12T04:15:40Z' progress=1.000000 cache=56.1MiB(492341txo) warning='33 of last 100 blocks have unexpected version'
2018-12-12T04:46:49Z Potential stale tip detected, will try using extra outbound peer (last tip update: 1867 seconds ago)
2018-12-12T04:47:02Z New outbound peer connected: version: 70015, blocks=553483, peer=108
2018-12-12T04:52:53Z Pre-allocating up to position 0xb00000 in rev01459.dat
2018-12-12T04:52:54Z UpdateTip: new best=0000000000000000001cf0ebf56562a16461f1fea099950fcec09e1f2686b8a7 height=553484 version=0x20000000 log2_work=90.117795 tx=363842195 date='2018-12-12T04:51:50Z' progress=1.000000 cache=57.0MiB(500865txo) warning='33 of last 100 blocks have unexpected version'
2018-12-12T05:02:43Z UpdateTip: new best=00000000000000000001396826d49b72e9e1061665039553ac783a26a75e6dfc height=553485 version=0x20000000 log2_work=90.117821 tx=363844765 date='2018-12-12T05:02:33Z' progress=1.000000 cache=57.2MiB(502640txo) warning='32 of last 100 blocks have unexpected version'
2018-12-12T05:18:31Z UpdateTip: new best=0000000000000000001b8496b762072161e06bda6507937bba91f09cba143ea8 height=553486 version=0x20000000 log2_work=90.117848 tx=363847213 date='2018-12-12T05:18:00Z' progress=1.000000 cache=58.0MiB(510269txo) warning='31 of last 100 blocks have unexpected version'
2018-12-12T05:19:29Z UpdateTip: new best=0000000000000000001589bf541d78bc3f70f26baf9a1feb39bc05a56f404a1a height=553487 version=0x20000000 log2_work=90.117874 tx=363848016 date='2018-12-12T05:19:01Z' progress=1.00000 [... cut ...]
2018-12-12T11:05:12Z Potential stale tip detected, will try using extra outbound peer (last tip update: 2040 seconds ago)
2018-12-12T11:05:13Z New outbound peer connected: version: 70015, blocks=553515, peer=111
2018-12-12T11:05:16Z New outbound peer connected: version: 70015, blocks=553515, peer=112
2018-12-12T11:05:36Z UpdateTip: new best=0000000000000000001d7053b6cc53765bb4316d4c6f82e977778137d10321da height=553516 version=0x20000000 log2_work=90.118629 tx=363903205 date='2018-12-12T11:05:23Z' progress=1.000000 cache=12.1MiB(81063txo) warning='36 of last 100 blocks have unexpected version'

RyanSepio, 2018-12-12:
hardwire1010 - did you find any solution?

rootzoll - I posted the bitcoind logs you requested above.  Is there a problem having a special character in any password?

hardwire1010, 2018-12-13:
@RyanSepio  No not yet, i wonder if it could be as simple as having special charters. over the next few days i might test it with a different password and see if it works.

RyanSepio, 2018-12-13:
@rootzoll When I start lnd I get the following and lnd is shutdown.  Any ideas what is happening here?

lnd
2018-12-13 17:24:48.945 [INF] LTND: Version: 0.5.0-beta commit=v0.5-beta-101-g61e867741926bcb318432a6344b80161fabd1455, build=production, logging=default
2018-12-13 17:24:48.945 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2018-12-13 17:27:27.476 [INF] CHDB: Checking for schema update: latest_version=6, db_version=6
2018-12-13 17:27:27.482 [INF] RPCS: password RPC server listening on [::]:10009
2018-12-13 17:27:27.482 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
2018-12-13 17:27:27.483 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
2018-12-13 17:28:28.867 [INF] LNWL: Opened wallet
2018-12-13 17:28:29.288 [INF] LTND: Primary chain is set to: bitcoin
2018-12-13 17:28:29.326 [INF] LNWL: Started listening for bitcoind block notifications via ZMQ on tcp://*:28332
2018-12-13 17:28:29.326 [INF] LTND: Initializing bitcoind backed fee estimator
2018-12-13 17:28:29.327 [INF] LNWL: Started listening for bitcoind transaction notifications via ZMQ on tcp://*:28333
2018-12-13 17:28:29.330 [DBG] LNWL: Using minimum fee rate of 253 sat/kw
2018-12-13 17:28:35.359 [INF] LNWL: The wallet has been unlocked without a time limit
2018-12-13 17:28:35.360 [INF] LTND: LightningWallet opened
2018-12-13 17:28:35.414 [INF] LNWL: Started rescan from block 0000000000000000002363b3563bd9823068b7c5eb787d409d62d3c3bdef17d9 (height 552930) for 29 addresses
2018-12-13 17:28:35.416 [INF] LNWL: Starting rescan from block 0000000000000000002363b3563bd9823068b7c5eb787d409d62d3c3bdef17d9
2018-12-13 17:28:35.899 [INF] HSWC: Restoring in-memory circuit state from disk
2018-12-13 17:28:35.932 [INF] HSWC: Payment circuits loaded: num_pending=0, num_open=0
2018-12-13 17:28:35.959 [INF] HSWC: Trimming open circuits for chan_id=552704:1191:1, start_htlc_id=0
2018-12-13 17:28:35.959 [INF] HSWC: Trimming open circuits for chan_id=552704:1190:1, start_htlc_id=0
2018-12-13 17:28:35.959 [INF] HSWC: Trimming open circuits for chan_id=552703:2668:1, start_htlc_id=0
2018-12-13 17:28:35.988 [INF] SRVR: Scanning local network for a UPnP enabled device
2018-12-13 17:28:36.096 [DBG] LNWL: Filtering block 552931 (0000000000000000001e9f559a4794ba476b81df52baf014eeb1387730872705) with 1371 transactions
2018-12-13 17:28:37.684 [DBG] LNWL: Filtering block 552932 (0000000000000000000c6e0c753347fd4d71f1fd6cd55f461bbfccd6886f36a3) with 2976 transactions
2018-12-13 17:28:39.203 [DBG] LNWL: Filtering block 552933 (0000000000000000000b47d8eff9fd6eb358732adaadde0052bbfd74623308b7) with 2263 transactions
2018-12-13 17:28:40.607 [DBG] LNWL: Filtering block 552934 (000000000000000000238503c7257da249e93361adc24c81c784b0bd14c83c5d) with 1735 transactions
2018-12-13 17:28:41.996 [DBG] LNWL: Filtering block 552935 (000000000000000000133db99e5f95db01b597bf59bcd6c3f4340c6101156a19) with 2851 transactions
2018-12-13 17:28:44.354 [DBG] LNWL: Filtering block 552936 (00000000000000000026cdb6403ccf715cc23113e2327f7da2ea744e143331c7) with 2618 transactions
2018-12-13 17:28:44.355 [DBG] LNWL: Unable to parse output script in 5a5ba72039745007899c8328a230541e7c333dc5fe76bed2886b50b8d2ad2548:2: opcode OP_DATA_75 requires 76 bytes, but script only has 39 remaining
2018-12-13 17:28:45.988 [ERR] SRVR: Unable to discover a UPnP enabled device on the local network: context canceled
2018-12-13 17:28:45.988 [INF] SRVR: Scanning local network for a NAT-PMP enabled device
2018-12-13 17:28:46.017 [ERR] SRVR: Unable to discover a NAT-PMP enabled device on the local network: fork/exec /usr/bin/ip: no such file or directory
2018-12-13 17:28:46.018 [ERR] SRVR: unable to create server: Unable to discover a NAT-PMP enabled device on the local network: fork/exec /usr/bin/ip: no such file or directory

2018-12-13 17:28:46.018 [INF] LTND: Shutdown complete
Unable to discover a NAT-PMP enabled device on the local network: fork/exec /usr/bin/ip: no such file or directory

RyanSepio, 2018-12-14:
I finally solved the issue.

The issue was the nat=true setting.  

**2018-12-13 17:28:46.018 [ERR] SRVR: unable to create server: Unable to discover a NAT-PMP enabled device on the local network: fork/exec /usr/bin/ip: no such file or directory**

I had set the inbound port forwarding manually on my firewall and did not need the automatic upnp or nat-pmp configuration coming from the RaspiPI.  This was causing LND not to start.  Once I removed that setting from lnd.conf all started fine.

I now have a running node on a public IP.