raspiblitz/raspiblitz #201: LND locks up a few seconds/minutes after lncli unlock command

wakedog, 2018-12-17:
Running the RaspiBlitz (latest version) and bitcoin chain is fully synced, as well as lnd. After bootup, I unlock with: lncli --network=mainnet unlock

LND unlocks, starts and after a few seconds. After running: sudo journalctl -f -u lnd, I see it starts scanning the chain and then fails with:

Dec 17 08:41:48 RaspiBlitz lnd[24439]: 2018-12-17 08:41:48.046 [INF] HSWC: Trimming open circuits for chan_id=554110:2114:1, start_htlc_id=0
Dec 17 08:41:48 RaspiBlitz lnd[24439]: 2018-12-17 08:41:48.207 [DBG] LNWL: Filtering block 554223 (00000000000000000005a5e722ad27cdeeaedc06f4c6a40f66d283e61a196c16) with 1621 transactions
Dec 17 08:41:48 RaspiBlitz lnd[24439]: 2018-12-17 08:41:48.251 [ERR] SRVR: unable to create server: can't create router: unexpected EOF
Dec 17 08:41:48 RaspiBlitz lnd[24439]: 2018-12-17 08:41:48.253 [INF] LTND: Shutdown complete
Dec 17 08:41:48 RaspiBlitz lnd[24439]: can't create router: unexpected EOF
Dec 17 08:41:48 RaspiBlitz systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Dec 17 08:41:48 RaspiBlitz systemd[1]: lnd.service: Unit entered failed state.
Dec 17 08:41:48 RaspiBlitz systemd[1]: lnd.service: Failed with result 'exit-code'.
Dec 17 08:42:48 RaspiBlitz systemd[1]: lnd.service: Service hold-off time over, scheduling restart.
Dec 17 08:42:48 RaspiBlitz systemd[1]: Stopped LND Lightning Daemon.
Dec 17 08:42:48 RaspiBlitz systemd[1]: Started LND Lightning Daemon.
Dec 17 08:42:48 RaspiBlitz lnd[25620]: 2018-12-17 08:42:48.683 [INF] LTND: Version: 0.5.0-beta commit=v0.5-beta-101-g61e867741926bcb318432a6344b80161fabd1455, build=production, logging=default
Dec 17 08:42:48 RaspiBlitz lnd[25620]: 2018-12-17 08:42:48.684 [INF] LTND: Active chain: Bitcoin (network=mainnet)
Dec 17 08:42:48 RaspiBlitz lnd[25620]: 2018-12-17 08:42:48.684 [INF] CHDB: Checking for schema update: latest_version=6, db_version=6
Dec 17 08:42:48 RaspiBlitz lnd[25620]: 2018-12-17 08:42:48.689 [INF] RPCS: password RPC server listening on [::]:10009
Dec 17 08:42:48 RaspiBlitz lnd[25620]: 2018-12-17 08:42:48.689 [INF] RPCS: password gRPC proxy started at 127.0.0.1:8080
Dec 17 08:42:48 RaspiBlitz lnd[25620]: 2018-12-17 08:42:48.690 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.

Any ideas why it would work, then fail almost immediately after?

I have disabled UFW to rule out any blocked ports and verified passwords are all valid. Thanks in advance.

wakedog

fluidvoice, 2018-12-18:
Are you logged in as user "bitcoin"?  If you look at the shell script AAunlockLND.sh you will see it starts up using:
`sudo -u bitcoin /usr/local/bin/lncli --chain=${network} unlock`

wakedog, 2018-12-18:
Thanks @fluidvoice I got it going. Was using admin account instead of bitcoin. Seems to be working now. Thanks for the help - this is resolved!

closed 2018-12-18 by wakedog