raspiblitz/raspiblitz #204: Research: Connecting with Joule Browser Extension

rootzoll, 2018-12-19:
Check the integration of the Joule Browser extension: https://github.com/wbobeirne/joule-extension

wuestenigel, 2019-01-07:
Nice, what is the NODE URL / API REST URL and basically the Port?

rootzoll, 2019-01-09:
@wuestenigel port 8080 should be correct. Did it worked when using `http://` instead of `https://` ?

wuestenigel, 2019-01-10:
Hey, unfortunately not working with http://
https:// is working, but there is this handshake error. Shall I just wait for 1.00 release?

rootzoll, 2019-01-10:
I have to run tests myself - hope to get this fixed for 1.0 , but if there is some "deeper" problem still it might get shifted further back. Any additonal debug input welcome.

rootzoll, 2019-01-15:
@wuestenigel I got it working: Yes its https. In the error message there is the "click this link" - this link needs to bne opened in the same browser in a new tab (should do automatically). Then the browser complains that its a self-signed cert, but gives you the option to add it to the certs you trust. When you do so and go back - now the URL `https://[YOUR-LOCAL-IP]:8080` works.

On the next page you need to upload the admin and the readonly macaroon. To add those click the small "hexstring option" in the form. Now instead of uploads you should insert two strings into a form. You get these strings from your RaspiBlitz Terminal with:

`sudo xxd -ps -u -c 1000 /mnt/hdd/lnd/data/chain/bitcoin/mainnet/admin.macaroon`
`sudo xxd -ps -u -c 1000 /mnt/hdd/lnd/data/chain/bitcoin/mainnet/readonly.macaroon`

And then Joule is all setuo and ready to go :)

I plan to add an additional feature to the main menu to export macaroon and cert more easily - if this works fast I will add it to version 1.0.

justrossthings, 2019-01-17:
This is awesome

wuestenigel, 2019-01-17:
Thanks for the work! :)
However, the Joule Github commented that I should change my local lnd configuration in order to get it running. So now I'm a little bit confused: https://github.com/wbobeirne/joule-extension/pull/43#issuecomment-454827154

Just wait vor v1.0? ;)

rootzoll, 2019-01-17:
@wuestenigel The RaspiBlitz already works with `tlsextraip=0.0.0.0` ... you should be fine.

When you open the URL in the browser: `https://[YOUR-LOCAL-RASPIBLITZ-IP]:8080` ... what does your browser say?

wuestenigel, 2019-01-17:
@rootzoll The output of https://192.168.178.65:8080/ is: 

> Not Found

The output of https://192.168.178.65:8080/v1/getinfo is:

> {"error":"all SubConns are in TransientFailure, latest connection error: connection error: desc = \"transport: authentication handshake failed: x509: certificate is valid for 127.0.0.1, ::1, 192.168.178.65, not 0.0.0.0\"","code":14}

rootzoll, 2019-01-17:
OK please check quick if you have the `tlsextraip=0.0.0.0` in your lnd.conf - check also if commented out or something - use:
`sudo cat /mnt/hdd/lnd/lnd.conf`

If the lnd.conf looks OK, try to generate a new TLS cert:
`sudo /home/admin/config.scripts/lnd.newtlscert.sh`

Then reboot and try again:
`sudo shutdown -r now`

... if that is not helping, try again with next RaspiBlitz release.

wuestenigel, 2019-01-18:
Worked perfectly. Unfortunately you don't have a tippin.me account, so I had to test it with yalls! :-) 

Also appreciate that you are in communication with Zap about the mobile wallet topic.

https://twitter.com/wuestenigel/status/1086165539574243329

Dropin16, 2019-03-09:
> If the lnd.conf looks OK, try to generate a new TLS cert:
sudo /home/admin/config.scripts/lnd.newtlscert.sh

Did this and now my node is stuck in a continuous boot loop. Sync from around 99.96% -> then I get this error:

error: Could not connect to the server 127.0.0.1:8332

Make sure the bitcoind server is running and that you are connecting to the correct RPC port.
tail: cannot open '/mnt/hdd/lnd/logs/bitcoin/net/lnd.log' for reading: No such file or directory
tail: cannot open '/mnt/hdd/lnd/logs/bitcoin/net/lnd.log' for reading: No such file or directory
tail: cannot open '/mnt/hdd/lnd/logs/bitcoin/net/lnd.log' for reading: No such file or directory
(standard_in) 1: syntax error
tail: cannot open '/mnt/hdd/lnd/logs/bitcoin/net/lnd.log' for reading: No such file or directory

Then the next screen with this message pops up: x error code: -28 error message: Loading block index... 
Followed by:  error code: -28 error message: Verifying blocks...    
Then I'm prompted to enter the LND password and after entering the node starts syncing from 99.96% again.

rootzoll, 2019-03-09:
@Dropin16 that sounds strange. Would recommend to shutdown RaspiBlitz (cut power if needed), reflash the sd card with a fresh version of the v1.0 image and go thru the update/recover process. 

Your data is on the HDD, so this process is just to get the OS setup correctly again. On final login you just have to set a new PasswordA.