raspiblitz/raspiblitz #205: Recover funds

jetlagv1, 2018-12-20:
My raspiblitz system doesn't want to start lnd anymore after rebooting. It looks like it the harddrive is corrupted. How can I recover the funds? Some funds are in channels and some in the wallet.

CryptoSigmund, 2018-12-20:
Hello jetlagv1,

Rootzoll write some helpfull informations about this state of the raspiblitz.

https://github.com/rootzoll/raspiblitz#recover-your-coins-from-a-failing-raspiblitz

hope that helps you!

jetlagv1, 2018-12-20:
Hi CryptoSigmunt. Thank you for posting the link but it doesn't seem to take me anywhere. I'm not that familiar with github and I'm using it on my phone right now but the link only takes me to the open issues page. Can you point me a little further?

CryptoSigmund, 2018-12-20:
Then it's a browser issue... it should bring you to the bottom of the first page where Rootzoll discribes how to recover funds.
Headline is "Recover your Coins from a failing RaspiBlitz" ;)

fluidvoice, 2018-12-20:
"It looks like it the hard drive is corrupted." why do you think this?  You should post debug log output so we can see the errors. Try the terminal command 
`sudo journalctl -f -u lnd`

fluidvoice, 2018-12-22:
I'm thinking there is nothing specific about Raspiblitz with respect to wallets and keys.  It's running Bitcoin Core under the hood. So you might get better results searching and posting this question on Bitcoin core discussion sites. I don't know the answer off the top of my head.  See: [https://bitcoin.org/en/community](https://bitcoin.org/en/community)

rootzoll, 2019-01-09:
To clear things up: Bitcoind is not managing any wallet - you can see that wallet funtion is deactivated in `/mnt/hdd/bitcoin/bitcoin.conf`. @WetPuppyKisses thats why a dumpprivatekey command is not working on bitcoind.

LND is doing all the wallet handling of your on-chain and off-chain funds. To backup and recover your private key - thats the word seed list you get when creating your LND wallet.

Sadly it seems that LND has still no export function after that:
https://github.com/lightningnetwork/lnd/issues/732

When you setup a fresh RaspiBlitz and you come to the point to create the LND wallet you can say that you already have a Mnemonic Seed (the word list) and enter it. Then LND will work with your old private key and recover all the on-chain funds for you.

I never tried, but the Mnemonic Seed (word list) should be standard and should also work in other bitcoin wallets.

The thing with your off-chain funds (bitcoin locked in multisig lighting channel) is that the Mnemonic Seed does not recover all your channel states. Here it gets tricky .. LND should be able on rescanning the blockhain discover that you have a funds in a channel - but cannot force close it. If you are lucke your channel counter part will force close and unfreeze the funds - but there is also the danger that the channel counter part will try to cheat with an old state and you cannot stop it.

That brings us to the topic of backup on your node ... I started a FAQ on that:
https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md

For more details on recover funds please checkout the LND Dev Slack channel https://lightningcommunity.slack.com 

I am closing this issue for RaspiBlitz for now. If anybody finds good resources on the internet about this topic please still post here.

closed 2019-01-09 by rootzoll

walletofsatoshi, 2019-04-09:
Seed is not working on other bip39 Wallets like electrum

walletofsatoshi, 2019-05-02:
so I recovered funds on new setup today with lnd folder from previous setup on a new setup. When I check funds directly via raspiblitz menu I am getting this error but when I connected via zapwallet reckless can access all funds and all channels were recovered. So if someone needs help if not finding funds after recovery just connect via zapwallet reckless and it should show up. If you need any help ping me.
./00mainMenu.sh: line 419: lnbalance: command not found
Press ENTER to return to main menu.

rootzoll, 2019-05-13:
The `./00mainMenu.sh: line 419: lnbalance: command not found` error is not critical an will be fixed in next release.

@walletofsatoshi can you describe more in detail on how you managed the recover with "zapwallet reckless" - is that a special version of zap wallet?

walletofsatoshi, 2019-05-13:
No I am connecting zap wallet over internet . I and ran few commands from terminal for command not found from earlier similar issue which fixed it.

The balance was not showing in raspiblitz since it was giving error with command not found issue but was showing on my zapwallet.

rootzoll, 2019-05-15:
@walletofsatoshi ah ok got it, thx.