raspiblitz/raspiblitz #214: Add BTCPayServer as service option

rootzoll, 2018-12-25:
See this great base work from @normandmickey
https://gist.github.com/normandmickey/6d6e74ebb6b3ddbd317bc1450f48f08e

Test and convert into a 'config.script' for RaspiBlitz.

gabrieldov, 2019-02-08:
Nice, will try it next week

rektlesspinko, 2019-02-15:
I followed the instructions and got it working. But for some reason can't integrate joule with it, don't see the LND services listed in crypto services in BTCPay / server settings / services so can't get the QR details to link it to joule. Be interested if you get the same issue @gabrieldov

DaReaLDeviL, 2019-02-25:
Is this still a thing? A install script with in the "services" menu whould be awesome!

normandmickey, 2019-02-25:
Yes, I haven't had any time to try to script it.  But if you can follow the
instructions they should work.

rootzoll, 2019-02-26:
@normandmickey let me know if you have a basic script ready .. I can then integrate it into the rest of RaspiBlitz.

normandmickey, 2019-07-09:
Updated Gist to install BTCPayServer on Raspiblitz V1.3 for RPI4B
https://gist.github.com/normandmickey/3f10fc077d15345fb469034e3697d0d0

Appears to work but would consider it a WIP.  Would still like to "scriptify" so it could be added as a menu option.  RPI4B improved performance over RPI3 makes this a viable option now.

openoms, 2019-09-10:
Made the script, but be aware that it is experimental, recommended only for test nodes first. Got it working once already, but SSL certificate asks are limited (5 in an hour - https://letsencrypt.org/docs/rate-limits/) so need to give it a break.
Nginx is very fussy, the setup tends to break previous Nginx configs (Like Electrs SSL), so added a warning if previous nginx.conf is found.
Script: https://raw.githubusercontent.com/openoms/bitcoin-tutorials/master/BTCPayServer/btcpay_to_blitz.sh

to download and run: 
`wget https://raw.githubusercontent.com/openoms/bitcoin-tutorials/master/BTCPayServer/btcpay_to_blitz.sh && bash btcpay_to_blitz.sh`

There is a readme too for the manual process based on @normandmickey`s gist:
https://github.com/openoms/bitcoin-tutorials/blob/master/BTCPayServer/BTCPayServer_on_the_RaspiBlitz.md

openoms, 2019-09-11:
With the help of the script above now I am running BTCPayServer with my longstanding LN-node.
The same RPi4 runs Electrs with Nginx serving both.
Minimal manual debugging needed in nginx.conf only.

See BTCpay run on check.DIYnodes.com.

normandmickey, 2019-09-11:
That's awesome!  Can't wait to try your script out.  I started to write one
a while ago but gave up.  Glad you picked up the ball and ran with it!

grnqrtr, 2019-09-13:
I tried again with fresh 1.3 SD card, but still didn't work. Here is a [pastbin](https://pastebin.com/sgsqEx0a 
) of most of my session. I see that there is a "Build FAILED" in there.

After running script once, I checked /home/admin/.nbxplorer/Main/settings.config and noticed that the rpc user and password weren't set again. I also realized that /home/admin/.btcpayserver/Main/settings.config didn't even exist so I re ran script, then still set some things manually in the file.

openoms, 2019-09-13:
@grnqrtr thanks for testing and giving log (and good job hiding your RPC password - will modify the script to hide that in the output)

You seem to get errors while building the BTCPay component. Strange that it was running well on mine with a fresh 1.3 image. Are you running as the user `admin` from `/home/admin/` ?

Could use the commands one-by-one manually or normandmickey-s original instructions to catch the fault.

grnqrtr, 2019-09-14:
Yes, I'm running as `admin` from `/home/admin` . I think my SD cards are okay. I've been going back and forth between two different ones, but was having problems on both.

I ended up getting it running fine by using your manually instructions here: https://github.com/openoms/bitcoin-tutorials/blob/master/BTCPayServer/BTCPayServer_on_the_RaspiBlitz.md

[Up and running](https://grnqrtr.raspberryip.com/embed/7uwG9JgHEJSe8neMM4H9ywGQPwQqQS3Dry2Jwao5oCv1/BTC/ln).

openoms, 2019-09-18:
@grnqrtr 
> I'd be curious to hear if anyone else is having problems with the script or just me.

Caught the problem. NBXplorer needs some time to start and create the `/home/admin/.nbxplorer/Main/settings.config` . Added a check and `sleep 10` while the file is not there, so should be sorted now.

Also tested running on top of previous installation. Worked so far.

grnqrtr, 2019-09-18:
I ran on top of previous installation and seemed to work fine!

One small thing I noticed was at the beginning if there is an existing Nginx configuration it warns and says `"Press CRTL+C to abort or any key to continue"` but I have to hit ENTER to continue, any key doesn't progress the script.

Thanks for putting this script together @openoms!

frennkie, 2019-09-19:
Just one remark. The Lets Encrypt certificate will expire after 90 days. So a renew (cronjob) will be needed.

openoms, 2019-09-19:
Thanks for the feedback @frennkie and @grnqrtr , will modify and test again. 
Do you guys think we should run BTTPayServer as it`s own user (just like it has been setup for Electrs too)? .NET is hog of dependecies, might be good to just keep it in it`s own enclave without sudo access and password disabled.

openoms, 2019-09-20:
@frennkie the `webroot` method failed when using a ddns and `standalone` worked so sticked to it. The time difference is couple seconds between nginx start-stop and reload in every 2 months.
Used a systemd service and timer to renew with certbot because it is logged better and has been already tested well for other services. 
Made the certificate generation less interactive with the help of the link provided.

The script is running quite reliably on the Rpi4 now. The .NET build of NBXplorer fails on the Odroid (same 32bit armv7las the RPi) for some reason.

grnqrtr, 2019-09-21:
> might be good to just keep it in it`s own enclave without sudo access and password disabled.

I honestly don't know, but it sounds cleaner and more secure to have it's own user.

openoms, 2019-09-22:
Update: https://github.com/openoms/bitcoin-tutorials/blob/master/BTCPayServer/btcpay_to_blitz.sh
* BTCpayServer, NBXplorer and the .NET Core is now confined to the user `btcpay`.
* The BTCPay data and store settings are stored on the HDD `/mnt/hdd/.btcpayserver` and recoverable on a reinstall / SDcard update.

jodobear, 2019-09-23:
@openoms I haven't tested your script on my rpi3 yet (still trying to recover funds), but as far as i understand from reading this thread, you are basically deploying the BTCPay docker image. So, now the question is how is this different from the Hack0? Or rather what advantages does the Hack0 have over Raspiblitz now?

normandmickey, 2019-09-23:
The script doesn't use Docker, it manually installs BTCPayServer and its dependencies (Dot Net and Nbxplorer) and uses the existing Bitcoin and LND Nodes of the Blitz.  A lot of Raspiblitz users have expressed a desire to add BTCPayServer instead of running a second node.  This is one way to do that.

jodobear, 2019-09-23:
So, then i'd argue that this is a better way to deploy BTCPay than the docker since the docker is used primarily to make VPS hosting simpler. Am i right? Also, is it also good to have docker since it encapsulates BTCPay in it's own container?

normandmickey, 2019-09-23:
I am not qualified to say whether Docker is good or bad.  It definitely
makes installing and updating BTCPayServer very simple. There are a lot of
moving parts to a Lightning Nodes  I would just say the Raspiblitz as more
techie (maker/hacker) focused and the BTCPayServer more merchant focused.
But there is a lot of overlap.

openoms, 2019-09-23:
Positives of using this script with the RaspiBlitz:
* avoiding Docker altogether
* NBXplorer and BTCPayServer are built from source - a middleman is out
* connects to the existing bitcoind and lnd - no more synching/funding/no need for new channels
* the user has the seed and the backup of the LND node unlike in the BTCPay docker setup
* overall flexibilty - you can still use RaspiBlitz to anything else
* should be able to keep all the store data during an update

Negative:
* less (more complicated) support from the BTCPay team
* Can break when updating either RaspiBlitz or BTCPayServer - will need continuous testing

openoms, 2019-09-23:
The hack0 is great to physically deploy BTCPayServer and decentralize further. As you correctly said @jodobear it uses the same dockerised process as the deployment to a VPS like LunaNode, but running on your own hardware. 
I bought it myself for two reasons:
* wanted a RockPro64 anyway - the hardware is the same as what the BitBoxBase uses and it costed what I would pay anywhere
* It has the signature of Nicolas Dorier (priceless)

For the reasons detailed above I will probably run the RaspiBlitz on it see: https://github.com/rootzoll/raspiblitz/tree/master/alternative.platforms#armbian-buster. Can even make an SDcard/ eMMC image for that hardware if it works well. Will also test the BitBoxBase. As discussing the RockPro64/hack0 is off-topic here I opened a separate issue: https://github.com/openoms/raspiblitz/issues/58

jodobear, 2019-09-24:
> It has the signature of Nicolas Dorier (priceless)
I bought it for the exact same reason and want to start learning about this stuff better.

I like the flexibility i get when working with linux, i am free to build it as i want and not dependent on docker images as you precisely point out, though it makes sense from a plug n play pov.

I'm going to build BTCPay on my blitz and compare.

Apologies for hijacking the thread.

frennkie, 2019-09-26:
@openoms I assume you have seen and read this message

```
Telemetry
---------
The .NET Core tools collect usage data in order to help us improve your experience. The data 
is anonymous and doesn't include command-line arguments. The data is collected by Microsoft 
and shared with the community. You can opt-out of telemetry by setting the 
DOTNET_CLI_TELEMETRY_OPTOUT environment variable to '1' or 'true' using your favorite shell.

Read more about .NET Core CLI Tools telemetry: https://aka.ms/dotnet-cli-telemetry
```

My strong strong recommendation would be to turn this into "opt-in" by adding this step in the script: 

```
echo "DOTNET_CLI_TELEMETRY_OPTOUT=1" | sudo tee -a /etc/environment
```

Anybody **wanting** to send this data to Microsoft can remove it.

openoms, 2019-09-27:
@frennkie thanks for pointing this out. Added the opt out line to the script.

frennkie, 2019-10-03:
@openoms Maybe it makes sense the *pin* the script to a specific - known to work - releases. E.g after https://github.com/openoms/bitcoin-tutorials/blob/master/BTCPayServer/btcpay_to_blitz.sh#L88 this `sudo -u btcpay git checkout v2.0.0.58` could be done so that the result of running the script is somewhat deterministic (same versions).

I just now ran into this (with the implicit `master`)

```
$ sudo -u btcpay ./build.sh
Microsoft (R) Build Engine version 15.9.20+g88f5fadfbe for .NET Core
Copyright (C) Microsoft Corporation. All rights reserved.

/home/btcpay/dotnet/sdk/2.2.102/Sdks/Microsoft.NET.Sdk/targets/Microsoft.NET.TargetFrameworkInference.targets(137,5): error NETSDK1045: The current .NET SDK does not support targeting .NET Core 3.0.  Either target .NET Core 2.2 or lower, or use a version of the .NET SDK that supports .NET Core 3.0. [/home/btcpay/NBXplorer/NBXplorer/NBXplorer.csproj]

Build FAILED.

/home/btcpay/dotnet/sdk/2.2.102/Sdks/Microsoft.NET.Sdk/targets/Microsoft.NET.TargetFrameworkInference.targets(137,5): error NETSDK1045: The current .NET SDK does not support targeting .NET Core 3.0.  Either target .NET Core 2.2 or lower, or use a version of the .NET SDK that supports .NET Core 3.0. [/home/btcpay/NBXplorer/NBXplorer/NBXplorer.csproj]
    0 Warning(s)
    1 Error(s)
```

openoms, 2019-10-03:
@frennkie Good point will do this for btcpayserver and NBxplorer too. Also. NET needs to be kept at the tested version. The BTCpay team is just starting to test building on .NET 3.0.

openoms, 2019-10-04:
Updated the script to checkout both NBxplorer and BTCpayserver to the last commit which was tested successfully. Will only update after testing so there should be no surprise errors again. https://github.com/openoms/bitcoin-tutorials/commit/ed0833f2735defb9176c8dadcb48054128402303

frennkie, 2019-10-06:
My BTCPayServer was not responding just now. 502 Bad Gateway from nginx. I checked and the Node process was running on IPv4 127.0.0.1:23000.

Nginx ist configured to connect to `localhost`. But that could mean either `127.0.0.1` or `::1`. And that was my issue... In the nginx logs I saw that it was trying to connect to `[::1]:23000`.

I think Btcpay should also listen to the IPv6 socket... But that might require some code changes... Therefore I have changed the nginx reverse proxy from. `localhost` to `127.0.0.1`.

grnqrtr, 2019-10-11:
I just finally had the chance to run this updated script again after re-syncing up a fresh node. Everything seemed to install fine, though I seem to only be able to access BTCPay from the Tor hidden service address but not from the node's local IP and not from the DDNS address I setup.

Is that the way it should be? I have TOR service enabled on RaspiBlitz and also set up a TOR relay before using your script, could that be why? Previously I had TOR enabled on RaspiBlitz, but was able to access BTCPay from the local IP and DDNS. Though I don't remember what order I did things previously. Maybe I had installed BTCPay with your script and then enabled TOR on RaspiBlitz?

openoms, 2019-10-11:
@grnqrtr it should work on LAN for sure and on the domain/dyndns too if you have the ports 80 and 443 forwarded.
Check if nginx is serving ok with:
`sudo nginx -t`

grnqrtr, 2019-10-11:
It seems okay, `sudo nginx -t` returns:
```
nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
nginx: configuration file /etc/nginx/nginx.conf test is successful
```

openoms, 2019-12-12:
FYI BTCPay Server with the current install script does make many regular outside calls which might not be a good fit with a privacy focused node.

Nicolas Dorier wrote that they are likely the price queries made often to avoid delays with invoice creation.

It is slightly different at every restart (reinstalled too). Took out the localhost connections and in the rest of the list there are connections to:
* Microsoft
* Digital Ocean
* Cloudflare
* amazonaws.com
* cloudfront.net
This is without connecting to a domain, only a self-signed self-signed certificate and Tor.

```
admin@RaspiBlitz:~ $ sudo -u btcpay lsof -i
COMMAND   PID   USER   FD   TYPE  DEVICE SIZE/OFF NODE NAME
dotnet  24686 btcpay  113u  IPv4 6793639      0t0  TCP 192.168.1.236:43617->52.169.64.244:https (ESTABLISHED)
dotnet  24686 btcpay  122u  IPv4 6794477      0t0  TCP 192.168.1.236:42627->52.169.64.244:https (ESTABLISHED)
dotnet  24718 btcpay  406u  IPv4 6797664      0t0  TCP 192.168.1.236:38121->159.203.144.149:https (ESTABLISHED)
dotnet  24718 btcpay  417u  IPv4 6797708      0t0  TCP 192.168.1.236:45663->104.16.214.191:https (ESTABLISHED)
dotnet  24718 btcpay  418u  IPv4 6797706      0t0  TCP 192.168.1.236:38735->159.203.144.149:https (ESTABLISHED)
dotnet  24718 btcpay  429u  IPv4 6797710      0t0  TCP 192.168.1.236:45559->ec2-99-79-108-36.ca-central-1.compute.amazonaws.com:https (ESTABLISHED)
dotnet  24718 btcpay  431u  IPv4 6797712      0t0  TCP 192.168.1.236:43719->server-13-224-77-74.man50.r.cloudfront.net:https (ESTABLISHED)
dotnet  24718 btcpay  433u  IPv4 6797714      0t0  TCP 192.168.1.236:40157->104.18.91.37:https (ESTABLISHED)
```

rootzoll, 2019-12-20:
Just saw this on install test:
```
[Install]
WantedBy=multi-user.target

Created symlink /etc/systemd/system/multi-user.target.wants/btcpayserver.service → /etc/systemd/system/btcpayserver.service.
**tee: /home/btcpay/.btcpayserver/Main/settings.config: No such file or directory**
```

rootzoll, 2019-12-20:
After test install - calling `https://192.168.178.99` brings up no page (I choosed run behind TOR during install). @openoms whats the best debug path from here

openoms, 2019-12-20:
@rootzoll It seems that BTCPayserver sis not install if you did not have the directory for the settings file created.

Is the `/mnt/hdd/.btcpayserver` dir there?
```
sudo systemctl status nbxplorer
sudo systemctl status btcpayserver

sudo nginx -t
```

show any errors?

A reinstall might also help.

openoms, 2019-12-20:
Also the self signed certificate leads to errors in all the browsers, not easily dealt with in some.

Try the Hidden Service address in Tor browser too  that bypasses Nginx.

openoms, 2019-12-23:
@rootzoll managed to reproduce this: 
```
$ sudo nginx -t
nginx: [emerg] open() "/etc/nginx/sites-enabled/btcpayserver" failed (2: No such file or directory) in /etc/nginx/nginx.conf:62
nginx: configuration file /etc/nginx/nginx.conf test failed
```
the problem is that the bonus.btcpayserver.sh cleanes out `/etc/nginx/sites-enabled/btcpayserver` after it has been created.
Will test the solution and make a PR soon.

openoms, 2019-12-23:
Fixed in the commits above.

openoms, 2019-12-24:
Added lines to store the btcpay data in the app-data dir on the disk to be able to preserve setings and acccounts:  https://github.com/rootzoll/raspiblitz/pull/927/commits/ad150eee2eeb2489a8ff03122e754aafb5c6dd60

bjadel, 2020-01-01:
I get the following error:

sudo journalctl -u  btcpayserver.service
...
dotnet[22799]: The application to execute does not exist: '/home/btcpay/btcpayserver/BTCPayServer/bin/Release/netcoreapp2.1/BTCPayServer.dll
...

What can I do?

That is true, the folder "/home/btcpay/btcpayserver/BTCPayServer/bin" doesn't exist.
`

openoms, 2020-01-01:
@bjadel best is to use the 1.4 RC1 image and install/reinstall from menu.

How did you install so far?

bjadel, 2020-01-01:
> @bjadel best is to use the 1.4 RC1 image and install/reinstall from menu.
> 
> How did you install so far?

I use the version 1.4 RC1 and install the server from service menu.

bjadel, 2020-01-01:
I have solved the problem. In my case the service menu script has not build the BTCPayServer.

openoms, 2020-01-02:
> I have solved the problem. In my case the service menu script has not build the BTCPayServer.

Did you just reinstall or used some other method? C

bjadel, 2020-01-02:
> > I have solved the problem. In my case the service menu script has not build the BTCPayServer.
> 
> Did you just reinstall or used some other method?
No, the reinstall didn't solved the problem. I switch to user btcpay and enter the folder '/home/btcpay/btcpayserver/'. In this folder I run the build script. After that the folder 'bin' was available.

openoms, 2020-01-03:
> No, the reinstall didn't solved the problem. I switch to user btcpay and enter the folder '/home/btcpay/btcpayserver/'. In this folder I run the build script. After that the folder 'bin' was available.

What you describe is the same what is happening in the script: https://github.com/rootzoll/raspiblitz/blob/v1.4/home.admin/config.scripts/bonus.btcpayserver.sh#L172

Not sure what caused your error.
Did you delete directories or did other things manually?

bjadel, 2020-01-03:
I had the problem that I started the build script first as admin user. This caused errors. Then I wanted to start the script with the user btcpay. There was an error message that a lock is set. So I did the following:

sudo rm /tmp/NuGetScratch/lock/83712a22a258d25bd50ce3d8d718d0b49a03a0f9

Afterwards the build command could be executed successfully.

openoms, 2020-01-04:
@bjadel ok, so as you can see in the script all installation is done with the btcpay user so runinng manually in admin should cause errors

Could your initial error have been because a download error? In that case the uninstall and reinstall should have helped since you have done the same manually.

Maybe when the install errorred out early the SERVICE menu did not show BTCpay as on so the `./config.scripts/bonus.btcpayserver.sh off` was not run to clean.
Will try to reproduce this.

bjadel, 2020-01-05:
> @bjadel ok, so as you can see in the script all installation is done with the btcpay user so runinng manually in admin should cause errors
> 
> Could your initial error have been because a download error? In that case the uninstall and reinstall should have helped since you have done the same manually.
Now, it is not easy to reproduce that. I believe it was not a download error. 

> 
> Maybe when the install errorred out early the SERVICE menu did not show BTCpay as on so the `./config.scripts/bonus.btcpayserver.sh off` was not run to clean.
> Will try to reproduce this.
That would be possible. 

Would it still be a good idea to clean the /tmp/NuGetScratch folder during the clean process? Maybe with the command:

`dotnet nuget locals all --clear`

or

`dotnet nuget locals temp --clear`

rootzoll, 2020-01-06:
@openoms this is your call - let me know if there is anything to add to the BTCPayServer script before final v1.4 release

openoms, 2020-01-13:
Included `dotnet nuget locals all --clear` and made some other improvements in:
https://github.com/rootzoll/raspiblitz/pull/974/commits/6ce189b3f8ff4b8f0e75370ca8779d23789a2d76

BTCPay is now at the latest:  v1.0.3.146

rootzoll, 2020-01-20:
@openoms I tested the BTCPayServer install on a fresh install again and it seems not to work yet. I ran behind TOR and in the end it told me to go to a onion.address - but there was nothing reachable. 

Also there is no file called: ´/home/btcpay/.btcpayserver/Main/settings.config´ there is just a link `/home/btcpay/.btcpayserver` pointing to `/mnt/hdd/app-data/.btcpayserver` that is empty.

Where should I start to debug? Maybe we should push this feature to 1.5 until its better tested?

openoms, 2020-01-20:
The problem must be with the links. Are you testing on a BTRFS disk?
For me it gave no errors on RC1 and RC2 on the EXT4 disks.
Will build again and find out.

openoms, 2020-01-20:
BTCPay is of course not essential, it should not hold back the 1.4 release.
It can always be offered as a manual setup later and I do prefer running it as a separate node.

openoms, 2020-01-21:
Made a large overhaul and a good couple of rounds of testing. Give BTCPay another chance to get in 1.4!

rootzoll, 2020-01-21:
I am testing on EXT4 hdd/ssd ... thx for the fix - will test again.

For the v1.4 we can keep the bonus-BTC script as experimental in there - but if we cannot fix install issues within this week I would like to uncomment BTC-PayServer from the service menu for that release and push it to v1.5.

[17 more replies]

closed 2020-01-28 by rootzoll