raspiblitz/raspiblitz #215: Add Spruned as service option

fluidvoice, 2018-12-25:
Allow to choose between installing a full-sync-node or a partial-sync-node via Spruned: https://github.com/gdassori/spruned
This allows running a node cheaply only on a small SD card without the need for a big HD/SSD.
And of course it syncs in a very short amount of time.

rootzoll, 2019-01-03:
@fluidvoice You know thats another if/else in the code, right - more Spaghetti ;)

But seriously - I had spruned in mind before. Some say its working, some say there are problems. I am not sure if its recommended to run with LND yet.

Also when thinking into that direction there is also this simple wrapper from jonas maybe:
https://gist.github.com/jonasnick/e4a2b41215a35a461099a5ca2d8affc4

It would be great if someone will run some longer test cases with a bitcoind wrapper like of solution on the raspiblitz ... it those tests run fine and its not limiting all the above services, that would be great.

fluidvoice, 2019-01-03:
I think Spruned runs on top of c-lightning not lnd, no?

rootzoll, 2019-01-07:
spruned was talked most about in the c-lightning context, but just from the concept it should "simulate" a bitcoind non-spruned backend .. that may also work with LND as a replacement for full blockchain bitcoind. I just heard that some people were experimenting with it. If someone could try it on a RaspiBlitz/LND and confirm that it is running stable it could be an option.

geco91, 2019-02-26:
recommend to relable this as RESEARCH

frennkie, 2020-08-08:
out-of-scope - closing.

closed 2020-08-08 by frennkie