raspiblitz/raspiblitz #216: Dynamic DNS not asking for domain name

ElectricTiger, 2018-12-26:
When I go to "Services" then "DynamicDNS",  it just pumps out this info and goes back to the main screen.
[image]

Can you think of anything obvious that I could have done wrong that would cause it to skip past asking me for the Domain name?

ElectricTiger, 2018-12-26:
I closed my issue. I think it was my old router / my fault. It only supports www.dyndns.org and I don't want to use that service.

closed 2018-12-26 by ElectricTiger

rootzoll, 2019-01-09:
Someone else had the a similar problem - did you used the spacebar to "check" the "DynamicDNS" Option before hitting OK/Enter?