raspiblitz/raspiblitz #217: bitcoind not listening on RPC port 8332 in some cases

goodc0re, 2018-12-26:
bitcoind is not listening on RPC port 8332 in some cases 
Adding rpcport = 8332 to bitcoin.conf fixes this issue.

fluidvoice, 2018-12-26:
in what cases?  Any idea what could be causing a port change? I thought 8332 is the default.

goodc0re, 2018-12-26:
The node was moved from a LAN IP to a public WAN IP without changing any settings whatsoever. Upon reboot RPC did not listen on port 8332 anymore. 

Adding rpcport=8332 alone did also not help, now I added: 

rpcport=8332
rpcallowip=127.0.0.1
rpcbind=127.0.0.1:8332

And only now is it listening on the port again.

fluidvoice, 2018-12-29:
> The node was moved from a LAN IP to a public WAN IP without changing any settings whatsoever. Upon reboot RPC did not listen on port 8332 anymore.
>
> Adding rpcport=8332 alone did also not help, now I [...]

I had the same problem and adding these 3 lines seems to have fixed it. thanks!

rootzoll, 2019-01-10:
OK .. needs final test run with migration and fresh install if scripts/config run without error

rootzoll, 2019-01-17:
fresh install runs OK with new settings. RPC is listening on 8332.
I had no old node to test migration - keep fingers crossed ;)

closing the issue for release.

closed 2019-01-17 by rootzoll