raspiblitz/raspiblitz #219: waiting for d to get ready screen

TidalWavesNode, 2018-12-28:
Had everything setup and upgraded to 0.99 after reboot my screen has been sitting at

Waiting for d to get ready
---> Starting Up
Can take longer if device was off.

its been sitting for 3 straight days now and the bitcoin wallet is synced up to the most recent block and continues to sync normally.

any help would be appreciated!

TIA

RobertSundh, 2018-12-28:
Hi. I had the same. My solution was to make a new sd card image with 0.99... Maybe there is another solution I hope

fluidvoice, 2018-12-28:
> Had everything setup and upgraded to 0.99 after reboot my screen has been sitting at
>
> Waiting for d to get ready
> ---> Starting Up
> Can take longer if device was off.
>
> its been sitting for 3 straight [...]

Paste the output of running ./XXdebugLogs.sh from an SSH shell

GeorgeSpark-Stahl, 2018-12-29:
I'm seeing the same issue - looks like it might be the same as issue #217.  In the logs below I see that it can't bind to port 8332.  Here's the output of ./XXdebugLogs.sh:

```*** RASPIBLITZ LOGS ***
blitzversion: 0.99
chainnetwork: bitcoin / main
 19:06:46 up 0 min,  2 users,  load average: 0.34, 0.09, 0.03

*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Fri 2018-12-28 19:06:13 PST; 33s ago
  Process: 669 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 679 (code=exited, status=1/FAILURE)
   CGroup: /system.slice/bitcoind.service

Dec 28 19:06:13 razblitz systemd[1]: bitcoind.service: Failed with result 'exit-code'.

*** LAST 20 CHAINNETWORK LOGS ***
2018-12-29T03:06:13Z Bitcoin Core version v0.17.0.1 (release build)
2018-12-29T03:06:13Z InitParameterInteraction: parameter interaction: -whitelistforcerelay=1 -> setting -whitelistrelay=1
2018-12-29T03:06:13Z Assuming ancestors of block 0000000000000000002e63058c023a9a1de233554f28c7b21380b6c9003f36a8 have valid signatures.
2018-12-29T03:06:13Z Setting nMinimumChainWork=0000000000000000000000000000000000000000028822fef1c230963535a90d
2018-12-29T03:06:13Z Using the 'standard' SHA256 implementation
2018-12-29T03:06:13Z Default data directory /home/bitcoin/.bitcoin
2018-12-29T03:06:13Z Using data directory /mnt/hdd/bitcoin
2018-12-29T03:06:13Z Using config file /home/bitcoin/.bitcoin/bitcoin.conf
2018-12-29T03:06:13Z Using at most 40 automatic connections (1024 file descriptors available)
2018-12-29T03:06:13Z Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2018-12-29T03:06:13Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2018-12-29T03:06:13Z Using 4 threads for script verification
2018-12-29T03:06:13Z scheduler thread start
2018-12-29T03:06:13Z libevent: getaddrinfo: address family for nodename not supported
2018-12-29T03:06:13Z Binding RPC on address 127.0.0.1 port 8332 failed.
2018-12-29T03:06:13Z Unable to bind any endpoint for RPC server
2018-12-29T03:06:13Z Error: Unable to start HTTP server. See debug log for details.
2018-12-29T03:06:13Z Shutdown: In progress...
2018-12-29T03:06:13Z scheduler thread interrupt
2018-12-29T03:06:13Z Shutdown: done

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2018-12-28 19:06:13 PST; 33s ago
 Main PID: 680 (lnd)
   CGroup: /system.slice/lnd.service
           └─680 /usr/local/bin/lnd --externalip=

Dec 28 19:06:14 razblitz lnd[680]: 2018-12-28 19:06:14.954 [INF] RPCS: password gRPC proxy started at [::]:8080
Dec 28 19:06:14 razblitz lnd[680]: 2018-12-28 19:06:14.954 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of a…g wallet and unlock it.
Hint: Some lines were ellipsized, use -l to show in full.

*** LAST 20 LND LOGS ***
-- Logs begin at Fri 2018-12-28 19:06:01 PST, end at Fri 2018-12-28 19:06:47 PST. --
Dec 28 19:06:13 razblitz systemd[1]: Started LND Lightning Daemon.
Dec 28 19:06:14 razblitz lnd[680]: 2018-12-28 19:06:14.914 [INF] LTND: Version: 0.5.1-beta commit=, build=production, logging=default
Dec 28 19:06:14 razblitz lnd[680]: 2018-12-28 19:06:14.914 [INF] LTND: Active chain: Bitcoin (network=mainnet)
Dec 28 19:06:14 razblitz lnd[680]: 2018-12-28 19:06:14.947 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
Dec 28 19:06:14 razblitz lnd[680]: 2018-12-28 19:06:14.953 [INF] RPCS: password RPC server listening on [::]:10009
Dec 28 19:06:14 razblitz lnd[680]: 2018-12-28 19:06:14.954 [INF] RPCS: password gRPC proxy started at [::]:8080
Dec 28 19:06:14 razblitz lnd[680]: 2018-12-28 19:06:14.954 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.

- RTL is OFF by config```

fluidvoice, 2018-12-29:
Did you apply the 3 line conf file change stated #217 ?

GeorgeSpark-Stahl, 2018-12-30:
I did - no luck.   It looks like bitcoin may be starting too early in the boot process. The same error is described here:  https://github.com/bitcoin/bitcoin/issues/7701

That tracks with my debugging - restarting the bitcoin service manually after boot seems to work fine.

hcarrega, 2018-12-30:
Yesterday after a small hiccup I have to reboot after that lightning stop working I read your help here and right now it’s working fine again

rootzoll, 2019-01-02:
@GeorgeSpark-Stahl the bitcoind gets started by systemd when everything else is ready ... and if it is not able to start it will try to restart it again. How do you manually restart it to get it to work in your situation?

GeorgeSpark-Stahl, 2019-01-02:
Maybe I was just impatient then?  That could be.  I was using the command `sudo systemctl restart bitcoind` to restart the service manually, but maybe I should have just waited.  I'll try that.

ErwanLetard, 2019-01-02:
Same problem here. Try to follow a guide to link Shango app to my node (https://github.com/neogeno/shango-lightning-wallet) and since then i m blocked on this screen. I try to come back to the previous lnd.conf (as i deleted .conf and .cert files as stated in the guide) but it do not work.

fluidvoice, 2019-01-02:
> Same problem here. Try to follow a guide to link Shango app to my node (https://github.com/neogeno/shango-lightning-wallet) and since then i m blocked on this screen. I try to come back to the previous lnd.conf (as i deleted .conf and .cert files as stated in the guide) but it do not work. Any ideas ?

What is your output of running:
`./XXdebugLogs.sh`
and
`tail -n 100 /mnt/hdd/bitcoin/debug.log`

ErwanLetard, 2019-01-02:
For ./XXdebugLog.sh :
*** RASPIBLITZ LOGS ***
blitzversion: 0.99
chainnetwork: bitcoin / main
 19:44:33 up 1 min,  2 users,  load average: 1.34, 0.45, 0.16

*** CHAINNETWORK SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-01-02 19:43:13 GMT; 1min 20s ago
  Process: 672 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 683 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─683 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin…id

Jan 02 19:43:12 Gabinou systemd[1]: Starting Bitcoin daemon...
Jan 02 19:43:13 Gabinou systemd[1]: Started Bitcoin daemon.

*** LAST 20 CHAINNETWORK LOGS ***
2019-01-02T19:43:13Z Cache configuration:
2019-01-02T19:43:13Z * Using 2.0MiB for block index database
2019-01-02T19:43:13Z * Using 12.2MiB for transaction index database
2019-01-02T19:43:13Z * Using 8.0MiB for chain state database
2019-01-02T19:43:13Z * Using 77.8MiB for in-memory UTXO set (plus up to 47.7MiB of unused mempool space)
2019-01-02T19:43:13Z init message: Loading block index...
2019-01-02T19:43:13Z Opening LevelDB in /mnt/hdd/bitcoin/blocks/index
2019-01-02T19:43:13Z Opened LevelDB successfully
2019-01-02T19:43:13Z Using obfuscation key for /mnt/hdd/bitcoin/blocks/index: 0000000000000000
2019-01-02T19:43:37Z LoadBlockIndexDB: last block file = 1482
2019-01-02T19:43:37Z LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=66, size=67323732, heights=556665...556730, time=2019-01-02...2019-01-02)
2019-01-02T19:43:37Z Checking all blk files are present...
2019-01-02T19:43:38Z Opening LevelDB in /mnt/hdd/bitcoin/chainstate
2019-01-02T19:43:48Z Opened LevelDB successfully
2019-01-02T19:43:48Z Using obfuscation key for /mnt/hdd/bitcoin/chainstate: eadc35e843c5834c
2019-01-02T19:43:49Z Loaded best chain: hashBestChain=0000000000000000000de6e814d0f09c74edf5820b4182adaf8dce704100dff5 height=556730 date=2019-01-02T19:33:46Z progress=0.999996
2019-01-02T19:43:49Z init message: Rewinding blocks...
2019-01-02T19:43:57Z init message: Verifying blocks...
2019-01-02T19:43:57Z Verifying last 6 blocks at level 3
2019-01-02T19:43:57Z [0%]...[16%]...
*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-01-02 19:43:13 GMT; 1min 20s ago
 Main PID: 684 (lnd)
   CGroup: /system.slice/lnd.service
           └─684 /usr/local/bin/lnd --externalip=

Jan 02 19:43:14 Gabinou lnd[684]: 2019-01-02 19:43:14.941 [INF] RPCS: passw…0009
Jan 02 19:43:14 Gabinou lnd[684]: 2019-01-02 19:43:14.941 [INF] LTND: Waiti… it.
Hint: Some lines were ellipsized, use -l to show in full.

*** LAST 20 LND LOGS ***
-- Logs begin at Wed 2019-01-02 19:43:03 GMT, end at Wed 2019-01-02 19:44:34 GMT. --
Jan 02 19:43:13 Gabinou systemd[1]: Started LND Lightning Daemon.
Jan 02 19:43:14 Gabinou lnd[684]: 2019-01-02 19:43:14.730 [INF] LTND: Version: 0.5.1-beta commit=, build=production, logging=default
Jan 02 19:43:14 Gabinou lnd[684]: 2019-01-02 19:43:14.737 [INF] LTND: Active chain: Bitcoin (network=mainnet)
Jan 02 19:43:14 Gabinou lnd[684]: 2019-01-02 19:43:14.903 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
Jan 02 19:43:14 Gabinou lnd[684]: 2019-01-02 19:43:14.941 [INF] RPCS: password gRPC proxy started at [::]:8080
Jan 02 19:43:14 Gabinou lnd[684]: 2019-01-02 19:43:14.941 [INF] RPCS: password RPC server listening on [::]:10009
Jan 02 19:43:14 Gabinou lnd[684]: 2019-01-02 19:43:14.941 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.

- RTL is OFF by config

fluidvoice, 2019-01-02:
That looks OK. What does this show:
`sudo tail -n 100 /mnt/hdd/bitcoin/debug.log`

ErwanLetard, 2019-01-02:
Ok, just make it through. My node seems to be running but when I want to 'listchannels' or other lncli command I got : 
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: authentication handshake failed: x509: certificate signed by unknown authority

ErwanLetard, 2019-01-02:
I added : 
rpcport=8332
rpcallowip=127.0.0.1
rpcbind=127.0.0.1:8332

in the .conf but when I run 'lnd' i got : /home/admin/.lnd/lnd.conf:10: unknown option: rpcport
any idea ?

fluidvoice, 2019-01-03:
> I added :
> rpcport=8332
> rpcallowip=127.0.0.1
> rpcbind=127.0.0.1:8332
> 
> in the .conf but when I run 'lnd' i got : /home/admin/.lnd/lnd.conf:10: unknown option: rpcport
> any idea ?

You put those lines in the **lnd.conf** file?  The belong in `/mnt/hdd/bitcoin/bitcoin.conf`

ErwanLetard, 2019-01-03:
oups, ok. made the change, but I still got the same errors. 
I don't have access to my machine right now. I'll try new stuff later.
Thank you for the help

fluidvoice, 2019-01-03:
> Ok, just make it through. My node seems to be running but when I want to 'listchannels' or other lncli command I got :
> [lncli] rpc error: code = Unavailable desc = all SubConns are in [...]

This problem was found here and resolved: https://github.com/rootzoll/raspiblitz/issues/35

ErwanLetard, 2019-01-03:
Hello !
Thank you, I tried 'su bitcoin' and when I do 'lnd' i got that : 
bitcoin@Gabinou:/home/admin $ lnd
2019-01-03 17:08:54.868 [INF] LTND: Version: 0.5.1-beta commit=, build=production, logging=default
2019-01-03 17:08:54.868 [INF] LTND: Active chain: Bitcoin (network=mainnet)

and nothing happen (i no longer have the hand on the shell) even ctrl+C doesnt give me the hand back.

when I run 'lnd' from "admin" not "bitcoin" I got : 
admin@Gabinou:~/.lnd $ lnd
2019-01-03 17:05:36.578 [INF] LTND: Version: 0.5.1-beta commit=, build=production, logging=default
2019-01-03 17:05:36.578 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2019-01-03 17:05:36.579 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
2019-01-03 17:05:36.583 [ERR] LTND: password RPC server unable to listen on 0.0.0.0:10009
2019-01-03 17:05:36.583 [INF] LTND: Shutdown complete
listen tcp 0.0.0.0:10009: bind: address already in use

The wierdest is that my node seems to be running but I can't do anything.

ErwanLetard, 2019-01-03:
Ok, I think I just broke all my lnd.conf file. Can someone send some "clean" .conf file ?

fluidvoice, 2019-01-03:
> Ok, I think I just broke all my lnd.conf file. Can someone send some "clean" .conf file ?

```
# lnd configuration

[Application Options]
debuglevel=debug
restlisten=0.0.0.0:8080
tlsextraip=0.0.0.0
maxpendingchannels=5
alias=thundahub
color=#68F442
nat=false

# RPC open to all connections on Port 10009
rpclisten=0.0.0.0:10009
# Domain, could use https://freedns.afraid.org
#tlsextradomain=lightning.yourhost.com  

[Bitcoin]
bitcoin.active=1
bitcoin.node=bitcoind
# enable either testnet or mainnet
#bitcoin.testnet=1
bitcoin.mainnet=1

[Bitcoind]
bitcoind.rpcuser=raspibolt
bitcoind.rpcpass=satnak
bitcoind.rpchost=127.0.0.1
bitcoind.zmqpubrawblock=tcp://*:28332
bitcoind.zmqpubrawtx=tcp://*:28333

[autopilot]
autopilot.active=0
autopilot.maxchannels=5
autopilot.allocation=0.6
```

fluidvoice, 2019-01-03:
> Hello !
> Thank you, I tried 'su bitcoin' and when I do 'lnd' i got that :
> bitcoin@Gabinou:/home/admin $ lnd
> 2019-01-03 17:08:54.868 [INF] LTND: Version: 0.5.1-beta commit=, build=production, [...]

You shouldn't have to do these things by hand. The scripts should do it for you when you reboot/startup.
You could try running the LND setup script:
`./70initLND.sh`

ErwanLetard, 2019-01-03:
ok, I just change my .conf with yours, and ran ./70initLND.sh
and i got that at the end : 
(30/33) still waiting ..
(31/33) still waiting ..
(32/33) still waiting ..
(33/33) still waiting ..
*** TAKES LONGER THEN EXCEPTED ***
1546540541
result()
error(./70initLND.sh: line 30: -cli: command not found)
tail: cannot open '/mnt/hdd//debug.log' for reading: No such file or directory
If you see an error -28 relax, just give it some time.
Waiting 1 minute and then trying again ...

fluidvoice, 2019-01-03:
this is prob caused by your **raspiblitz.info** file does not have correct content.
did you try rebooting?  I think the boot script and other scripts automatically sets this up on every boot.

fluidvoice, 2019-01-03:
You could try putting these lines in raspiblitz.info and running 70initLND.sh again:

```
state=
network=bitcoin
chain=main
message=
setupStep=0
```

ErwanLetard, 2019-01-03:
Anywhere in the file ?

ErwanLetard, 2019-01-03:
Ok I did it again because i noticed that my raspiblitz.info haad change to its former form.
here is what I get : 
admin@Gabinou:~ $ sudo nano raspiblitz.info
admin@Gabinou:~ $ ./70initLND.sh

*** Checking bitcoin ***
OK - bitcoind is running

*** Wait until bitcoind is ready ...
OK - chainnetwork is working

*** LND Config ***
OK - exists

*** Starting LND ***
OK - LND is running

walletExists(1)
--> lets wait 60 seconds for LND to get ready

then : 
[image]

So I restarted and got : 

[image]
then
[image]
then
[image]
then
[image]

And here it loops, asking to unlock my wallet again without incrementing the progress. I finnally go through after unlocking my wallet 3-4 times : 
[image]

And the error I got is still there ...

fluidvoice, 2019-01-03:
Cool. that is good progress.

fluidvoice, 2019-01-03:
That info screen shows you have channels open and 3382710 satoshis in there.

ErwanLetard, 2019-01-03:
I do.

ErwanLetard, 2019-01-03:
My node worked perfectly well until I tried to link it with Shango. 
And I don't understand what I broke following this tutorial : https://github.com/neogeno/shango-lightning-wallet
But it do not work well at all now (needless to say that the connexion with shango was a fail)

fluidvoice, 2019-01-03:
What do you see when you run from the SSH terminal:
`lnbalance bitcoin`

fluidvoice, 2019-01-03:
well from that info screen, it looks at least like your node is up and running correctly, so that's good progress. I don't know enough about how the channel stuff works yet, so cannot help much there.

ErwanLetard, 2019-01-03:
Here is what I see

[image]

fluidvoice, 2019-01-03:
You actually have TOR working which is more than I do.  I tried enabling it but it didn't work first time for me so I reverted to non-TOR.  It's not important for me right now.

fluidvoice, 2019-01-03:
and if you run `./00infoBlitz.sh` do you see the same info screen as you posted above?

ErwanLetard, 2019-01-03:
Here we are :
[image]

fluidvoice, 2019-01-03:
Damn. I think that last error "line 70" is due to your raspiblitz.info file still not correct.
If I run $ `lnbalance +` I get the same error and all values zero...

> /usr/local/bin/lnbalance: line 70: +  : syntax error: operand expected (error token is "+  ")

ErwanLetard, 2019-01-03:
for what reason my raspiblitz.info is back to : 
[image]

fluidvoice, 2019-01-03:
My node is:
038d3eb5339f78774697448ce2414b429b41edd8f65dcc2edc33e591001476c948@179.125.20.7:9735
how do I set the minimum channel sats?

ErwanLetard, 2019-01-03:
huuum
[image]

ErwanLetard, 2019-01-03:
always the same error on lncli ...

fluidvoice, 2019-01-03:
> always the same error on lncli ...

`sudo tail -n 50 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log`

ErwanLetard, 2019-01-03:
admin@Gabinou:~ $ sudo tail -n 50 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
2019-01-03 20:40:37.932 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=606455429588254721, flag=2, update_time=2019-01-03 20:40:01 +0000 GMT) from 138.68.14.104:9735
2019-01-03 20:40:37.933 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=596669776075292672, flag=3, update_time=2019-01-03 20:40:15 +0000 GMT) from 138.68.14.104:9735
2019-01-03 20:40:37.933 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=606376264783101953, flag=258, update_time=2019-01-03 20:39:49 +0000 GMT) from 138.68.14.104:9735
2019-01-03 20:40:37.934 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=606376264783101953, flag=259, update_time=2019-01-03 20:39:52 +0000 GMT) from 138.68.14.104:9735
2019-01-03 20:40:37.934 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=605060149341585408, flag=1, update_time=2019-01-03 20:40:10 +0000 GMT) from 138.68.14.104:9735
2019-01-03 20:40:37.934 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=612134407107903489, flag=0, update_time=2019-01-03 20:40:09 +0000 GMT) from 138.68.14.104:9735
2019-01-03 20:40:37.934 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=605733050395066368, flag=1, update_time=2019-01-03 20:40:15 +0000 GMT) from 138.68.14.104:9735
2019-01-03 20:40:37.935 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=610558807011164160, flag=0, update_time=2019-01-03 20:40:10 +0000 GMT) from 138.68.14.104:9735
2019-01-03 20:40:37.935 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=605108527812444160, flag=1, update_time=2019-01-03 20:40:10 +0000 GMT) from 138.68.14.104:9735
2019-01-03 20:40:40.425 [DBG] RPCS: [walletbalance] Total balance=0.01859908 BTC
2019-01-03 20:40:40.514 [DBG] RPCS: [walletbalance] Total balance=0.01859908 BTC
2019-01-03 20:40:40.839 [INF] RPCS: [listchannels] fetched 5 channels from DB
2019-01-03 20:40:44.750 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=608498322163105792, flag=3, update_time=2019-01-03 19:46:26 +0000 GMT) from 74.108.13.152:9735
2019-01-03 20:40:44.759 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=600049674878713857, flag=2, update_time=2019-01-03 19:46:19 +0000 GMT) from 74.108.13.152:9735
2019-01-03 20:40:44.763 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=605060149341585408, flag=1, update_time=2019-01-03 20:40:10 +0000 GMT) from 74.108.13.152:9735
2019-01-03 20:40:44.766 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=608495023714861056, flag=2, update_time=2019-01-03 20:40:00 +0000 GMT) from 74.108.13.152:9735
2019-01-03 20:40:44.768 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=605108527812444160, flag=1, update_time=2019-01-03 20:40:10 +0000 GMT) from 74.108.13.152:9735
2019-01-03 20:40:44.841 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=603179984429711361, flag=256, update_time=2019-01-03 20:39:53 +0000 GMT) from 74.108.13.152:9735
2019-01-03 20:40:44.842 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=608234439447740416, flag=3, update_time=2019-01-03 20:39:57 +0000 GMT) from 74.108.13.152:9735
2019-01-03 20:40:44.842 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=608695134822596609, flag=2, update_time=2019-01-03 19:46:31 +0000 GMT) from 74.108.13.152:9735
2019-01-03 20:40:44.843 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=612177288038973440, flag=0, update_time=2019-01-03 20:39:51 +0000 GMT) from 74.108.13.152:9735
2019-01-03 20:40:44.843 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=608774299582922752, flag=0, update_time=2019-01-03 20:39:54 +0000 GMT) from 74.108.13.152:9735
2019-01-03 20:40:44.844 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=605733050395066368, flag=1, update_time=2019-01-03 20:40:15 +0000 GMT) from 74.108.13.152:9735
2019-01-03 20:40:44.844 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=610558807010639872, flag=0, update_time=2019-01-03 20:40:10 +0000 GMT) from 74.108.13.152:9735
[... cut ...] 7.77.11:9735
2019-01-03 20:40:50.676 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=608498322163105792, flag=3, update_time=2019-01-03 19:46:26 +0000 GMT) from 23.237.77.11:9735
2019-01-03 20:40:50.676 [DBG] PEER: Received ChannelUpdate(chain_hash=000000000019d6689c085ae165831e934ff763ae46a2a6c172b3f1b60a8ce26f, short_chan_id=608695134822596609, flag=2, update_time=2019-01-03 19:46:31 +0000 GMT) from 23.237.77.11:9735

fluidvoice, 2019-01-03:
That "huuum" output, did that come from using the menu system to connect or are you running from the command line?

ErwanLetard, 2019-01-03:
it is just a line I type in git up answer before pasting the screeshot :)

fluidvoice, 2019-01-03:
but the screen shot is the result of selecting a menu option or from running something in the shell?

fluidvoice, 2019-01-03:
what do you see if you run:
`sudo -u bitcoin lnbalance bitcoin`
is it still all zeros with errors?

ErwanLetard, 2019-01-03:
[image]

ErwanLetard, 2019-01-03:
works with bitcoin but not admin ...

fluidvoice, 2019-01-03:
yep, that's what I thought.  This is what they were talking about here:
https://github.com/rootzoll/raspiblitz/issues/35#issuecomment-411467669

ErwanLetard, 2019-01-03:
I'm not sure to understand everything in this. I tried to copy the cert and tls from btc to admin earlier, didn't seems to work ..

fluidvoice, 2019-01-03:
according to #35 you don't need to do that.  Just run as user (bitcoin). The question is why the scripts are not running as user (bitcoin) as they should be.

ErwanLetard, 2019-01-03:
And once i m in shell mode, is there a command to come back to the 'menu mode' (with blue back etc ..)

ErwanLetard, 2019-01-03:
Still can't connect to your node :
[image]

fluidvoice, 2019-01-03:
try
`sudo -u bitcoin lncli connect ...`

ErwanLetard, 2019-01-03:
I got that : 
bitcoin@Gabinou:~ $ sudo -u bitcoin lncli connect 038d3eb5339f78774697448ce2414b429b41edd8f65dcc2edc33e591001476c948@179.125.20.7:9735
bitcoin is not in the sudoers file.  This incident will be reported.

fluidvoice, 2019-01-03:
and these:
sudo -u bitcoin lncli --chain=bitcoin --network=mainnet connect ...
sudo -u admin lncli --chain=bitcoin --network=mainnet connect ...

ErwanLetard, 2019-01-03:
All three give me the same mistake ...
bitcoin is not in the sudoers file. This incident will be reported.

fluidvoice, 2019-01-03:
> All three give me the same mistake ...
> bitcoin is not in the sudoers file. This incident will be reported.

then how did you get that "connection refused" screenshot then?

ErwanLetard, 2019-01-03:
just with : 
bitcoin@Gabinou:~ $ lncli connect 038d3eb5339f78774697448ce2414b429b41edd8f65dcc2edc33e591001476c948@179.125.20.7:9735
[lncli] rpc error: code = Unknown desc = dial tcp 179.125.20.7:9735: connect: connection refused

[16 more replies]

closed 2019-01-09 by rootzoll