raspiblitz/raspiblitz #222: ZapDesktop connect to Node

DaReaLDeviL, 2018-12-29:
The new App for IOs is working like it should with the newest version of Raspiblitz.

Now i try it with ZapDesktop Version 0.3.2beta to Rapsiblitz 0.99 but have some problems to give it the right TLS-Certificate and Macaroon path. Can someone please point me in the right direction?

Thank you in advanced!

hcarrega, 2018-12-29:
I can’t even take iOS app working :) I think e need to read more about lnd , but everything it’s working on RaspiBlitz

DaReaLDeviL, 2018-12-29:
Where can I assist you on the iOS app? Come into the IRC of freenode #raspiblitz I will try to help you.

hcarrega, 2018-12-29:
i think the problem is easy nothing is setup on lnd.conf on IP's

DaReaLDeviL, 2018-12-29:
Maybe you can test the desktop app and help me on there?

DaReaLDeviL, 2018-12-30:
Ok now I get it. You have to copy the tls.cert and admin.macaroon from the raspiblitz to your local pc and give this folder in your zapdesktop. dont forget to point to the right port and now it works!

fluidvoice, 2018-12-30:
I sure hope they improve the ease of use of Lightning in the future. This stuff is impossible for the general public.

closed 2018-12-30 by DaReaLDeviL

wuestenigel, 2019-01-07:
Hey,
where do I find the admin.macaroon file and where do I find my ip:port settings to connect with?

zer0trst, 2019-01-07:
@wuestenigel 

Using filezilla, connect via SFTP into your raspi and copy the following files to your pc, point to these two files where it asks you in Zap.

      /home/admin/.lnd/tls.cert

      /home/admin/.lnd/data/chain/bitcoin/mainnet/admin.macaroon

wuestenigel, 2019-01-07:
Great, what is username and PW for connecting with SFTP?

What will be the URL and  port I have to enter later in Zap?

zer0trst, 2019-01-07:
Connect to sftp:// local IP of your node:21 user name Admin and the same PW you set for admin when setting up the node

If you’re only connecting zap locally then use the same local ip of your node in zap with port 10009

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

wuestenigel, 2019-01-07:
Instantly worked! So much appreciated!

wuestenigel, 2019-01-07:
Any ideas what would be the REST API / Node URL in Joule chrome Extension?

Basically I think I need the port.