raspiblitz/raspiblitz #227: Finally get rid of txindex?

rootzoll, 2018-12-31:
Check:
https://twitter.com/Stadicus3000/status/1079677441956868097

And for future Electrum Server:
https://twitter.com/Stadicus3000/status/1079677444032995328

If everything works without txindex, maybe also change torrents to:
https://getbitcoinblockchain.com

geco91, 2019-02-26:
would recommend to check this again for v1.1 .. RaspiBolt has now also removed the txindex .. see:
https://github.com/Stadicus/guides/pull/325

Ask emzy if he can also test a bitcoind with electrum server and give feedback on.

CommanderPoe, 2019-03-01:
> would recommend to check this again for v1.1 .. RaspiBolt has now also removed the txindex .. see:
> [Stadicus/guides#325](https://github.com/Stadicus/guides/pull/325)
> 
> Ask emzy if he can also test a bitcoind with electrum server and give feedback on.

i was indeed running bitcoind without txindex on the pi when i first used the raspibolt.

Emzy, 2019-03-03:
I have my RaspiBlitz all the time with txindex=0 running. No issues so far.
electrs and Electrum Personal Server don't need it. Only the Electrumx would need txindex=1

vmavromatis, 2019-03-04:
I'm fully synced with txindex=1. Do I gain something by switching to txindex=0?

Emzy, 2019-03-04:
Faster sync time and less disk space. 
The txindex is 19GB in size. But I have no measure how much faster it would be without txindex.

rootzoll, 2019-06-18:
Testing with torrents from getbitcoinblockchain.com .. they download fast (good seeding), lets see if this works.

rootzoll, 2019-06-23:
OK getbitcoinblockchain.com sources seem to work in testing

TODO: before final v1.3 package use latest files from that page or add realtime download of latest torrents

rootzoll, 2019-06-28:
Ah shoot - seeing just now that getbitcoinblockchain.com stopped updating torrent sources last month. OK so ...

Create new torrents for blockchain without index switched on:
TODO: New Torrents for Bitcoin with mainnet/testnet
TODO: New Torrents for Litecoin with mainnet

rootzoll, 2019-07-01:
OK - tests with new torrents were good. 
Requested help for good seeding:
https://github.com/rootzoll/raspiblitz/issues/285#issuecomment-507224479

rootzoll, 2019-07-01:
OK. See new torrents are working and seeding is getting good. txindex=0 is now the new default.

closed 2019-07-01 by rootzoll