raspiblitz/raspiblitz #228: linux noob here hdd not mounting

adz4u28, 2018-12-31:
i already have an external usb hdd that works perfectly fine in raspibolt, i formatted it etc for raspiblitz, but when booting it refuses acknowledge it exists. however i can traverse it in terminal, make files etc. any idea why it wont see it on boot? i have it listed on fstab but for the life of me i cant get the raspibolt initial setup to recognise it :( could you point me in the right direction? i even renamed it BLOCKCHAIN to no avail. its mounted fine but im lost and any assistance would be greatly appreciated.

fluidvoice, 2019-01-01:
Almost everything is done with the various scripts in the /home/admin folder. Inspect the 30initHDD.sh and 40addHDD.sh scripts and see what they do. But these get run by 10setupBlitz.sh which itself gets run by 00mainMenu.sh when you start setup or SSH into your device and choose "Bitcoin" to setup. So that should setup your HDD correctly.

fluidvoice, 2019-01-01:
a somewhat similar problem here. Might help to take a look: https://github.com/rootzoll/raspiblitz/issues/233

adz4u28, 2019-01-01:
yeah all that happens for me is:
admin@RaspiBlitz:~ $ ./00mainMenu.sh
Starting the main menu ...
***********************************************************
WARNING: NO HDD FOUND -> Shutdown, connect HDD and restart.
***********************************************************
i can write to the hd, i can list it,copy to it, rename etc  you name it i can do it. except this will not see it and im really stumped. 
like i said, im a linux noob, but looking through the script 30initHDD.sh shows me its looking for sda1 or sda2 and my external drive is called sda ?

fluidvoice, 2019-01-01:
> NO HDD FOUND

If I grep for "WARNING:" I don't find this warning message anywhere.  What script has this message?
Are you sure you're running the latest raspiblitz release?

adz4u28, 2019-01-01:
100% i literally downloaded it yesterday and wrote it to a card :(

fluidvoice, 2019-01-01:
we just need to figure out why sda1 does not exist for you. For me...

$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0 465.8G  0 disk 
└─sda1        8:1    0 465.8G  0 part /mnt/hdd
mmcblk0     179:0    0  29.8G  0 disk 
├─mmcblk0p1 179:1    0  43.8M  0 part /boot
└─mmcblk0p2 179:2    0  29.8G  0 part /

fluidvoice, 2019-01-01:
It's def looking for sda1 in the bootstrap script: https://github.com/rootzoll/raspiblitz/blob/ed748e433ff3548da05530d97756143e5b4f5991/home.admin/_bootstrap.sh#L75

fluidvoice, 2019-01-01:
does your /home/admin/raspiblitz.log show anything interesting?

adz4u28, 2019-01-01:
***********************************************
Running RaspiBlitz Bootstrap 0.99
Mon 31 Dec 18:23:30 GMT 2018
***********************************************
:(

fluidvoice, 2019-01-01:
The problem may be that you manually did some things with fstab and/or creating other things. The scripts depend on the state of fstab. Dont' do anything manually. I would reformat EXT4 and reboot letting the scripts do all the init/setup.

adz4u28, 2019-01-01:
yeah gives me an error saying sda1 doesnt exist, removed the entry in fstab, made no difference. still says no hd 

how do i make my external usb disk sda1 instead of sda ? using the cmd above wont work.

fluidvoice, 2019-01-01:
> yeah gives me an error saying sda1 doesnt exist, removed the entry in fstab, made no difference. still says no hd
> 
> how do i make my external usb disk sda1 instead of sda ? using the cmd above wont work.

This doesn't work?:
`sudo mkfs.ext4 /dev/sda1 -F -L BLOCKCHAIN`

adz4u28, 2019-01-01:
okay, sorted it. i fdisked it and removed the lot. rebooted now working. wow.. that was a very intense introduction to linux... haha
now finally i have the bitcoin menu up... wow... :) 
thankyou so much for your assistance!

fluidvoice, 2019-01-01:
cool. I'm surprised it actually needed to be re-partitioned and not just reformatted with that `mkfs` command I gave above.  Is that true?

adz4u28, 2019-01-01:
no the cmd you posted above wouldnt work because it couldnt find sda1. hence why i used fdisk to redo it.

fluidvoice, 2019-01-01:
what was the fdisk command line you used? for the sake of others who might have same problem.

adz4u28, 2019-01-01:
sudo fdisk /dev/sda
accept defaults then
w to write it out.
sudo fdisk -l /dev/sda to check

closed 2019-01-01 by adz4u28

fluidvoice, 2019-01-01:
It's funny because nowhere in the scripts or the README does it talk about partitioning or `fdisk`
This command is not run anywhere.

frennkie, 2019-01-11:
@rootzoll As you pointed out `sda1` is the first partition on the first disk, If there is no `sda1` then most likely reason is that there are no partitions at all...