raspiblitz/raspiblitz #230: Add bluewallet Support

rootzoll, 2018-12-31:
https://bluewallet.io

and check their LndHub wrapper:
https://github.com/BlueWallet/LndHub

Could one Blitz host multipüle accounts this way?

rootzoll, 2019-01-07:
just a reminder: https://lightning-wallet.com .. seems like a standalone, but to keep on mind.

openoms, 2019-02-17:
Testing the guide on this one: https://github.com/dangeross/guides/blob/master/raspibolt/raspibolt_6B_lndhub.md

openoms, 2019-02-17:
got stuck on this already:
```admin@openomsRaspiBlitz:~/download/redis-5.0.3 $ sudo systemctl start redis
admin@openomsRaspiBlitz:~/download/redis-5.0.3 $ sudo systemctl status redis
● redis.service - Redis In-Memory Data Store
   Loaded: loaded (/etc/systemd/system/redis.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2019-02-17 08:32:53 GMT; 4s ago
  Process: 4961 ExecStop=/usr/local/bin/redis-cli shutdown (code=exited, status=203/EXEC)
  Process: 4959 ExecStart=/usr/local/bin/redis-server /etc/redis/redis.conf (code=exited, status=203/EX
 Main PID: 4959 (code=exited, status=203/EXEC)

Feb 17 08:32:53 openomsRaspiBlitz systemd[1]: redis.service: Unit entered failed state.
Feb 17 08:32:53 openomsRaspiBlitz systemd[1]: redis.service: Failed with result 'exit-code'.
Feb 17 08:32:53 openomsRaspiBlitz systemd[1]: redis.service: Service hold-off time over, scheduling res
Feb 17 08:32:53 openomsRaspiBlitz systemd[1]: Stopped Redis In-Memory Data Store.
Feb 17 08:32:53 openomsRaspiBlitz systemd[1]: redis.service: Start request repeated too quickly.
Feb 17 08:32:53 openomsRaspiBlitz systemd[1]: Failed to start Redis In-Memory Data Store.
Feb 17 08:32:53 openomsRaspiBlitz systemd[1]: redis.service: Unit entered failed state.
Feb 17 08:32:53 openomsRaspiBlitz systemd[1]: redis.service: Failed with result 'exit-code'.```

geco91, 2019-02-26:
recommend to give this a shot for v1.1 - but when takes too long to delay this feature

rootzoll, 2019-03-04:
Recheck on the guide for v1.1 release: https://twitter.com/bluewalletio/status/1096460638799450113

rootzoll, 2019-03-07:
Also take a look at the pebble wallet while on mobile wallet integration.
https://github.com/IndieSquare/pebblewallet

openoms, 2019-03-22:
I am very reluctant to use the Bluewallet until they stop sending the xpubs to blockchain.info. https://twitter.com/richblov8s/status/1108883868399095810?s=19

openoms, 2019-11-21:
see https://github.com/rootzoll/raspiblitz/issues/810 for developments

openoms, 2019-11-21:
> Also take a look at the pebble wallet while on mobile wallet integration.
> https://github.com/IndieSquare/pebblewallet

Pebble is now discontinued. Some of the team is working on the Nayuta wallet, which is promising, but not open source yet. I tried and paired with my Blitz successfully.

openoms, 2020-03-04:
Looked at BlueWallet again, but I don't feel it being ready for recommendation/implementation:
* thankfully instead of blockchain.info the wallet connects to Electrum servers. Possible to use a custom server, but only with TCP (unencrypted) and it is not planned to change anytime soon: https://github.com/BlueWallet/BlueWallet/issues/579
[image]

* Communication with LND is over clearnet too:
 https://github.com/BlueWallet/LndHub/issues/33
Despite the screen suggests a https endpoint all tutorials use TCP.
[image]

* positive is that logging seems to be not too intruding:
https://github.com/BlueWallet/BlueWallet/blob/master/analytics.js

ncoelho, 2020-03-28:
Just to give an update on these issues (thanks for pointing them out, sometimes we get to overwhelmed on our own little world...)

Since release [v5.1.0](https://github.com/BlueWallet/BlueWallet/releases/tag/v5.1.0) all calls to 3rd party APIs were removed and all wallet types are connecting to our Electrum Servers. There may be some code leftovers, but this are not being used, the heavy work is done.

TLS support for Electrum was merged to master on https://github.com/BlueWallet/BlueWallet/commit/7b5db85c9b6443f9f39289dc6217025b79b7c646 and it will be released by default on the next store update.

Regarding the LNDhub connection, it supports TLS and the connection is made over SSH. 

Any feedback just let me know. Thanks!

ncoelho, 2020-04-09:
Another update.

https://github.com/BlueWallet/BlueWallet/releases/tag/v5.3.0 introduces TLS for Electrum Personal Server and also for Electrs, we had to disable batching for this specific implementations since they are not following the specifications completely.

INSG2020, 2020-06-03:
@ncoelho Does the LNDhub connection also support TOR?

@openoms With TOR and the above mentioned changes for Electrum, would you then feel it´s ready for implementation on Raspiblitz?

openoms, 2020-06-03:
@INSG2020 my concerns above have been fixed, so would be happy to see LNDhub on the RaspiBlitz as an optional service.

The connection for clients would need en exposed port / remote-ssh-tunnel or Tor2IP service.
Not sure if BlueWallet plays well with Orbot to connect to an LNDHub serving on a .onion.

You can already connect BlueWallet to the Electrum Rust Server Running on the Blitz.
Also there is a browser based LN onboarding service in the form of LNbits.

rootzoll, 2020-10-14:
Just to mention the lastest LNbits (as of v1.6.1 RaspiBlitz release) .. has a LNDhub plugin. Maybe that is already a start for some ...
[image]

nyxnor, 2021-05-25:
I stated a LndHub installation script in https://github.com/rootzoll/raspiblitz/issues/2320, but had problems setting the correct path for nodejs. If someone wants to continue ...

rootzoll, 2021-06-14:
It looks like that Redis will get a default service for the WebUI BackendAPI with v1.8 .. so I push this to v1.8.1 to then take a look again.

nyxnor, 2021-06-19:
redis.service is working, just having trouble with babel lines that copy files. So this is close to be accomplished if someone debugs further

xanoni, 2021-08-02:
I installed this recently on RaspiBolt. Is the script above still the latest? I see a few problems. I can't test it without a RaspiBlitz, but here are the things I would change: 

1. Your script runs as user `admin`, right? Then the `git clone` command shouldn't need a `sudo`
2. The `sudo npm install` command should be run from the `/home/bitcoin/.lndhub` directory, but seems it's run from `/home/admin` instead? It has to move down a few lines in the script.
3. Besides, we can run `sudo -u bitcoin npm install` instead of running it as `root`
4. The command `sudo npm config set prefix '/home/bitcoin/.npm-global'` should be run as user `bitcoin` instead (`sudo -u bitcoin [...]`) ... otherwise it will append the prefix to file `/root/.npmrc` instead of to file `/home/bitcoin/.npmrc`
5. Babel should be installed via `sudo -u bitcoin npm install -g @babel/core @babel/cli` (which will install it into the above-defined npm prefix directory: `/home/bitcoin/.npm-global/`) .... the current script sets `admin`'s `$PATH` to `/home/bitcoin/.npm-global/bin` but then installs `babel` somewhere else, so it won't be found by anyone
6. Once the above is fixed, we can run `sudo -u bitcoin babel ./ --out-dir ./build --copy-files --ignore node_modules` without the `npm run` prefix ... but we need to make sure that `bitcoin`'s `$PATH` includes `/home/bitcoin/.npm-global/bin`
7. Delete this line: `sudo node /home/bitcoin/.lndhub/build/index.js` ... this launches LndHub and shouldn't be part of the build script

xanoni, 2021-08-02:
Regarding the port forwarding:

I decided to only expose LndHub over Tor. BlueWallet supports that, you just enter the Onion link in the app.

Thus, one could delete these two lines:
```bash
sudo ufw allow 3050 comment 'allow LndHub'
sudo ufw reload
```

And instead add a Hidden Service in `/etc/tor/torrc`:
```
# External port 3000, internal port 3050
HiddenServiceDir /var/lib/tor/hs_lndhub_3000/
HiddenServicePort 3000 127.0.0.1:3050
```

nyxnor, 2021-08-02:
Thanks for the corrections. Local IP access needs to be a allowed, a self signed certificate can be used.
You are in the right direction, finish this to claim the bounty.

xanoni, 2021-08-02:
Okay I can finish it this week. The Blitz environment is close enough to the Bolt, so I should be able to test it on a Bolt with some modifications.

jmillionaire, 2021-08-10:
@xanoni , how's it going? i'd love to contribute by adding to the 200k bounty.

it seems like a number of us are trying to execute this "Uncle Jim" BlueWallet/LndHub/LNBits/Raspiblitz plan. i'm not programming savvy, perhaps i should've bought an Umbrel but i am enjoying the struggle at times when things actually work.

xanoni, 2021-08-10:
> @xanoni , how's it going? i'd love to contribute by adding to the 200k bounty.
>
>
>
> it seems like a number of us are trying to execute this "Uncle Jim" BlueWallet/LndHub/LNBits/Raspiblitz plan. i'm [...]

Good that you're responding here because I wasn't sure if anyone still cared about this, and there's always endless other stuff.

It shouldn't be difficult, I already have it running on RaspiBolt. Just needs to be repackaged and tested.

jmillionaire, 2021-08-10:
> Good that you're responding here because I wasn't sure if anyone still cared about this, and there's always endless other stuff.
>
> It shouldn't be difficult, I already have it running on RaspiBolt. [...]

do you have a Blitz to do so?

xanoni, 2021-08-10:
> do you have a Blitz to do so?

Not right now but I know well enough how the Blitz works so I think I can fake one on the Bolt. 

Someone might have to do a proper test for me, though :)

jmillionaire, 2021-08-10:
> Not right now but I know well enough how the Blitz works so I think I can fake one on the Bolt.
>
> Someone might have to do a proper test for me, though :)

i'll volunteer!

xanoni, 2021-08-10:
> i'll volunteer!

Thank you that makes it a lot easier ... my main fear was missing something and frustrating people if there's a bug. I'll try to message you somewhere.

jmillionaire, 2021-08-10:
> Thank you that makes it a lot easier ... my main fear was missing something and frustrating people if there's a bug. I'll try to message you somewhere.

@jcmillionaire on Reddit.

xanoni, 2021-08-13:
> @jcmillionaire on Reddit.

Not a Redditor but I pinged you on Telegram 2 days ago (saw you in the Blitz channel)... seems you no longer use that?

xanoni, 2021-08-25:
FYI: So I've started putting this together, but ran out of time and need to finish it later ... don't use this in the current form, please. (It's quite close, though. Stay tuned.)

https://github.com/xanoni/raspiblitz/blob/ab22f59399b5e2a554033d2230a7cae87737897f/home.admin/config.scripts/bonus.lndhub.sh

Diff: https://github.com/xanoni/raspiblitz/commit/ab22f59399b5e2a554033d2230a7cae87737897f

xanoni, 2021-08-30:
Okay! thank you I have what I need then

P1xelfehler, 2021-12-06:
So what's the status here? Do you need help @xanoni ?
My linux skills are not enough to really help, but I can test it on my Raspiblitz if you want.

xanoni, 2022-01-10:
Sorry but I'm currently somewhat disconnected from this world here. Life.

seinzfiction, 2022-05-19:
@rootzoll is there some support required on this item? I'd be interested to do some work on this.

openoms, 2022-05-20:
@seinzfiction thank you for the offer. Before working to integrate LNDhub itself check if the the latest LNbits might provide an LNDHub plugin which would make it available for the LND and CLN funding source as well.

seinzfiction, 2022-05-29:
@rootzoll The current LNBits offers a LNDHub extension already. But as far as I understood and experienced it, this extension offers LNDHub for the LNBits users wallets. Via this way it is not possible to offer LNDHub access for the Raspiblitz Lightning wallet directly.

openoms, 2022-05-29:
So multiple users must be handled by LNbits then - they need to create a wallet and get the LNDhub link individually as I undestand
This is indeed less convenient then just having one LNDhub link to share so might justify the standalone LNDhub implementation.

seinzfiction, 2022-05-30:
@openoms you're right. LNBits provides custodial Lightning wallets to 3rd party users and the LNDHub extension in LNBits can be used to access those wallets through Blue Wallet directly.

prusnak, 2022-07-13:
There is a very promising reimplementation of LndHub in Go language here: https://github.com/getAlby/lndhub.go
I recommend to use this as a LndHub/BlueWallet server.

Smiggel, 2022-09-11:
Tried LNbits with LNDHub but Raspiblitz does not save the settings. It loses the wallet and the LNDHub installation. When I create a wallet, and close LNBits, and re-open it again, the wallet is gone.

rootzoll, 2022-09-13:
> Tried LNbits with LNDHub but Raspiblitz does not save the settings. It loses the wallet and the LNDHub installation. When I create a wallet, and close LNBits, and re-open it again, the wallet is gone.

@callebtc can you take a look or patch us to the right LNbits person what persistence error this could be with LNDhub plugin?

jmillionaire, 2022-09-13:
> Tried LNbits with LNDHub but Raspiblitz does not save the settings. It loses the wallet and the LNDHub installation. When I create a wallet, and close LNBits, and re-open it again, the wallet is gone.

I had to bookmark my LNBits on Tor Browser

callebtc, 2022-09-13:
If the wallet is gone (not the extension) it would rather sound like a deeper db persistance problem. But then nothing in LNbits would work.

@Smiggel what do you mean exactly by "the wallet is gone"?

@rootzoll how do I test this without a blitz?

rootzoll, 2022-09-13:
> @rootzoll how do I test this without a blitz?

Maybe @arcbtc can sponsor a RaspiBlitz from FULMO for future testing?

Smiggel, 2022-09-13:
> @Smiggel what do you mean exactly by "the wallet is gone"?

Perhaps I am understanding incorrectly how it should, but here is what I do:

1. Installed LND bits.
2. Installed LND Hub
3. Create a wallet
4. Close the browser
5. Open the browser and browse to LND bits

After point 5 I expect to see my wallet from step 3. However, I don’t see the wallet anymore. When I look at the extensions, I see the “install” button again for LND hub.

I would except the wallet to be there and the extension installed.

arbadacarbaYK, 2022-10-09:
> I would except the wallet to be there and the extension installed.

Did you open the wallet by its URL or just open LNbits ? You either need to save the URL of the wallet or import it to e.g. BlueWallet by using the "Export-via-QR-to-phone" feature, see the popup you click when generating a wallet 😋 Sats are still on your node so nevermind about the lost link.

Smiggel, 2022-10-09:
Thanks, I will give that a try.

cstenglein, 2023-06-07:
BlueWallet doesn't support Lightning anymore

@rootzoll closing this for now

closed 2023-06-07 by cstenglein