raspiblitz/raspiblitz #234: files deleted without warning

fluidvoice, 2019-01-01:
https://github.com/rootzoll/raspiblitz/blob/ed748e433ff3548da05530d97756143e5b4f5991/home.admin/XXupdateScripts.sh#L33

This should not be done - removing all shell scripts in a directory without warning the user.
I had additional scripts for other things running on my node that got deleted.

rootzoll, 2019-01-02:
Ouch OK. Maybe the delete should be a "--clear" option on the script.

The XXupdateScripts is not the recommended for "Update" anyway ... maybe I need to change/rename that script anyway. Its more for syncing with your GitHub during development.

Regular system updates should always be by replacing the complete SD card. Own scripts would need to be placed on the HDD and need some manual reactivation/install after an update.

fluidvoice, 2019-01-02:
redoing the SD card seems overkill to me, unless it's really necessary. One thing I wondered about, which needs upfront documentation visibility, is which upgrade method to choose and that is ensured not to delete the already synced blockchain DB.  In my case running XXupdateScripts and rebooting worked quite well after I fixed one problem with it (which I opened an issue on).