raspiblitz/raspiblitz #235: Zapconnect

eurtnl, 2019-01-01:
OK, so i finally got it up and running. Now i would like to connect Zap. But i get the next error:
*** Setup ***

Installing zapconnect. Please wait...

Getting github.com/LN-Zap/zapconnect (please wait) ...
./97addMobileWalletZap.sh: line 17: cd: /usr/local/gocode/src/github.com/LN-Zap/zapconnect: No such file or directory

and then nothing...:(

eurtnl, 2019-01-01:
Oh wait...now i got this:
*** STEP 2 : Click on Scan (make whole QR code fill camera) ***
./97addMobileWalletZap.sh: line 51: zapconnect: command not found
(To shrink QR code: OSX->CMD- / LINUX-> CTRL-) Press ENTER when finished.

but no qr code

fluidvoice, 2019-01-01:
echo "If its not working - check issues on GitHub:"
echo "https://github.com/LN-Zap/zap-iOS/issues"
echo "https://github.com/LN-Zap/zapconnect/issues"

fluidvoice, 2019-01-01:
the script failed to build zapconnect so nothing is running.

eurtnl, 2019-01-01:
no kidding ;-)
So what tot do now?

fluidvoice, 2019-01-01:
LNDConnect is talked about here: https://medium.com/@JimmyMow/announcing-zap-desktop-0-3-0-beta-7eb3b7df4a8c

eurtnl, 2019-01-01:
ok, did `sudo apt-get install golang` and `go get -d github.com/LN-Zap/lndconnect`
now it says
package github.com/LN-Zap/lndconnect: cannot download, $GOPATH not set. For more details see: go help gopath
should i reboot first?

fluidvoice, 2019-01-01:
try running these first:
```
export GOROOT=/usr/local/go
export PATH=$PATH:$GOROOT/bin
export GOPATH=/usr/local/gocode
export PATH=$PATH:$GOPATH/bin
```

eurtnl, 2019-01-01:
Did that and than make. now i get:
```
go get ./...
import cycle not allowed
package github.com/LN-Zap/lndconnect
	imports encoding/base64
	imports encoding/binary
	imports errors
	imports runtime
	imports internal/bytealg
	imports internal/cpu
	imports runtime
Makefile:4: recipe for target 'dep' failed
```

fluidvoice, 2019-01-01:
I can't even get past the `go get` command it keeps telling me 
> cannot download, $GOPATH not set
I don't know squat about go programming and apparently they like to over complicate their build environment.

CryptoSigmund, 2019-01-02:
Try to run the mobile install a second time. This does trigger the download. First it failed like you stated.

fluidvoice, 2019-01-02:
> Try to run the mobile install a second time. This does trigger the download. First it failed like you stated.

what do you mean by "mobile install"?

CryptoSigmund, 2019-01-02:
The menu point
-Main Menu -> Mobile -> Yes -> Zap

I had it to rune 2 times until I get a QR Code. Then it worked from mobile and desktop.

fluidvoice, 2019-01-02:
This only works if, as mentioned above, if you already have 'go' installed.  The script does not install go for you so it will fail to build zapconnect or lndconnect.  What directions did you use to install go?

eurtnl, 2019-01-02:
I get this error when installing ZAP:
```
*** Setup ***

Installing zapconnect. Please wait...

Getting github.com/LN-Zap/zapconnect (please wait) ...
import cycle not allowed
package github.com/LN-Zap/zapconnect
	imports encoding/base64
	imports encoding/binary
	imports errors
	imports runtime
	imports internal/bytealg
	imports internal/cpu
	imports runtime

Building github.com/LN-Zap/zapconnect ...
go get ./...
import cycle not allowed
package github.com/LN-Zap/zapconnect
	imports encoding/base64
	imports encoding/binary
	imports errors
	imports runtime
	imports internal/bytealg
	imports internal/cpu
	imports runtime
Makefile:4: recipe for target 'dep' failed
make: *** [dep] Error 1
```

fluidvoice, 2019-01-02:
I think you need to learn how to install golang correctly on Raspbian.  I think what is failing is the make (go building) is not finding the build dependencies (imports).  Try searching the Raspbian and Raspberri Pi forums, or maybe golang forums or Zap forums.

rootzoll, 2019-01-02:
In the lastest SD card build. GO should be installed.

@eurtnl did you started with the lastest sd card relase or did you some update with XXupdateScripts in between?

zer0trst, 2019-01-02:
I've hit the same wall - fresh install of 0.99 on sd card today.

after running

      go get -d github.com/LN-Zap/lndconnect

all i get is this

[image]

Here is my go Env file

[image]

zer0trst, 2019-01-02:
Works for me now.

I removed previous install of golang

      sudo apt purge golang
      sudo apt autoremove   
      sudo apt update

then I ran

      wget https://storage.googleapis.com/golang/go1.11.linux-armv6l.tar.gz
      sudo tar -C /usr/local -xzf go1.11.linux-armv6l.tar.gz
      sudo rm *.gz
      sudo mkdir /usr/local/gocode
      sudo chmod 777 /usr/local/gocode
      export GOROOT=/usr/local/go
      export PATH=$PATH:$GOROOT/bin
      export GOPATH=/usr/local/gocode
      export PATH=$PATH:$GOPATH/bin

went to mobile wallet menu in raspbiblitz, ran it and voila it works!

Thanks!

fluidvoice, 2019-01-02:
Nice! I'll remove previous install and try it again.

fluidvoice, 2019-01-02:
Damn, I didn't realize the raspiblitz script installs lndconnect for you.  I already built it before choosing the raspiblitz wallet menu.

eurtnl, 2019-01-02:
> In the lastest SD card build. GO should be installed.
> 
> @eurtnl did you started with the lastest sd card relase or did you some update with XXupdateScripts in between?

I did a fresh install yesterday, no update.

eurtnl, 2019-01-02:
> Works for me now.
>
> I removed previous install of golang
>
> ```
>   sudo apt purge golang
>   sudo apt autoremove
>   sudo apt update
> ```
> then I ran
>
> ```
>   wget [...]

This did work for me too, thanks guys!

fluidvoice, 2019-01-03:
> This did work for me too, thanks guys!

Please close this issue. thanks.

zer0trst, 2019-01-03:
Okay I have reset my blitz again (due to other reasons) and can confirm I did have to also run the following

      go get -d github.com/LN-Zap/zapconnect
 
and

      go get -d github.com/LN-Zap/lndconnect

Not sure why this made it work but it did..?

rootzoll, 2019-01-10:
I will recheck for verison 1.0 - thx.

rootzoll, 2019-01-17:
OK. I had to fix the GOPATH again (for the next release), but then it installed zapconnect correctly and displayed the QR code. I was also able to scan the QR code with the ZAp mobile app and macaroon+TLS showed up correctly in the app.

But I was not able to connect with my dyndns or with publicIP - I had to replace it in the app with the local LAN IP an the the app was working with my RaspiBlitz. The macaroons/TLS are OK with the dyndns/publicIP, because they are working with no problem in Shango and Zap Desktop. So this looks like a bug on the Zap mobile site.

I wrote I tweet to the iOS dev: https://twitter.com/rootzoll/status/1086028470646394880

So I close this issue for now for the next release - if you have further problems, please request reopen.

closed 2019-01-17 by rootzoll