raspiblitz/raspiblitz #236: 70initLND.sh missing bitcoin-cli -datadir parameter.

fluidvoice, 2019-01-01:
After downloading and running XXupdateScripts.sh , the system was not starting up LND.
When I ran 70initLND.sh it was stuck in a waiting loop because of wrong rpc user/password. 
The problem was the two lines below do not include the -datadir parameter.  changing these lines to the following fixed it: 
`${network}-cli -datadir=/home/bitcoin/.${network} getblockchaininfo`

https://github.com/rootzoll/raspiblitz/blob/52c2132b8e6b42be2129607ed64acc62e0e7c5f4/home.admin/70initLND.sh#L12

https://github.com/rootzoll/raspiblitz/blob/ed748e433ff3548da05530d97756143e5b4f5991/home.admin/70initLND.sh#L30

rootzoll, 2019-01-15:
OK. Added the datadir to that lines. Marked for final testing before release.

rootzoll, 2019-01-17:
I ran the new 70initLND.sh two times during testing - looks good. Closing issue.

closed 2019-01-17 by rootzoll