raspiblitz/raspiblitz #237: Tor V3 on version 0.99

zer0trst, 2019-01-02:
I Posted this issue over on https://github.com/Stadicus/guides/issues/328

Just thought I would try here as well for anyone who may have the time to check it out?

Hoping to get some assistance on this one. I am trying to run through TOR v3 onion service. I have Setup my Raspiblitz today using the latest image (raspiblitz-0.99-2018-12-24.img.gz)

I tried the new option of setting from the main menu:  Go to SERVICES and Run behind TOR
[image]

It gives me this screen for a split second and then takes me back to the main menu. From there it does nothing further.

[image]

I rebooted in the hopes it would do something but still nothing..

So I went and Followed the steps outlined here: https://github.com/Stadicus/guides/blob/master/raspibolt/raspibolt_69_tor.md

added sources and installed tor-arm...

Added "bitcoin" to the group "debian-tor"

      $ cat /etc/group | grep debian-tor
         debian-tor:x:123:bitcoin

un-commented and added in "torrc" file

      $ sudo nano /etc/tor/torrc

[image]

Added lines to "bitcoin.conf" file

      $ sudo nano /home/bitcoin/.bitcoin/bitcoin.conf

[image]

Archived log file and restarted bitcoin core

      $ sudo mv /home/bitcoin/.bitcoin/debug.log /home/bitcoin/.bitcoin/debug.log.old
      $ sudo systemctl start bitcoind

ran the verification 

      $ sudo tail /home/bitcoin/.bitcoin/debug.log -f -n 200

[image]
[image]

and 

      $ bitcoin-cli getnetworkinfo

[image]

Everything looks right i think? but here's where I run into a brick wall....

I run 

    $ curl icanhazip.com

and my IP shows up:

[image]

Now when I run 

      $ bitcoin-cli getpeerinfo | grep  local

my IP is not included but my onion address is...

[image]

So i then go through the steps for LND as follows: 

      $ sudo nano /home/bitcoin/.lnd/lnd.conf

[image]

and when I run

      lncli getinfo 
      lncli getnodeinfo [YOUR_PUBKEY]

My IP is exposed here as well

[image]
[image]

So final thing I have tried is adding these lines to "/etc/tor/torrc" as someone on reddit had pointed out a few months back

[image]

then I grabbed the address out of /var/lib/tor/lnd_service/hostname and added it to /home/bitcoin/.lnd/lnd.conf in "externalip"

not sure i've put this in the right place or not??

[image]

This seems to have done something because after waiting about 10 minutes, when I run 

     lncli getinfo

Under URI's I now get both my public key against my exposed IP but also my public key against what appears to be a V3 onion address but it's not the address in /var/lib/tor/lnd_service/hostname or what I put under "externalip" in lnd.conf

[image]
[image]

[image]

so i'm stuck here..

any ideas? something I've missing along the way perhaps?

rootzoll, 2019-01-02:
Lets start at the beginning. Did you switched the TOR option in the settings menu "ON" by hitting spacebar before continue with OK?

zer0trst, 2019-01-02:
no I hit enter but see here https://github.com/Stadicus/guides/issues/328#issuecomment-450841426 

I think I have got it working now but  still not sure as to why I see my IP Still or whether I m supposed to see it or not?

zer0trst, 2019-01-03:
Okay so i feel like a dummy,  re-did everything, hit space on run behind Tor and it works.

Is there any reason why its not using onion v3, it seems to only run v2 this way only?

Thanks for your assistance rootzoll

fluidvoice, 2019-01-03:
> Lets start at the beginning. Did you switched the TOR option in the settings menu "ON" by hitting spacebar before continue with OK?

I tried enabled TOR in my already working node and upon reboot I just get tons of these in the bitcoind log and it never syncs:
`2019-01-03T13:33:12Z Cannot create socket for fno4aakpl6sg6y47.onion:8333: unsupported network`

rootzoll, 2019-01-03:
@zer0trst the UI for activating service is not optimal at that point - a webUI with better known is on the todo list. And its TorV2 because it had better integration with bitcoind and LND and the the hidden services addresses in v2 are much shorter with a change that you can still transfer them manually if needed. I know v3 provides better security, but its a lightning node - not a dark market .. so I think its OK to keep it to v2 for a bit more ease of use.

@fluidvoice what do you get if you run to check if tor is running:
`curl --connect-timeout 10 --socks5-hostname 127.0.0.1:9050 https://check.torproject.org`
Did you see any errors during the TOR install script?

fluidvoice, 2019-01-03:
> @zer0trst the UI for activating service is not optimal at that point - a webUI with better known is on the todo list. And its TorV2 because it had better integration with bitcoind and LND and the [...]

Not sure yet. I just ran `internet.tor.sh off` to put it back to running state.
I did notice an error here and wonder if it could be related to TOR not working for me:

> *** Remove NYX ***
> Cannot uninstall requirement nyx, not installed

zer0trst, 2019-01-06:
I’m happy to close this one if everyone has their answers?

Thanks

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

rootzoll, 2019-01-07:
@fluidvoice there was an report and suggested fix I want to add for next version on this issue for getting tor working: https://github.com/rootzoll/raspiblitz/issues/249

I will close this issue - please use issue above for further problems with TOR setup.

joernroeder, 2019-02-08:
running into the same issue as @fluidvoice. selected Tor in menu, and it installed afterwards.

when i run `internet.tor.sh on` after reboot again i get the following:

```
./internet.tor.sh on
making sure services are not running
switching the TOR ON

*** Check if TOR service is functional ***
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0curl: (7) Failed to connect to 127.0.0.1 port 9050: Connection refused
TOR not running ... proceed with switching to TOR.

dpkg-query: package 'tor-arm' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files,
and dpkg --contents (= dpkg-deb --contents) to list their contents.
*** Install TOR repo keys ***
gpg: keyserver receive failed: No data
gpg: WARNING: nothing exported
gpg: no valid OpenPGP data found.

gpg: keyserver receive failed: No data
gpg: WARNING: nothing exported
gpg: no valid OpenPGP data found.

*** Adding Tor Sources to sources.list ***
deb https://deb.torproject.org/torproject.org stretch main
deb-src https://deb.torproject.org/torproject.org stretch main
OK

*** Updating System ***
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:3 https://deb.nodesource.com/node_11.x stretch InRelease
Get:4 https://deb.torproject.org/torproject.org stretch InRelease [4,965 B]
Ign:4 https://deb.torproject.org/torproject.org stretch InRelease
Fetched 4,965 B in 1s (2,597 B/s)
Reading package lists... Done
W: GPG error: https://deb.torproject.org/torproject.org stretch InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 74A941BA219EC810
W: The repository 'https://deb.torproject.org/torproject.org stretch InRelease' is not signed.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: Target Sources (main/source/Sources) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:7
W: Target Packages (main/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6
W: Target Translations (main/i18n/Translation-en_GB) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6
W: Target Translations (main/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6
W: Target Sources (main/source/Sources) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:7
W: Target Packages (main/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6
W: Target Translations (main/i18n/Translation-en_GB) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6
W: Target Translations (main/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6

*** Installing dirmngr ***
Reading package lists... Done
Building dependency tree
Reading state information... Done
dirmngr is already the newest version (2.1.18-8~deb9u3).
The following packages were automatically installed and are no longer required:
  python-geoip python-torctl torsocks
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 63 not upgraded.

*** Install Tor ***
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  libzstd1 tor-geoipdb
Suggested packages:
  mixmaster torbrowser-launcher socat apparmor-utils obfs4proxy
The following NEW packages will be installed:
  libzstd1 tor tor-arm tor-geoipdb
0 upgraded, 4 newly installed, 0 to remove and 63 not upgraded.
Need to get 3,437 kB of archives.
After this operation, 12.9 MB of additional disk space will be used.
WARNING: The following packages cannot be authenticated!
  tor tor-geoipdb
E: There were unauthenticated packages and -y was used without --allow-unauthenticated

*** Tor Config ***
chown: cannot access '/var/run/tor/': No such file or directory

*** Installing NYX - TOR monitoring Tool ***
NYX already installed

*** Activating TOR system service ***
ReadWriteDirectories=-/mnt/hdd/tor
The unit files have no installation config (WantedBy, RequiredBy, Also, Alias
settings in the [Install] section, and DefaultInstance for template units).
This means they are not meant to be enabled using systemctl.
Possible reasons for having this kind of units are:
1) A unit may be statically enabled by being symlinked from another unit's
   .wants/ or .requires/ directory.
2) A unit's purpose may be to act as a helper for some other unit which has
   a requirement dependency on it.
3) A unit may be started when needed via activation (socket, path, timer,
   D-Bus, udev, scripted systemctl call, ...).
4) In case of template units, the unit is meant to be enabled with some
   instance name specified.

*** Changing bitcoin Config ***
Chain network already configured for TOR
*** Putting LND behind TOR ***
Make sutre LND is disabled
editing /etc/systemd/system/lnd.service
Enable LND again
Created symlink /etc/systemd/system/multi-user.target.wants/lnd.service → /etc/systemd/system/lnd.service.
OK

OK - TOR is now ON
needs reboot to activate new setting
```

openoms, 2019-08-26:
> @zer0trst the UI for activating service is not optimal at that point - a webUI with better known is on the todo list. And its TorV2 because it had better integration with bitcoind and LND and the [...]

@rootzoll do you think we could change to Tor V3 now as it is matured a lot in lnd and in the Tor stack too?
The only drawback I can see is the longer address, but I think that gets copy pasted most of the time from either the menu or from a LN explorer.

Had a chat on Mastodon with FreePietje@x0f.org about using Tor V3 vs V2: 

>Benefits of v3:
https://2019.www.torproject.org/docs/tor-onion-service.html.en#four

>Better crypto (replaced SHA1/DH/RSA1024 with SHA3/ed25519/curve25519)

>SHA1 has not been considered a safe hashing algo for a while now.
RSA1024 should've been banned about more then 5 years now. I think OpenSSL 1.1 actively refuses to connect with such weak security.
New default (in Debian at least) is 2048, but I use/generate 4096 for several years now.

I have changed to --tor.v3  in the `/etc/systemd/system/lnd.service` on two of my nodes and running without problems. 
[image]

rootzoll, 2019-08-28:
@openoms I think changing to TOR v3 is now something to consider. I added it to the "future release" milestone to consider it for one of the upcoming releases.

openoms, 2019-09-11:
Other Tor related optimizations:

from: https://github.com/lightningnetwork/lnd/blob/master/sample-lnd.conf
 ```
Enable Tor stream isolation by randomizing user credentials for each connection. With this mode active, each connection will use a new circuit.
This means that multiple applications (other than lnd) using Tor won't be mixed
in with lnd's traffic.
tor.streamisolation=1
```

Also should add the `bitcoin` use to the `debian-tor` group, so avoiding errors in the log and can run `nyx` without `sudo`:

`$ sudo usermod -a -G debian-tor bitcoin`

woeisme, 2019-09-12:
> Other Tor related optimizations:
>
> from: https://github.com/lightningnetwork/lnd/blob/master/sample-lnd.conf
>
> ```
> Enable Tor stream isolation by randomizing user credentials for each connection. With [...]

yes so much this!

openoms, 2019-09-12:
OK, testing the modified `/etc/systemd/system/lnd.service` with:
```
ExecStart=/usr/local/bin/lnd --tor.active --tor.v3 --listen=localhost --tor.streamisolation
```
Running `$ sudo usermod -a -G debian-tor bitcoin`
Still leads to an error for `nyx` (without sudo):
```
We were unable to read tor's authentication cookie...

  Path: /mnt/hdd/tor/sys/control_auth_cookie
  Issue: Authentication failed: '/mnt/hdd/tor/sys/control_auth_cookie' doesn't exist
```

openoms, 2019-09-15:
Related PR https://github.com/rootzoll/raspiblitz/pull/777

rootzoll, 2019-09-17:
Thx. Will merge and test once I start working on v1.4 branch.

rootzoll, 2019-12-19:
First test looks good .. at least ready for v1.4RC1 for more depth testing.

rootzoll, 2020-01-21:
Seems to run stable - closing issue.

closed 2020-01-21 by rootzoll