raspiblitz/raspiblitz #239: raspiblitz stuck on: *** Test if chainnetwork is ready ... ***

rolandbitcoinspot, 2019-01-02:
Hi,

my raspiblitz is tuck on the following error message. *** Test if chainnetwork is ready ... *** (see image)
device is online and has an internet connection as far as i can see. had VPN on but shut that down also.

[image]

rootzoll, 2019-01-02:
Either this is some older version of raspiblitz or did you copy over your own blockchain from another computer?

fluidvoice, 2019-01-02:
If you ever started up bitcoind using SSH the blockchain DB can get into a state where the files are owned by the user "admin" instead of the user "bitcoin".  I had some problems because of this. And to fix it I had to use the command `chown` to recursively change all the files rights to `bitcoin:bitcoin`.  I don't remember the exact syntax I used. See the man page for `chown`.

rolandbitcoinspot, 2019-01-02:
> If you ever started up bitcoind using SSH the blockchain DB can get into a state where the files are owned by the user "admin" instead of the user "bitcoin". I had some problems because of this. And to fix it I had to use the command `chown` to recursively change all the files rights to `bitcoin:bitcoin`. I don't remember the exact syntax I used. See the man page for `chown`. Note, performing these two commands should show nothing if your blockchain DB is correct:
> $ `ls -al /mnt/hdd/bitcoin/blocks | grep admin`
> $ `ls -al /mnt/hdd/bitcoin/chainstate | grep admin`

Hi 

> Either this is some older version of raspiblitz or did you copy over your own blockchain from another computer?

I reinstalled the microsd card, the bc files were probably still there from the previous install. any steps i can take in this instance ?

fluidvoice, 2019-01-02:
> I reinstalled the microsd card, the bc files were probably still there from the previous install. any steps i can take in this instance ?

As I said you need to check the file rights with the `ls` commands I gave and change them to `bitcoin:bitcoin` with the `chown` command if they are wrong.

rootzoll, 2019-01-09:
@fluidvoice To automatically fix that onwership problem of blockchain data I added a I chown line into the bootstrap - getting executed on every startup - just in case for the future.

@rolandbitcoinspot I closing this issue for now. Please post if you get stuck again at that spot.

closed 2019-01-09 by rootzoll