raspiblitz/raspiblitz #240: Change of hardware RBP (From B to B+)

RobertSundh, 2019-01-02:
So I tryed switching to a model b+ from b, same SD card, SSD and screen. Node was running and synced on the b. But when I start it up I come to the sync screen : 
Lightning Filtering Blockchain
Progress : ?/       (Weird)
please wait etc etc...

and also this keeps looping : 

./00mainMenu.sh: line 97: 25523 Illegal instruction     sudo -u bitcoin ${network}-cli -datadir=/home/bitcoin/.${network} getblockchaininfo > /dev/null 2> error.tmp
tail: cannot open '/mnt/hdd/lnd/logs/bitcoin/net/lnd.log' for reading: No such file or directory
tail: cannot open '/mnt/hdd/lnd/logs/bitcoin/net/lnd.log' for reading: No such file or directory
tail: cannot open '/mnt/hdd/lnd/logs/bitcoin/net/lnd.log' for reading: No such file or directory
(standard_in) 1: syntax error
tail: cannot open '/mnt/hdd/lnd/logs/bitcoin/net/lnd.log' for reading: No such file or directory

Any ideas? :) I think it should work its just a new RBP.

fluidvoice, 2019-01-02:
What is the contents of:
`/home/admin/raspiblitz.log`
`/home/admin/raspiblitz.info`
The 3B+ will need different firmware. What do you see here:
`sudo ls -al /boot | grep bcm`
It should include **bcm2710-rpi-3-b-plus.dtb**

fluidvoice, 2019-01-02:
Also, see this thread for upgrading from 3B to 3B+.
In general always wise to search the RPI forums when it's RPI related (wink-wink): https://www.raspberrypi.org/forums/viewtopic.php?p=1400579#p1400483

RobertSundh, 2019-01-02:
Thanks for the answer. Sorry I didnt think of that lol I thought it was raspiblitz releated only but ofc ppl upgrade other builds to.

admin@RaspiBlitz:~ $ cat /home/admin/raspiblitz.log

***********************************************
Running RaspiBlitz Bootstrap 0.99
Wed  2 Jan 17:22:35 GMT 2019
***********************************************
Check if HDD contains configuration ..
load and update publicIP
update value (46.39.111.252)
DONE BOOTSTRAP
////////////////////////////////////////
admin@RaspiBlitz:~ $ cat /home/admin/raspiblitz.info
state=ready
network=
chain=
message='waiting login'
////////////////////////////////////////
admin@RaspiBlitz:~ $ sudo ls -al /boot | grep bcm
-rwxr-xr-x  1 root root   23315 Sep 19 19:06 bcm2708-rpi-0-w.dtb
-rwxr-xr-x  1 root root   23071 Sep 19 19:06 bcm2708-rpi-b-plus.dtb
-rwxr-xr-x  1 root root   22812 Sep 19 19:06 bcm2708-rpi-b.dtb
-rwxr-xr-x  1 root root   22589 Sep 19 19:06 bcm2708-rpi-cm.dtb
-rwxr-xr-x  1 root root   24115 Sep 19 19:06 bcm2709-rpi-2-b.dtb
-rwxr-xr-x  1 root root   25574 Sep 19 19:06 bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x  1 root root   25311 Sep 19 19:06 bcm2710-rpi-3-b.dtb
-rwxr-xr-x  1 root root   24087 Sep 19 19:06 bcm2710-rpi-cm3.dtb

fluidvoice, 2019-01-02:
The failure you saw is because raspiblitz.info has "network=" where is should be "network=bitcoin" and should also contain "chain=main".
I think but not sure this is because the Raspiblitz boot script over-writes this file after every reboot.
Is your HDD connected and has your synced blockchain?
You might want to try running the following commands which worked for me when I was having problems:
// download XXupdateScripts script
`sudo wget https://raw.githubusercontent.com/rootzoll/raspiblitz/master/home.admin/XXupdateScripts.sh`
// run it
`sudo chmod +x XXupdateScripts.sh`
`sudo ./XXupdateScripts.sh`

RobertSundh, 2019-01-02:
Jepp HDD is connected and sync just before i switched pie.

Well that didnt work, same text looping.. 

Thanks for the help anyway :+1:  :) I might keep the B on the node and do something else with the b+ !

fluidvoice, 2019-01-02:
You could try this. Edit the **/home/admin/raspiblitz.info** file so it looks like:
```
state=ready
network=bitcoin
chain=main
setupStep=0
```
Then run ./10setupBlitz.sh

fluidvoice, 2019-01-03:
close this issue if you're not going to go on with it. Thanks.

closed 2019-01-07 by RobertSundh