raspiblitz/raspiblitz #241: Opening Channel rpc error: code = Unknown desc = reservation timed out waiting for peer

zer0trst, 2019-01-03:
Anyone have any idea what could be causing this? 

Connected to peers fine but when I go to open channel with them It hangs for about 15 minutes and spits out this.

`[lncli] rpc error: code = Unknown desc = reservation timed out waiting for peer (peerID:&{0x13432d0 540979851843736396442121175776881883303619030496332697958684396087317895828 30791537473803715475186086513457980142719295813684146174162300728314291292571}, chanID:01a3596606648be78aa6656a5fbd4d4b258ed137322b6d106e2857de158bae1e)`

It's done this with 5 different nodes I've tried to connect to.

Any other information needed do diagnose?

zer0trst, 2019-01-03:
Have just managed to connect to a few clearnet nodes, must have just been the onion ones I was trying to open channels with, strange they could connect but couldnt open channel

closed 2019-01-03 by zer0trst

fluidvoice, 2019-01-03:
What happens when you run from the SSH terminal:
`lnbalance bitcoin`

fluidvoice, 2019-01-09:
> Have just managed to connect to a few clearnet nodes, must have just been the onion ones I was trying to open channels with, strange they could connect but couldn't open channel

Myself and one other person have seen opening a channel to a **c-lightning** (Blockstream) node from an **LND** node (ie Raspiblitz) fails like this. For me it's very repeatable.  Other LND nodes are not a problem for me.  I will try to look into the problem.