raspiblitz/raspiblitz #243: Line failure caused raspiblitz not booting

lucbran, 2019-01-04:
Hi,
one week ago installed Raspiblitz and got my full node running. Opened LN channel with some funds. Everything perfect (except still wondering how to set an alias name to the node..).
Yesterday after temporary internet line failure I cannot have raspiblitz booting. System is asking login password and then terminal takes me to root. How to re boot raspiblitz with terminal command?(I switched off and restarted but not working).
Thanks for help

fluidvoice, 2019-01-04:
sudo reboot

lucbran, 2019-01-04:
not working.. may be flash drive corrupted now? I am trying to reinstall with new image flash drive following documentation instruction but seems process is aborted when creating mnt/hdd folder as same already exists.. any idea?

fluidvoice, 2019-01-05:
> not working.. may be flash drive corrupted now? I am trying to reinstall with new image flash drive following documentation instruction but seems process is aborted when creating mnt/hdd folder as same already exists.. any idea?

try running the script `./XXcleanHDD.sh` first.

lucbran, 2019-01-05:
I tried everything as shown in documentation but no way.. I am now reinstalling with fresh sd card (had to download again all blockchain) and hope to retrieve my previous wallet by the seed.
My only concern this doesnt happen again as I dont wnt to gothrough the same process again.. There must be a way to boot raspiblitz in case of power or line failure and on raspberry shows: enter login for maintenance...
I entered login pwd and was back to root position but no idea how to restart raspiblitz (sudo reboot doesnt work)...

fluidvoice, 2019-01-05:
so if you unplug it from power and start it back up, what does it do?

lucbran, 2019-01-05:
when booing, raspberry returns an error asking pwd for maintenanec. I enter password and terminal takes me to root but no idea how to restart raspiblitz ..

fluidvoice, 2019-01-05:
A picture or screen capture would help

lucbran, 2019-01-05:
unfortunately i am in the process of reinstalling everything... 
but my question is: has anyone got power failure and was able to reboot raspiblitz without any particular command to be entered?
i am concerned this could happen again..

fluidvoice, 2019-01-05:
maybe get a UPS with spike/surge protection. Something unusual happened to your system. I've had multiple power failures before and also unplugged the power when running and didn't have any problem with the system coming back up as normal without any special commands entered.

lucbran, 2019-01-05:
this relieves me! may be I got something wrong then..
ok in the meantime I am still re validating the LN nodes.. when finished will see if i have been able to recover seed and satoshis I had in the previous wallet...(read some users couldnt get them back).
thanks for your help!