raspiblitz/raspiblitz #244: Req: Support for other single board computers (i.e. ROCK64)

unic0der, 2019-01-04:
For the reason of future-proofing and faster sync, it would be beneficial to introduce support for other, stronger single board computers such as the [ROCK64](https://www.pine64.org/?product=rock64-media-board-computer) with stronger CPU, a lot more RAM, as well as USB 3.0 support

rootzoll, 2019-01-09:
Yes that would be great and is in the list of ideas. But its a bigger step, because this would mean to exchange the Raspbian base OS to something like DietPi. 

The sd crad build script has already some preparations in to detect when building on something else then Raspian - that would need to be extended:
https://github.com/rootzoll/raspiblitz/blob/master/build.sdcard/raspbianStretchDesktop.sh

If someone can get the RaspiBlitz on a Rock64/DietPi running please share experience.

openoms, 2019-01-22:
Hi! I have an ODroid HC1 and HC2 originally bought to do the Thunderdroid guide, but never got to it.
I have tried the Raspiblitz script with DietPi unfortunately not successfully so far. I keep running into a strange error during the initial setup after the HDD is set up:
```
Failed to connect to bus: No such file or directory
1548192239
LND not ready yet ... waiting another 60 seconds.
If this takes too long (more then 10min total) --> CTRL+c and report Problem
```
see my full output here: https://github.com/openoms/droidblitz/blob/master/DietPi/initial_setup_output.html
And the whole context here: https://github.com/openoms/droidblitz/blob/master/DietPi/install.md

21isenough, 2019-01-23:
I just got my Odroid HC1 hardware delivered - ready to be loaded with bitcoin software :-D I'll be able to run that script over the weekend as well and report on my errors and outputs.

justinmoon, 2019-01-24:
I'll try to test as well. I've got an Odroid lying around ...

fluidvoice, 2019-01-29:
> Hi! I have an ODroid HC1 and HC2 originally bought to do the Thunderdroid guide, but never got to it.
> I have tried the Raspiblitz script with DietPi unfortunately not successfully so far. I keep [...]

Potential clues?... https://superuser.com/questions/759759/writing-a-service-that-depends-on-xorg

openoms, 2019-01-29:
making progress here: 
using systemctl with sudo allowed to step forward, thanks  for helping @21isenough !
Working on some more changes, most additions are already pushed here: https://github.com/openoms/HardwareNode/tree/OdroidHC1Debug

openoms, 2019-01-30:
the current working branch is: https://github.com/openoms/HardwareNode/tree/raspiblitz-dev updated with the raspiblitz dev branch code. Everything seems to work so far.

openoms, 2019-02-26:
There is this sad and concerning tweet regarding the future of the DietPi project: https://twitter.com/DietPi_/status/1098420068998938625

openoms, 2019-02-27:
There is hope, looking forward to the announcement. https://twitter.com/DietPi_/status/1100584695702736896
Will post updates here.

openoms, 2019-03-21:
I have managed to make the pi user autologin and the default GPIO LCD display work with DietPi on the RPi without issues, but I have a strange one with this  HDMI screen. (https://www.aliexpress.com/item/3-5-inch-LCD-HDMI-USB-Touch-Screen-Real-HD-1920x1080-LCD-Display-Py-for-Raspberri/32818537950.html)
I wonder if anyone experienced similar. The text outputs and dialog boxes appear fine, but the status screen only takes half of the screen as pictured. The resolution is set to the display's native 480x320. 
[image]
[image]
[image]
Any tips how to fix this?

fluidvoice, 2019-03-21:
> I have managed to make the pi user autologin and the default GPIO LCD display work with DietPi on the RPi without issues, but I have a strange one with this HDMI screen. [...]

https://www.raspberrypi.org/forums/viewtopic.php?t=175616#p1120534

> Here is what need to be set in /boot/config.txt to use this display (this is only a part of the file):
> 
> #. Force use HDMI
> hdmi_force_hotplug=1
> #. force a specific HDMI mode
> hdmi_group=2
> hdmi_mode=87
> 
> #. custom display resolution (480x320 looks good , and 720x480 is not too bad too,3rd res is: 810x540 )
> #.         /----------------- Width: 480px ( or 720px /or 810px)
> #.         |   /------------- Height: 320px ( or 480px /or 540px)
> #.         |   |   /--------- Refresh: 60Hz
> #.         |   |   | /------- Aspect ratio: 6=15:9 (1=4:3, 2=14:9, 3=16:9, 4=5:4, 5=16:10)
> #.         |   |   | | /----- Margins: 0=disabled (1=enabled)
> #.         |   |   | | | /--- Interlace: 0=progressive (1=interlaced)
> #.         |   |   | | | | /- Reduced blanking: 0=normal (1=reduced)
> #.         |   |   | | | | |
> hdmi_cvt 480 320 60 6 0 0 0
> 
> #. uncomment to force a HDMI mode rather than DVI. This can make audio work in
> #. DMT (computer monitor) modes
> hdmi_drive=2
> 
>

fluidvoice, 2019-03-21:
Let us know how that screen it works.

openoms, 2019-03-21:
Thanks, but I think it suggests the same settings I have used given by the vendor: 
http://www.lcdwiki.com/3.5inch_HDMI_Display-B
https://github.com/goodtft/LCD-show/blob/master/boot/config-35-480X320.txt
Will test more on an RPi on Raspbian and on the Odroid XU4 with DietPi.
Everything displays well except the status screen, I suspect it could be a font issue.
Anyway there are some folks, who were testing HDMI screens, I will try to ask more specifically if the problem persists.

fluidvoice, 2019-03-22:
The /boot/cmdline.txt has this included in the line `fbcon=map:10 fbcon=font:ProFont6x11`
Maybe that's it. I don't know where that is being set though. It's not in the build_sdcard.sh script.

rootzoll, 2019-03-22:
The dialogboxes can mange different screen/ascii resolutions. The status screen is fixed to the standard ascii width/height of the LCD. I think its a setting of the resolution.

BTW: It would be great if we can get the standard GPIO LCD display working on dietpi - that would open up the possibility to make a complete move at one point from raspbian to dietpi with an update even for all people running the raspberry with the GPIO LCD. Is there any changes to get that working?

fluidvoice, 2019-03-22:
> possibility to make a complete move at one point from raspbian to dietpi
Why? Is there some advantage, like perhaps lower memory usage, faster, or something?
Or just to have a second option for no specific reason?

rootzoll, 2019-03-22:
Dietpi is a more universal platform that supports a broader range of SBC - including the raspberryPi. Raspbian is very limited just to their own hardware. So being based on dietpi, would give people more choice in choosing their hardware in the future. Also it claims to be `lightweight` - but thats something to be seen in comparison in the end.

fluidvoice, 2019-03-22:
> The dialogboxes can mange different screen/ascii resolutions. The status screen is fixed to the standard ascii width/height of the LCD. I think its a setting of the resolution.
>
> BTW: It would be [...]

Has @openoms or anyone tried with the standard LCD and what happens?

openoms, 2019-03-22:
@rootzoll the default GPIO LCD is working together with the automatic install in DietPi on the RPi. I am just waiting with the PR until the 1.1 changes are merged into the dev branch. 
Playing along with this HDMI screen for the sake of my Odroid XU4.

openoms, 2019-03-22:
> The /boot/cmdline.txt has this included in the line fbcon=map:10 fbcon=font:ProFont6x11 
Maybe that's it. I don't know where that is being set though. It's not in the build_sdcard.sh script.

The cmdline.txt is being copied in at the end of the build_sdcard.sh by the https://github.com/goodtft/LCD-show/blob/master/LCD35-show or the MPI3508-show in the HDMI screen's case.
DietPi is running the https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-autostart at every start overwriting some files with ones from the /DietPi folder depending on the settings.
I will try to put this cmdline.txt in both locations.

> I think its a setting of the resolution
It is unlikely  as it is set to the display's native 480x320, same as the default GPIO LCD. Also I have tried numerous other (higher  and/or different aspect ratio) resolutions and the status screen showed either the same or worse.

fluidvoice, 2019-03-22:
> Dietpi is a more universal platform that supports a broader range of SBC - including the raspberryPi. Raspbian is very limited just to their own hardware. So being based on dietpi, would give people [...]

from the Readme... "DietPi is an extremely lightweight Debian-based OS. With images starting at 400MB, that's 3x lighter than 'Raspbian Lite'. It is highly optimized for minimal CPU and RAM resource usage, ensuring your SBC always runs at its maximum potential."
Nice marketing ;-) I wonder what the real difference is in resource usage? Most importantly the RAM usage.

openoms, 2019-03-22:
>Nice marketing ;-) I wonder what the real difference is in resource usage? Most importantly the RAM usage.

It is lighter in the sense that it is stripped down of any unnecessary dependency with the option to install a extras through the dietpi-software.

It is an open source - community managed project, I don't think it needs marketing at all.

fluidvoice, 2019-03-22:
@rootzoll @openoms  I must say I really like this idea.  I'm interested in building on top of a RaspiBlitz type of platform (SBC+bitcoin+LN) and having the option of using a board with more CPU power and more importantly more RAM if I need it... and who knows if/when Raspberry Pi 4 will come out and what spec's it will have.

fluidvoice, 2019-03-22:
> It is an open source - community managed project, I don't think it needs marketing at all.
I was joking. I just meant they are touting the supposed benefits.  If it's true, it's great.

openoms, 2019-03-22:
>I was joking. I just meant they are touting the supposed benefits. If it's true, it's great.

To be fair they got some bashing lately from the guy managing armbian for the same reason: https://twitter.com/armbian/status/1108645865621737472?s=19
Despite taht I think that the project has a big advantage in the support (and testing) of multiple platforms. Making the RaspiBlitz less hardware dependent helps the whole network of Bitcoin and Lightning.
In the unlikely event of the DietPi community halting we would need to fall back onto specific builds for different hardwares.

fluidvoice, 2019-03-23:
How does Armbian compare to DietPi? I'm assuming you think DietPi is better for some reasons?

openoms, 2019-03-24:
@fluidvoice DietPi has this automated setup feature, so one only needs to burn the image and copy or modify two files to have a one button image setup. Please skim through my writeup here (it can and will be improved further) : https://github.com/openoms/raspiblitz/blob/DietPi/dietpi/RaspberryPi.md (to test need to use my repo for now until it is merged in rootzoll's dev branch) .

Here you can see how an image is converted to a DietPi image: https://github.com/MichaIng/DietPi/issues/1285#issue-280771944

So you can even make a DietPi image from an Armbian build. 
The two main reasons to use this distro is the automation and the compatibility with multiple SBCs.

openoms, 2019-03-26:
My HDMI siplay issue is solved with running: 
`sudo setfont /usr/share/consolefonts/Uni3-TerminusBold24x12.psf.gz`
(comment here: https://github.com/rootzoll/raspiblitz/issues/341#issuecomment-476664969)
Now just need to find out how to make it permanent. 
I have put the line to .profile and .bashrc in /home/dietpi (that equals to /home/pi in raspbian) without success so far.

fluidvoice, 2019-03-27:
> @fluidvoice DietPi has this automated setup feature, so one only needs to burn the image and copy or modify two files to have a one button image setup. Please skim through my writeup here (it can and will be improved further) : https://github.com/openoms/raspiblitz/blob/DietPi/dietpi/RaspberryPi.md (to test need to use my repo for now until it is merged in rootzoll's dev branch) .
> 
> Here you can see how an image is converted to a DietPi image: [MichaIng/DietPi#1285 (comment)](https://github.com/MichaIng/DietPi/issues/1285#issue-280771944)
> 
> So you can even make a DietPi image from an Armbian build.
> The two main reasons to use this distro is the automation and the compatibility with multiple SBCs.

@openoms So I tried this, burned your image to a USB drive and it starts booting up fine and gets pretty far along then gets stuck waiting for the network connection.  It just prints out a statement forever on the HDMI screen.  I tried editing the dietpi.txt file and setting both wifi and ethernet, using either DHCP or static IP nothing works.  I never gets past this.  Any ideas?

openoms, 2019-03-27:
@fluidvoice Sorry I was not explaining it fully about changing to my repo.
In the Automation_Custom_Script.sh change the 

`wget https://raw.githubusercontent.com/openoms/raspiblitz/raspiblitz-dev/build.sdcard/raspbianStretchDesktop.sh && sudo bash raspbianStretchDesktop.sh DietPi-dev openoms` 
to
`wget https://raw.githubusercontent.com/openoms/raspiblitz/DietPi-dev/build_sdcard.sh  && sudo bash build_sdcard.sh DietPi-dev openoms`

this will build exactly what will be there when my PR is merged in (https://github.com/rootzoll/raspiblitz/pull/463).

fluidvoice, 2019-03-27:
> @fluidvoice Sorry I was not explaining it fully about changing to my repo.
> 
> Start with a fresh SDcard.
> 
> Use this dietpi.txt: https://github.com/openoms/raspiblitz/raw/DietPi-dev/dietpi/boot/dietpi.txt
> 
> And in the Automation_Custom_Script.sh change the
> 
> `wget https://raw.githubusercontent.com/openoms/raspiblitz/raspiblitz-dev/build.sdcard/raspbianStretchDesktop.sh && sudo bash raspbianStretchDesktop.sh DietPi-dev openoms`
> to
> `wget https://raw.githubusercontent.com/openoms/raspiblitz/DietPi-dev/build_sdcard.sh && sudo bash build_sdcard.sh DietPi-dev openoms`
> 
> this will build exactly what will be there when my PR is merged in (#463).

That's what I did. Burned fresh image w/Etcher to USB drive (I seriously doubt SD card will be any different).  Then copied that dietpi.txt file.  
Note: I need to use WiFi as my router is too far away from my HDMI monitor and I need to see what's going on during bootup.  Why does WiFi not work?

openoms, 2019-03-27:
@fluidvoice This is great, I was not testing using wifi or booting from USB on DietPi yet, so it is all new info. 

But first of all I hope the default method works with the SD and ethernet works for you as well. I got an RPi with the same functionality as with Raspbian. Also an Odroid XU4 with the HDMI screen attached and working and an Odroid HC1 . All built from the same repo.

fluidvoice, 2019-03-28:
@openoms so I figured out how to get the WiFi working and everything built and booted up OK. The only problem is my LCD screen image is upside down.  
Note: I could not get any of the config files to get WiFi working so I had to change the wait setting from "2" (wait forever) to "1" (wait for 10 seconds), then once booted ran dietpi-config to enable WiFi then rebooted. Would be nice to figure out how to do this from config files without needing to run dietpi-config.

openoms, 2019-03-28:
>The only problem is my Raspiblitz LCD screen image is upside down.

This should solve it: 
https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#can-i-flip-the-screen
Except you need to edit at /DietPi/config.txt, which overwrites the /boot/config.txt on restart. I will change the default setting on install, thanks for pointing it out.

fluidvoice, 2019-03-28:
> > The only problem is my Raspiblitz LCD screen image is upside down.
> 
> This should solve it:
> https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#can-i-flip-the-screen
> Except you need to edit at /DietPi/config.txt, which overwrites the /boot/config.txt ong g restart. I will change the default setting on install, thanks for pointing it out.
> 
> Regarding the USB boot setup we need to extend the logic of 30addHDD.sh to look for the biggest partition on sdb as well. That would make it compatible with both the SD and USB booted systems. @fluidvoice would you be ok to test and make a PR for that? I can look into it in a couple of days also.

Yeah will look into it.  The LND and bitcoind services are also not running so there are prob other script changes needed to get USB install working.

fluidvoice, 2019-03-28:
> This should solve it:
> https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#can-i-flip-the-screen
> Except you need to edit at /DietPi/config.txt, which overwrites the /boot/config.txt ong g [...]
**Fixed.** The problem was **/DietPi/config.txt** was missing the rotate clause:
`dtoverlay=tft35a`
was changed to...
`dtoverlay=tft35a:rotate=270`

fluidvoice, 2019-03-28:
Regarding enabling WiFi on first boot. I found on my Raspi (after enabling WiFi via running dietpi-config) that the **/etc/wpa_supplicant/wpa_supplicant.conf** file contained my login info so maybe setting this will do it (need to test):

> country=US
> ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
> update_config=1
> network={
> ssid="XXX_SSID_XXX"
> scan_ssid=1
> key_mgmt=WPA-PSK
> psk="xxx_passkey_xxx"
> }

openoms, 2019-03-29:
Yes, the wpa_supplicant.conf is the way to set wifi up on Raspbian as well. 
On DietPi you will need to set the `AUTO_SETUP_NET_WIFI_ENABLED=` to `1` in the dietpi.txt to have it working on first boot. I am not sure were to put the credentials, but the wpa_supplicant.conf would make sense.

fluidvoice, 2019-03-29:
small note on the wifi:

> How wpa_supplicant.conf deals with priority and ordering
> -- Prefers to connect to a secured (WPA/WEP) SSID, regardless of lower signal strength against open SSID.
> -- Aside from that, connects to the SSID with the greatest strength, regardless of its order in wpa_supplicant.conf

fluidvoice, 2019-03-29:
ah, just saw this. Prob why it didn't work for me. Need to test:
https://dietpi.com/phpbb/viewtopic.php?f=8&t=9&p=9#p9
[image]

fluidvoice, 2019-03-29:
> The wpa_supplicant.conf is the way to set wifi up on Raspbian.
> 
> For DietPi see: https://dietpi.com/phpbb/viewtopic.php?f=8&t=9#p9
> 
> Optional (Wifi):
> Pre-configure Wifi Details for DietPi
> 
> ```
> Goto my computer, select the SD card and open it.
> Locate the file called dietpi.txt and open it with wordpad.
> Set AUTO_SETUP_NET_WIFI_ENABLED=1
> DietPi_v6.15 and lower images:
>     Enter your ssid AUTO_SETUP_NET_WIFI_SSID=MySSID and AUTO_SETUP_NET_WIFI_KEY=MyWifiKey (both are case sensitive).
> DietPi_v6.16 and higher images:
>     Open dietpi-wifi.txt and open it with wordpad
>     Change aWIFI_SSID[0]="MySSID" and aWIFI_KEY[0]="MyWifiKey"
> Save chanages to edited files
> Plug the Wifi adapter into your device, before turning it on.
> ```

LOL. I just found the same thing, then saw your post :)

fluidvoice, 2019-03-29:
Yes this worked.  The install via WiFi automatically progressed with these changes. After first boot the dietpi-wifi.txt file was auto-removed so that config file only works on first boot.

openoms, 2019-03-29:
> Yes this worked.  The install via WiFi automatically progressed with these changes. After first boot the dietpi-wifi.txt file was auto-removed so that config file only works on first boot.

Great! Please see the intended behaviour of the dietpi.txt: https://github.com/MichaIng/DietPi/blob/master/dietpi.txt
>IMPORTANT:
 Modifications to /boot/dietpi.txt will not be preserved on reboot.
 Please ensure you edit from the DietPi-RAMdisk location: /DietPi/dietpi.txt```

>Btw @openoms , the second time I tested this I used a big 64GB Samsung USB flash drive. There are a couple places where the bootup pauses for a god-aweful long time... it pauses a bit when auto-resizing the partition to max size (nice!) and then later for even longer when it's doing something unknown - no status is displayed. It looked like it was frozen and not doing anything but eventually it continued. Is there any way to show this output on the LCD screen during the initial install/upgrade process instead of just a white screen?

Thanks for pointing out. Unfortunately I see no way to make the LCD show the output of the automated building process, because it needs a reboot before it works. To see what is  happening you can do as described in the readme: https://github.com/rootzoll/raspiblitz/blob/dev/dietpi/RaspberryPi.md
>To follow the logs during the automated building process login with root and press CTRL+C.
tail -n1000 -f /tmp/DietPi-Update/dietpi-update.log - follow the dietpi-update process
tail -n1000 -f /var/tmp/dietpi/logs/dietpi-automation_custom_script.log follow the output of the build_sdcard.sh

fluidvoice, 2019-03-29:
> no way to make the LCD show the output
Not so sure.  Though perhaps it's not simple: https://github.com/MichaIng/DietPi/issues/2590#issuecomment-467189302

openoms, 2019-03-31:
> Btw, if that HDMI LCD you linked to above #244 (comment) is used, then event the bootup process is displayed, correct?

yes, it is displayed, but you cannot read it until installing the driver and reboot. See the linked issue above.

openoms, 2019-03-31:
@rootzoll I think this issue is best to be closed as it derailed towards a DietPi discussion anyway.
From my side the Rock64 is iced until  at least their next (v3) version is out.
https://twitter.com/openoms/status/1111544436054478848

rootzoll, 2019-04-02:
@openoms I renamed the issue to a DietPi discussion in general. There is no other issue covering this, so I keep it open - also because its an ongoing development.

openoms, 2019-04-02:
As it looks I will look into trialling Armbian. They have ready made images for multiple boards as well and looks to be a cleaner base to build on.
@rootzoll see the comment here: https://github.com/rootzoll/raspiblitz/issues/474#issuecomment-479006912 Is there anything else you consider when choosing a platform for the RaspiBlitz? WWhat you think of the Armbian vs using the manufacturer provided images (Raspbian or even the DietPi ones)

rootzoll, 2019-04-02:
@openoms as long as the OS is also running on raspberrys and support what we need to run the RaspiBlitz I am open and curious what you all think is giving us the most flexibility.

fluidvoice, 2019-04-03:
> As it looks I will look into trialling Armbian. They have ready made images for multiple boards as well and looks to be a cleaner base to build on.
> @rootzoll see the comment here: [#474 (comment)](https://github.com/rootzoll/raspiblitz/issues/474#issuecomment-479006912) Is there anything else you consider when choosing a platform for the RaspiBlitz? WWhat you think of the Armbian vs using the manufacturer provided images (Raspbian or even the DietPi ones)

I agree. I think we should test/try building on top of Armbian.  I think building on top of Debian base will prob be too much work - adding distro/release maintenance work we don't want to focus on.
I will try to run @openoms DietPi build script on top of Armbian n see if it works n what changes are required.

openoms, 2019-04-03:
> I will try to run @openoms DietPi build script on top of Armbian and see if it works on the Pi3B+ and what script changes are needed.

Don`t try the DietPi instructions on Armbian, that is not how it works.
The buidl_sdcard.sh chooses to use the Raspbian or DietPi instructions according to the hostname. 
Also there is no premade Armbian build for the RPi, because of the closed source part of the Pi.

I think for the Pi we should stick to the Raspbian as that is the best managed distro and for other SBC-s we should use the most often and recently updated manufacturer build. 

See my answer here: https://github.com/rootzoll/raspiblitz/issues/474#issuecomment-479387848

> My conclusion is:
> * so far we got a proof of concept of the RaspiBlitz system working reliably on other SBC-s and can be compatible with updates.
> *  it is far from being possible to provide a platform which can run easily on different boards without serious security trade-offs
> * if we are testing other SBC-s a best base distro should be picked individually for every board according to what is updated the most frequently and reliably.
> * this is the limitation of Linux on Arm, but we will continue to work with that given the price and form factor is unmatched in Intel based systems.

A minimal Debian or Ubuntu base is usually available like this one: https://wiki.odroid.com/odroid-xu4/os_images/linux/ubuntu_4.14/ubuntu_4.14 (based on an old kernel, because the board had trouble with the new 4.9. The 4.14 is the kernel the Armbian is built on also: https://www.armbian.com/odroid-xu4/) trouble with Kernels below 4.4, that they don`t support btrfs well (https://wiki.debian.org/Btrfs), so this needs further research

 If you want to test building on other repo-s   than it is probably the best to try to change the hostname to `Raspbian` before running the build_sdcard.sh because in most cases the directory architecture will be the same as in Raspbian.
Also make the user pi, if it would not exist already.

Will link my experimental repo once have more on this.

fluidvoice, 2019-04-03:
> I think for the Pi we should stick to the Raspbian as that is the best managed distro and for other SBC-s we should use the most often and recently updated manufacturer build.

I'm inclined to agree. Thinking on it a bit I started thinking about all the little tweaks in Raspbian which are hardware specific that I'm sure don't exist in other distros.  Things like startup overlays for enabling things on the GPIO pins and other headers.  Extended functionality such as intelligent on/off shutdown, etc.

gijswijs, 2019-05-03:
Not sure if this is the right thread, but the is there a possibility of using the raspiblits sh scripts on a intel based (mini) pc with Ubuntu?

fluidvoice, 2019-05-03:
> Not sure if this is the right thread, but the is there a possibility of using the raspiblits sh scripts on a intel based (mini) pc with Ubuntu?

I'm planning on testing/working-on this. @openoms created a distro branch here on Github to track progress, testing, etc. of this.  You're quite welcome to participate. Which mini-pc do you have or wanting to try?

openoms, 2019-05-03:
@gijswijs excellent idea, please join in with the testing. I've just received my Intel Atom based mini PC (AtomicPi) and ready for testing in a couple of days. There is a branch started for the x86 support: https://github.com/openoms/raspiblitz/tree/X86_64
So far only with Bitcoin and LND working. Will need to insert the logic to correct every platform sensitive part and update to the new 1.2 version.
Aiming to support VirtualBox as well to aid testing or just to be able to try the RaspiBlitz on a PC before buying any specific hardware.

Starting with a most basic Ubuntu image is the best choice I think. Many SBC-s come with Ubuntu as the default image, so it will be well tested.

gijswijs, 2019-05-04:
@openoms I'd love to help with testing. I'll receive my Partaker B13 i7 within 7 days, if all goes well.

openoms, 2019-05-30:
I don`t think we will ever change over to DietPi, but my Rock64 is back again.
A new kernel version updated in Armbian seem to have solved the memory chip issues. 
Details here: https://github.com/openoms/raspiblitz/issues/23#issuecomment-497284402

openoms, 2020-05-09:
I have pivoted to Armbian because of better support and more security focus.
Closing this, but panning to continue building to other SBC-s and distros.

Check the updates here: https://github.com/rootzoll/raspiblitz/tree/master/alternative.platforms
and the issue here:  https://github.com/rootzoll/raspiblitz/issues/1164

closed 2020-05-09 by openoms