raspiblitz/raspiblitz #248: Autonat problem

eurtnl, 2019-01-06:
Hi there,
I just turned on Autonat in services, but after reboot my RaspiBlitz gets in a loop where i have to unlock LND Wallet over and over.
How can i solve this problem or how to disable Autonat?

rootzoll, 2019-01-06:
yeah the lnd autonat is still broken
to deactivate manually there is a script to be called:
/home/admin/config.scripts/lnf.autonat.sh off

eurtnl, 2019-01-06:
OK, thanks i disabled the autonat

rootzoll, 2019-01-07:
I reopend the issue as a TODO for the 1.0 release to remove the option from the menu for now. This LND feature seems not ready to have it as an option for people to switch on generally. I will leave the script to switch it on manually for people how wnat to still test it.

rootzoll, 2019-01-10:
removed autonat from service menu - test if service menu is till working fine

rootzoll, 2019-01-15:
OK - service menu is tested and fixed.

closed 2019-01-15 by rootzoll