raspiblitz/raspiblitz #249: tor setup doesn't work

corbelantoine, 2019-01-07:
Hi,

I just tried to setup tor using the interface.
After restarting it seems like bitcoind fails to connect to tor:
`Cannot create socket for XXXX.onion:8333: unsupported network`
and LND fails to start.

I think tor failed to install because it's not running and I can install it manually.
During the setup I could notice an error while trying to mv the tor config file that didn't exist (probably because tor failed to install).

I found and rerun manually the tor script, looks like there is an issue with the gpg key used to check the signature of the tor package :

```
making sure services are not running
switching the TOR ON

*** Check if TOR service is functional ***
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0curl: (7) Failed to connect to 127.0.0.1 port 9050: Connection refused
TOR not running ... proceed with switching to TOR.

*** Updating System ***
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Get:2 http://deb.torproject.org/torproject.org stretch InRelease [4965 B]             
Ign:2 http://deb.torproject.org/torproject.org stretch InRelease                         
Hit:3 http://raspbian.raspberrypi.org/raspbian stretch InRelease                     
Fetched 4965 B in 3s (1325 B/s)                            
Reading package lists... Done
W: GPG error: http://deb.torproject.org/torproject.org stretch InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 74A941BA219EC810
W: The repository 'http://deb.torproject.org/torproject.org stretch InRelease' is not signed.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.

*** Install Tor ***
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  libzstd1 python-geoip python-torctl tor-geoipdb torsocks
Suggested packages:
  mixmaster torbrowser-launcher socat apparmor-utils obfs4proxy
The following NEW packages will be installed:
  libzstd1 python-geoip python-torctl tor tor-arm tor-geoipdb torsocks
0 upgraded, 7 newly installed, 0 to remove and 0 not upgraded.
Need to get 3542 kB of archives.
After this operation, 13.0 MB of additional disk space will be used.
WARNING: The following packages cannot be authenticated!
  tor tor-geoipdb
E: There were unauthenticated packages and -y was used without --allow-unauthenticated

*** Tor Config ***
rm: cannot remove '/etc/tor/torrc': No such file or directory
mv: cannot move './torrc' to '/etc/tor/torrc': No such file or directory
chmod: cannot access '/etc/tor/torrc': No such file or directory
chown: cannot access '/var/run/tor/': No such file or directory
```

corbelantoine, 2019-01-07:
Seems like the issue is related to debian and not tor only.
The following code fixes the issue:

```
sudo gpg --keyserver pgpkeys.mit.edu --recv-key  74A941BA219EC810
sudo gpg -a --export 74A941BA219EC810 | sudo apt-key add -
sudo apt-get update
```

rootzoll, 2019-01-07:
@corbelantoine much thanks. I added your addition to the 1.0 release todo list.

rootzoll, 2019-01-18:
Added the keys. Tests with TOR wirking fine. Closing this issue fro the release.

closed 2019-01-18 by rootzoll

joernroeder, 2019-02-08:
running into the same issue as @fluidvoice in #237. selected Tor in menu, and it installed afterwards.

when i run `internet.tor.sh on` after reboot again i get the following:

```
./internet.tor.sh on
making sure services are not running
switching the TOR ON

*** Check if TOR service is functional ***
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0curl: (7) Failed to connect to 127.0.0.1 port 9050: Connection refused
TOR not running ... proceed with switching to TOR.

dpkg-query: package 'tor-arm' is not installed and no information is available
Use dpkg --info (= dpkg-deb --info) to examine archive files,
and dpkg --contents (= dpkg-deb --contents) to list their contents.
*** Install TOR repo keys ***
gpg: keyserver receive failed: No data
gpg: WARNING: nothing exported
gpg: no valid OpenPGP data found.

gpg: keyserver receive failed: No data
gpg: WARNING: nothing exported
gpg: no valid OpenPGP data found.

*** Adding Tor Sources to sources.list ***
deb https://deb.torproject.org/torproject.org stretch main
deb-src https://deb.torproject.org/torproject.org stretch main
OK

*** Updating System ***
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:3 https://deb.nodesource.com/node_11.x stretch InRelease
Get:4 https://deb.torproject.org/torproject.org stretch InRelease [4,965 B]
Ign:4 https://deb.torproject.org/torproject.org stretch InRelease
Fetched 4,965 B in 1s (2,597 B/s)
Reading package lists... Done
W: GPG error: https://deb.torproject.org/torproject.org stretch InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 74A941BA219EC810
W: The repository 'https://deb.torproject.org/torproject.org stretch InRelease' is not signed.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: Target Sources (main/source/Sources) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:7
W: Target Packages (main/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6
W: Target Translations (main/i18n/Translation-en_GB) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6
W: Target Translations (main/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6
W: Target Sources (main/source/Sources) is configured multiple times in /etc/apt/sources.list:5 and /etc/apt/sources.list:7
W: Target Packages (main/binary-armhf/Packages) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6
W: Target Translations (main/i18n/Translation-en_GB) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6
W: Target Translations (main/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:4 and /etc/apt/sources.list:6

*** Installing dirmngr ***
Reading package lists... Done
Building dependency tree
Reading state information... Done
dirmngr is already the newest version (2.1.18-8~deb9u3).
The following packages were automatically installed and are no longer required:
  python-geoip python-torctl torsocks
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 63 not upgraded.

*** Install Tor ***
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  libzstd1 tor-geoipdb
Suggested packages:
  mixmaster torbrowser-launcher socat apparmor-utils obfs4proxy
The following NEW packages will be installed:
  libzstd1 tor tor-arm tor-geoipdb
0 upgraded, 4 newly installed, 0 to remove and 63 not upgraded.
Need to get 3,437 kB of archives.
After this operation, 12.9 MB of additional disk space will be used.
WARNING: The following packages cannot be authenticated!
  tor tor-geoipdb
E: There were unauthenticated packages and -y was used without --allow-unauthenticated

*** Tor Config ***
chown: cannot access '/var/run/tor/': No such file or directory

*** Installing NYX - TOR monitoring Tool ***
NYX already installed

*** Activating TOR system service ***
ReadWriteDirectories=-/mnt/hdd/tor
The unit files have no installation config (WantedBy, RequiredBy, Also, Alias
settings in the [Install] section, and DefaultInstance for template units).
This means they are not meant to be enabled using systemctl.
Possible reasons for having this kind of units are:
1) A unit may be statically enabled by being symlinked from another unit's
   .wants/ or .requires/ directory.
2) A unit's purpose may be to act as a helper for some other unit which has
   a requirement dependency on it.
3) A unit may be started when needed via activation (socket, path, timer,
   D-Bus, udev, scripted systemctl call, ...).
4) In case of template units, the unit is meant to be enabled with some
   instance name specified.

*** Changing bitcoin Config ***
Chain network already configured for TOR
*** Putting LND behind TOR ***
Make sutre LND is disabled
editing /etc/systemd/system/lnd.service
Enable LND again
Created symlink /etc/systemd/system/multi-user.target.wants/lnd.service → /etc/systemd/system/lnd.service.
OK

OK - TOR is now ON
needs reboot to activate new setting
```

so there seems to be an issue with Tor Keys.

rootzoll, 2019-02-10:
There were some problems with the key import. Not sure why that broke, but I changed it now and its working again. Fixed in #304 Will be part of version 1.0 release - just run the update once the realease is out: https://github.com/rootzoll/raspiblitz/tree/dev#updating-to-new-version

joernroeder, 2019-02-10:
Thanks! Seems like MIT keyserver is broken right now. Also unable to lookup keys via their interface

joernroeder, 2019-02-11:
just gave the new script a try and gpg keys are fetched now. i dont know if this is a follow up from the broken setup before but during tor installation the script halted an i was  asked to override the config file.

```
Configuration file '/etc/tor/torrc'
 ==> Modified (by you or by a script) since installation.
 ==> Package distributor has shipped an updated version.
   What would you like to do about it ?  Your options are:
    Y or I  : install the package maintainer's version
    N or O  : keep your currently-installed version
      D     : show the differences between the versions
      Z     : start a shell to examine the situation
 The default action is to keep your current version.
*** torrc (Y/I/N/O/D/Z) [default=N]
```

i manually approved but shouldn't this also be covered by the install script.

and there is a tiny typo as well:

*** Putting LND behind TOR ***
Make **sutre** LND is disabled
editing /etc/systemd/system/lnd.service
Enable LND again

rootzoll, 2019-02-11:
I will test again before realease, but I think that happened because you work with an old base image and just updated the scripts. The regular way to update is by reflashing the sd card: https://github.com/rootzoll/raspiblitz/blob/dev/FAQ.md#how-to-update-my-raspiblitz-after-version-098

Thanks for catching the typo :)