raspiblitz/raspiblitz #250: Displaying IPv6 address breaks Blitz Logo

rootzoll, 2019-01-07:
see https://twitter.com/wuestenigel/status/1082150083079163904

solution to implement: if there is an ipv6 instead of ipv4 just display the address:port and remove "Public" text and peer info so that it can fit into line without breaking it

wuestenigel, 2019-01-07:
Looking forward to v 1.0 ;)

fluidvoice, 2019-01-07:
Somewhat related. My node was displaying IPv6 address for many bootups then the last reboot it switched back to IPv4 despite me not changing anything. What is causing this?  In fact, I'd like to just make it default/stay on IPv4. How can I do this?

rootzoll, 2019-01-07:
Your router makes the choice between your public IP v4 or v6. The RaspiBlitz just tries to figure out what your running with and displays and uses it. 

If you want it to stay on IPv4 - see if you can find an option in your router to deactivate IPv6.

rootzoll, 2019-01-15:
marked for final testing ... even if no ipv6 abvailable to test, check if the modified script at least is not throwing any errors

rootzoll, 2019-01-15:
Was not able to test with IPv6 - but script now runs without errors. Closing issue.

closed 2019-01-15 by rootzoll

fluidvoice, 2019-01-16:
> Was not able to test with IPv6 - but script now runs without errors. Closing issue.

The reason this happens is not because of IPv6.  In my case the longer address showed up on the LCD when TOR was active.