raspiblitz/raspiblitz #251: generate new tls cert+key and macaroon

zer0trst, 2019-01-07:
Is there a reasonably easy way to generate new tls.cert, tls.key and new admin.macaroon file? and by reasonably easy I mean easier than making a new SD and starting again fresh..

I have tried deleting the files and then stop / start lnd but no new files. 

Any help would be awesome! Thanks

Smiggel, 2019-01-09:
Would like to know that as well. Especially if it's possible without losing open channels. I have some 2 way channels and I would like to keep them. However, RTL installation broke my ZAP wallet. Can no longer connect to it. So, hopefully there is a way to fix it.

rootzoll, 2019-01-09:
Aslo in latest RaspiBlitz release there is a script to create new tls certs. Try run on terminal:
`sudo /home/admin/config.scripts/lnd.newtlscert.sh`

If your Blitz does not have that script yet, check the source script and follow commands manually:
https://github.com/rootzoll/raspiblitz/blob/master/home.admin/config.scripts/lnd.newtlscert.sh

Please post if it worked, so I can close the issue. Thanks.

zer0trst, 2019-01-10:
Running version 0.99 and yes i do have that script. Tried running it and it appears success.

Here is the output

`making sure services are not running
deleting TLSCert
let lnd generate new TLSCert
wait until generated
(1/60) check for cert
2019-01-10 03:54:23.884 [INF] LTND: Version: 0.5.1-beta commit=, build=production, logging=default
2019-01-10 03:54:23.885 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2019-01-10 03:54:23.886 [INF] CHDB: Checking for schema update: latest_version=7, db_version=7
2019-01-10 03:54:23.886 [INF] RPCS: Generating TLS certificates...
2019-01-10 03:54:23.897 [INF] RPCS: Done generating TLS certificates
2019-01-10 03:54:23.901 [INF] RPCS: password RPC server listening on 127.0.0.1:10009
2019-01-10 03:54:23.902 [INF] RPCS: password gRPC proxy started at [::]:8080
2019-01-10 03:54:23.902 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
(2/60) check for cert
copy new cert to admin user
OK TLS certs are fresh`

So yes that achieves what I was wanting to do but still hasn't fixed the issue I am having after installing RTL which is why i wanted to try this. refer https://github.com/rootzoll/raspiblitz/issues/238

Thanks @rootzoll given this achieved what I asked, I will close topic.

Cheers :)

belgacoin, 2019-03-14:
Hello, to be clear: can I do this while LND is up and running?
Launch the script to create new tls certs. On terminal:
sudo /home/admin/config.scripts/lnd.newtlscert.sh

What does the node do after the script has run? Does it reboot, if yes, will I loose my channels?
Thanks :)

rootzoll, 2019-03-14:
This option can also be started from the SSH main menu if you go into the sub section "EXPORT" and the choose "RESET" - that will also create new macaroons.

No reboot needed using the path ybove. LND will just be restarted. You will not loose your channels, it might just take some minutes until all channel are back online after that procedure.

bmh2322, 2019-09-01:
Looks like I was experiencing a similar issue as the OP. I for some reason am unable to use my mobile zap wallet at all. It recognizes my lnd wallet but I have been unable to open it for days, tried a number of solutions. I followed the solution proposed by @smiggel , and was unable to get lnd started again. I produced the following error log.

`[3J[H[2J
***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.2
chainnetwork: bitcoin / main
 15:22:57 up 56 min,  2 users,  load average: 0.44, 0.56, 0.67

*** BCLOCKHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2019-09-01 14:29:32 BST; 53min ago
  Process: 6267 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
  Process: 6259 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
 Main PID: 6415 (bitcoind)
   CGroup: /system.slice/bitcoind.service
           └─6415 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

Sep 01 14:29:31 MerrimackValleyCrypto systemd[1]: Starting Bitcoin daemon...
Sep 01 14:29:32 MerrimackValleyCrypto systemd[1]: Started Bitcoin daemon.

*** LAST BCLOCKHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Sun 2019-09-01 14:26:51 BST, end at Sun 2019-09-01 15:22:58 BST. --
Sep 01 14:29:31 MerrimackValleyCrypto systemd[1]: Starting Bitcoin daemon...
Sep 01 14:29:32 MerrimackValleyCrypto systemd[1]: Started Bitcoin daemon.

*** LAST BCLOCKHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2019-09-01T13:33:21Z socket recv error Connection reset by peer (104)
2019-09-01T13:33:33Z New outbound peer connected: version: 70015, blocks=592764, peer=44
2019-09-01T13:33:39Z New outbound peer connected: version: 70015, blocks=592764, peer=46
2019-09-01T13:33:54Z New outbound peer connected: version: 70015, blocks=592764, peer=49
2019-09-01T13:35:52Z UpdateTip: new best=0000000000000000000e77b356f89c851169b413c9b0499ee6d65d798c0019a7 height=592765 version=0x20000000 log2_work=91.031424 tx=450811757 date='2019-09-01T13:35:27Z' progress=1.000000 cache=2.9MiB(25917txo) warning='52 of last 100 blocks have unexpected version'
2019-09-01T13:35:59Z Imported mempool transactions from disk: 4075 succeeded, 2879 failed, 0 expired, 9 already there
2019-09-01T13:39:10Z socket send error Broken pipe (32)
2019-09-01T13:45:22Z UpdateTip: new best=000000000000000000021f77f9ebbc3703e83e1031a9562ba385039dd5ef13c4 height=592766 version=0x2000e000 log2_work=91.031451 tx=450814037 date='2019-09-01T13:45:49Z' progress=1.000000 cache=3.7MiB(32759txo) warning='53 of last 100 blocks have unexpected version'
2019-09-01T13:50:07Z Leaving block file 1777: CBlockFileInfo(blocks=164, size=134147374, heights=592603...592766, time=2019-08-31...2019-09-01)
2019-09-01T13:50:07Z Pre-allocating up to position 0x1000000 in blk01778.dat
2019-09-01T13:50:09Z Pre-allocating up to position 0x100000 in rev01778.dat
2019-09-01T13:50:09Z UpdateTip: new best=0000000000000000000cc111a62385caac7fd72bd80215dde0734e4fac022d0d height=592767 version=0x20c00000 log2_work=91.031477 tx=450815611 date='2019-09-01T13:49:52Z' progress=1.000000 cache=4.1MiB(35990txo) warning='54 of last 100 blocks have unexpected version'
2019-09-01T13:50:40Z socket recv error Connection reset by peer (104)
2019-09-01T13:53:30Z socket recv error Connection reset by peer (104)
2019-09-01T13:55:15Z socket recv error Connection reset by peer (104)
2019-09-01T14:01:34Z socket recv error Connection reset by peer (104)
2019-09-01T14:06:09Z UpdateTip: new best=000000000000000000181d1d98a1e51dec516ce5417f2c291b012f47657d545d height=592768 version=0x20000000 log2_work=91.031504 tx=450818269 date='2019-09-01T14:05:57Z' progress=1.000000 cache=5.0MiB(44944txo) warning='54 of last 100 blocks have unexpected version'
2019-09-01T14:09:36Z UpdateTip: new best=00000000000000000014f89261168794a86b2620122a0fbbce945444f5b5b637 height=592769 version=0x20000000 log2_work=91.03153 tx=450820495 date='2019-09-01T14:08:44Z' progress=1.000000 cache=5.4MiB(48701txo) warning='53 of last 100 blocks have unexpected version'
2019-09-01T14:16:11Z socket recv error Connection reset by peer (104)
2019-09-01T14:20:29Z socket send error Connection reset by peer (104)

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2019-09-01 15:21:58 BST; 59s ago
  Process: 6203 ExecStart=/usr/local/bin/lnd --externalip=${publicIP}:${lndPort} (code=exited, status=1/FAILURE)
  Process: 6200 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 6203 (code=exited, status=1/FAILURE)

Sep 01 15:21:58 MerrimackValleyCrypto systemd[1]: lnd.service: Failed with result 'exit-code'.

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Logs begin at Sun 2019-09-01 14:26:51 BST, end at Sun 2019-09-01 15:22:58 BST. --
Sep 01 15:20:58 MerrimackValleyCrypto systemd[1]: Started LND Lightning Daemon.
[... cut ...]
Sep 01 15:21:58 MerrimackValleyCrypto systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Sep 01 15:21:58 MerrimackValleyCrypto systemd[1]: lnd.service: Unit entered failed state.
Sep 01 15:21:58 MerrimackValleyCrypto systemd[1]: lnd.service: Failed with result 'exit-code'.

*** LAST 30 LND INFO LOGS ***
sudo tail -n 30 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log

- RTL is OFF by config

*** HARDWARE TEST RESULTS ***
UndervoltageReports in Logs: 0
powerFAIL=0
powerWARN=0
powerMIN='1262500 microVolt'
tempFAIL=0
tempWARN=0
tempMAX='5150 centiGrad'`

rootzoll, 2019-09-01:
@bmh2322 There is something wrong in your lnd.conf file in line 10:
`/home/bitcoin/.lnd/lnd.conf:10: malformed key=value (RPC open to all connections on Port 10009)`

bmh2322, 2019-09-01:
@rootzoll , thank you. I went back and undid the changes where I removed the '#' comment symbol from 'rpclisten=0.0.0.0:10009' As far as I can tell that fixed the problem I had created. I am still having the issue with the mobile iOS zap wallet though. I will continue to tinker and see if it has anything to do with certs and macaroons.

The zap wallet recognized my lnd wallet but I am unable to open it.

Also thank you for creating such a comprehensive guide to making your own lightning node. I have really enjoyed this project and can't wait to see what development lies ahead of the lightning network!

rootzoll, 2019-09-01:
@bmh2322 there is a fundraising for RaspiBlitz running on tallycoin if you like :) https://tallyco.in/s/r5lx23/ .. about the ZAP Wallet: the v1.2 had some bugs regarding the walltes. I will hopefully release end of week the v1.3 with bugfixes for the wallets. If you you then have still issues after update, please open a new issue on that.

bmh2322, 2019-09-01:
@rootzoll I will certainly contribute to the fundraising. Eagerly awaiting v1.3, thank you and everyone else for all of your hard work!

P-Verbrugge, 2019-09-01:
It looks like zap corrupted my wallet dB. :-( I could open the wallet, but after updating the zap wallet is said it would work better with version 0.7 of lnd. After that message, the DB got corrupted. Can not restore my wallet, channels, etc anymore. :-( 

Channel backup with seed is not accepted. Says the channel already exists when I do a restore. After that I get the message, that the seed does not match. Not sure what is going on. Just be careful with zap and your node! I think I lost my funds forever.

Smiggel, 2019-09-01:
It looks like zap corrupted my wallet dB. :-( I could open the wallet, but after updating the zap wallet is said it would work better with version 0.7 of lnd. After that message, the DB got corrupted. Can not restore my wallet, channels, etc anymore. :-(

Channel backup with seed is not accepted. Says the channel already exists when I do a restore. After that I get the message, that the seed does not match. Not sure what is going on. Just be careful with zap and your node! I think I lost my funds forever.

rootzoll, 2019-09-01:
@Smiggel sounds like an issue to post over at the zap wallet github. "Channel backup" is to use with a fresh LND wallet, not sure whats up with your seed. For a last resort to recover funds see: https://medium.com/@guggero/did-you-lose-funds-on-the-lightning-network-because-of-a-disk-crash-8971b6a92494