raspiblitz/raspiblitz #252: Fatal LevelDB error: Corruption: CURRENT file does not end with newline

RobertSundh, 2019-01-07:
Hi.

I shutdown the node from meny because I had to change power on it. When I started up again it was stuck on the "----->starting up"

From log: 

2019-01-07T20:53:01Z Bitcoin Core version v0.17.0.1 (release build)
2019-01-07T20:53:01Z InitParameterInteraction: parameter interaction: -whitelistforcerelay=1 -> setting -whitelistrelay=1
2019-01-07T20:53:01Z Assuming ancestors of block 0000000000000000002e63058c023a9a1de233554f28c7b21380b6c9003f36a8 have valid signatures.
2019-01-07T20:53:01Z Setting nMinimumChainWork=0000000000000000000000000000000000000000028822fef1c230963535a90d
2019-01-07T20:53:01Z Using the 'standard' SHA256 implementation
2019-01-07T20:53:02Z Default data directory /home/bitcoin/.bitcoin
2019-01-07T20:53:02Z Using data directory /mnt/hdd/bitcoin
2019-01-07T20:53:02Z Using config file /home/bitcoin/.bitcoin/bitcoin.conf
2019-01-07T20:53:02Z Using at most 40 automatic connections (1024 file descriptors available)
2019-01-07T20:53:02Z Using 16 MiB out of 32/2 requested for signature cache, able to store 524288 elements
2019-01-07T20:53:02Z Using 16 MiB out of 32/2 requested for script execution cache, able to store 524288 elements
2019-01-07T20:53:02Z Using 4 threads for script verification
2019-01-07T20:53:02Z HTTP: creating work queue of depth 16
2019-01-07T20:53:02Z Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcauth for rpcauth auth generation.
2019-01-07T20:53:02Z scheduler thread start
2019-01-07T20:53:02Z HTTP: starting 4 worker threads
2019-01-07T20:53:02Z Cache configuration:
2019-01-07T20:53:02Z * Using 2.0MiB for block index database
2019-01-07T20:53:02Z * Using 12.2MiB for transaction index database
2019-01-07T20:53:02Z * Using 8.0MiB for chain state database
2019-01-07T20:53:02Z * Using 77.8MiB for in-memory UTXO set (plus up to 47.7MiB of unused mempool space)
2019-01-07T20:53:02Z init message: Loading block index...
2019-01-07T20:53:02Z Opening LevelDB in /mnt/hdd/bitcoin/blocks/index
2019-01-07T20:53:02Z Fatal LevelDB error: Corruption: CURRENT file does not end with newline
2019-01-07T20:53:02Z You can use -debug=leveldb to get more complete diagnostic messages
2019-01-07T20:53:02Z Fatal LevelDB error: Corruption: CURRENT file does not end with newline
2019-01-07T20:53:02Z : Error opening block database.
Please restart with -reindex or -reindex-chainstate to recover.
2019-01-07T20:53:02Z Aborted block database rebuild. Exiting.
2019-01-07T20:53:02Z Shutdown: In progress...
2019-01-07T20:53:02Z scheduler thread interrupt
2019-01-07T20:53:02Z Shutdown: done

So I guess the blockchain is screwed? I tryed running the -reindex-chainstate but I cant figure out how to enter the command.

**Also this might be related :** 

admin@Lightning:~ $ lncli unlock
Input wallet password: 

lnd successfully unlocked!
admin@Lightning:~ $ lncli getinfo
[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"

Any ideas guys?

Thanks /B

fluidvoice, 2019-01-08:
You could try this from SSH shell:
`sudo systemctl stop bitcoind.service`
`sudo bitcoind -daemon -reindex -conf=/home/bitcoin/.bitcoin/bitcoin.conf`
wait till you see lines that contain "progress=" then hit `ctrl+c` to cancel and then do
`sudo reboot`

RobertSundh, 2019-01-08:
wow it might have worked!

  Waiting for final Blockchain Sync                │  
            │  Progress: .0185130272827029500 %                 │  
            │  Please wait - this can take some long time.      │  
            │  Its OK to close terminal and ssh back in later.

Gonna wait for that sync then i get back if it worked. Thanks alot hopefully I can buy you a lightning beer someday.

fluidvoice, 2019-01-08:
Well that's basically started the sync from scratch. Was your blockchain 100% synced before this problem happened?

RobertSundh, 2019-01-08:
Lol yeah I realised that, thought it was just doing the missing one/two days.

Yes it was running synced before shutdown.

Maybe I will save time just closing channels and redo from scratch.🙂

rootzoll, 2019-01-09:
@RobertSundh if you have channels with funding open it might be better to wait for reindex - I think its not possible to close channels if bitcoind is not ready.

I will close this issue for now - but if anybody has a hint how to help RaspiBlitz prevent such LevelDB errors, please share.

closed 2019-01-09 by rootzoll